Koenkk/zigbee2mqtt #7871: Phillips Hue Remote - stuck in "update_available" loop?

SHxKM, 2021-06-24:
OS: MacOS
Adapter: Conbee II
HA version: 2021.6.4
z2m version: 1.19.1

I recently migrated to Z2M from ZHA. I noticed there was an update available for my Hue remote (324131092621), so I initiated the update the first time. Upon finishing the process, I saw something along the lines of date_code: `2016XXXX` and I thought it'd be weird if that was the last time Phillips updated this one. Immediately after, there was a message that an update is available for this device, again. I thought it's kinda weird but I went on with it.

But then after the *second* update finished, this "update available" message appeared again. Should I try to update for the 3rd time? is this normal? (the remote's battery is dropping around 8% on each update like this).

```
Zigbee2MQTT:info  2021-06-25 01:42:48: MQTT publish: topic 'zigbee2mqtt/bedroom_dimmer_switch', payload '{"action":null,"battery":87.5,"brightness":255,"counter":1,"linkquality":255,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-06-25 01:42:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"bedroom_dimmer_switch"},"meta":{"device":"bedroom_dimmer_switch","from":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"},"status":"update_succeeded","to":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}},"type":"ota_update"}'
Zigbee2MQTT:info  2021-06-25 01:42:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"id":"bedroom_dimmer_switch","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"status":"ok"}'
Zigbee2MQTT:info  2021-06-25 01:42:53: MQTT publish: topic 'zigbee2mqtt/bedroom_dimmer_switch', payload '{"action":null,"battery":87.5,"brightness":255,"counter":1,"linkquality":255,"update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-06-25 01:42:53: Update available for 'bedroom_dimmer_switch'
Zigbee2MQTT:info  2021-06-25 01:42:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update available for 'bedroom_dimmer_switch'","meta":{"device":"bedroom_dimmer_switch","status":"available"},"type":"ota_update"}'
```

SHxKM, 2021-07-02:
After the dimmer updated twice (I didn't take note of the versions it was updated from/to), I am now unable to update at all - and it seems stuck at firmware version 5.45.1.17846 (build date: 20160302). It seems that Zigbee2MQTT downgraded my version through the update mechanism because this was previously connected to a Hue bridge and was showing "up-to-date" status.

SHxKM, 2021-07-02:
Right after writing my last message I managed to get an update to start. As I suspected Z2M is "updating" the device to the same version over and over again (I suspect the device had newer firmware before Z2M "updated" the firmware). Here are the logs I could catch:

```
{"device":"glob_ph_dimmer_1","progress":98.23,"status":"update_progress"},"type":"ota_update"}'
Debug OTA update at 99.42%, remaining 6.954162937214861 seconds
Info Update of 'glob_ph_dimmer_1' at 99.42%, ≈ 0 minutes remaining
Info MQTT publish: topic 'zigbee2mqtt/glob_ph_dimmer_1', payload '{"action":null,"battery":99.5,"brightness":255,"counter":1,"linkquality":255,"update":{"progress":99.42,"remaining":7,"state":"updating"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'glob_ph_dimmer_1' at 99.42%, ≈ 0 minutes remaining","meta":{"device":"glob_ph_dimmer_1","progress":99.42,"status":"update_progress"},"type":"ota_update"}'
Debug Got upgrade end request for '0x0017880108096dd4': {"status":0,"manufacturerCode":4107,"imageType":265,"fileVersion":1107324829}
Debug Update succeeded, waiting for device to restart
Debug Received Zigbee message from 'glob_ph_dimmer_1', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":184}' from endpoint 2 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/glob_ph_dimmer_1', payload '{"action":null,"battery":92,"brightness":255,"counter":1,"linkquality":255,"update":{"progress":99.42,"remaining":7,"state":"updating"},"update_available":false}'
```

Two minutes after that I got a message "an update is available". So something seems off about Z2M's OTA mechanism.

SHxKM, 2021-07-03:
I have a Hue bridge handy so I've paired this remote with it and the firmware is indeed `5.45.1.17846`. I'm now using the Hue bridge to update the device, and will report if Z2M is at least able to tell that it should not have any more recent firmware after that.

I don't consider this a solution as I have XX devices in my network which I would like to update using Z2M. Just checking if Z2M is at least able to tell if the firmware is the most recent.

james-fry, 2021-08-16:
I am seeing the same behaviour.

SHxKM, 2021-08-16:
@Koenkk can we reopen this issue?

james-fry, 2021-08-16:
I just tried to reproduce it and the upgrade succeeded.
I have 4 of these dimmers and TBH they dont work that well hence the attempt to upgrade.
The upgrade that succeeded was from 5.45.1.17846 to 6.1.1.28573.
Despite completing two upgrades of these devices today, only one is on 6.1.1.28573.

This the log for update start and complete from the other dimmer that didnt succeed:
```
info  2021-08-16 12:03:17: Updating 'Hue Dimmer Garden' to latest firmware
info  2021-08-16 12:03:17: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":"Updating 'Hue Dimmer Garden' to latest firmware","meta":{"device":"Hue Dimmer Garden","status":"update_in_progress"},"type":"ota_update"}'
info  2021-08-16 12:03:17: MQTT publish: topic 'zigbee2mqtt_cc1352p/Hue Dimmer Garden', payload '{"action":null,"battery":44.5,"brightness":223,"counter":1,"last_seen":"2021-08-16T12:03:17+01:00","linkquality":84,"update":{"state":"available"},"update_available":true}'
info  2021-08-16 12:03:18: MQTT publish: topic 'zigbee2mqtt_cc1352p/Hue Dimmer Garden', payload '{"action":null,"battery":44.5,"brightness":223,"counter":1,"last_seen":"2021-08-16T12:03:18+01:00","linkquality":84,"update":{"state":"available"},"update_available":true}'
info  2021-08-16 12:03:19: MQTT publish: topic 'zigbee2mqtt_cc1352p/Hue Dimmer Garden', payload '{"action":null,"battery":44.5,"brightness":223,"counter":1,"last_seen":"2021-08-16T12:03:19+01:00","linkquality":87,"update":{"state":"available"},"update_available":true}'

info  2021-08-16 12:48:12: Update of 'Hue Dimmer Garden' at 100.00%
info  2021-08-16 12:48:12: MQTT publish: topic 'zigbee2mqtt_cc1352p/Hue Dimmer Garden', payload '{"action":null,"battery":34,"brightness":223,"counter":1,"last_seen":"2021-08-16T12:47:20+01:00","linkquality":72,"update":{"progress":100,"remaining":12,"state":"updating"},"update_available":false}'
info  2021-08-16 12:48:12: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":"Update of 'Hue Dimmer Garden' at 100.00%","meta":{"device":"Hue Dimmer Garden","progress":100,"status":"update_progress"},"type":"ota_update"}'
info  2021-08-16 12:48:35: Finished update of 'Hue Dimmer Garden'
info  2021-08-16 12:48:35: MQTT publish: topic 'zigbee2mqtt_cc1352p/Hue Dimmer Garden', payload '{"action":null,"battery":34,"brightness":223,"counter":1,"last_seen":"2021-08-16T12:47:20+01:00","linkquality":72,"update":{"state":"idle"},"update_available":false}'
info  2021-08-16 12:48:35: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":{"id":"Hue Dimmer Garden","transaction":"ee7vj-27"},"meta":{"device":"Hue Dimmer Garden","from":{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"},"status":"update_succeeded","to":null},"type":"ota_update"}'
info  2021-08-16 12:48:35: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"id":"Hue Dimmer Garden","to":null},"status":"ok","transaction":"ee7vj-27"}'
```

I'll try this one again and see what happens...

james-fry, 2021-08-16:
Another update.
I unpaired, reset and re-paired the controller that "failed" its update.
When it is re-paired the controller is reporting FW version 6.1.1.28573.

These are pretty problematic devices - too 3 pairing attempts to get it to announce, interview and configure properly...

closed 2021-08-24 by github-actions[bot]

SHxKM, 2021-11-27:
@james-fry Seeing as this has been closed again, maybe we should investigate further and open a new issue? I'm unable to update any of my remotes.

hidra000, 2022-01-08:
And i have the exact same problem

jbliesener, 2022-01-19:
Same here.

[image]

Note that one of the E1524/E1810's updated but now is stuck in a loop, showing a recent firmware number but an old firmware date. Also note that one of the repeaters shows a different firmware date with the same firmware version. Open another issue?

Raycharges, 2022-02-13:
Same here: Hue Amaze and remote bought together, it seems I'm not able to get them out of their current firmwares:

[image]

Everytime, at the end of the update, it says that the version has been ungraded from XXX to XXX (target version is the same as the starting one) then, if I do another update check, of course it says there's an update available.

I don't find any errors in the logs, everything seems fine but let me know if you could use the logs anyway.

Home Assistant version: 2022.2.6
Zigbee2Mqtt version: 1.23.0-1
Host: Raspberry 4
Gateway: Conbee II

BusinessPerson, 2023-11-15:
I have the same issue. The Zigbee2mqtt interface shows current version as : 5.45.1.17846
Update request starts ok from that screen, then fails. Logs show:
Debug 2023-11-15 07:43:37Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"HueDimmerStudy","transaction":"mgft9-2"}'
Info 2023-11-15 07:43:37Updating 'HueDimmerStudy' to latest firmware
Debug 2023-11-15 07:49:46Received Zigbee message from 'HueDimmerStudy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Debug 2023-11-15 07:49:46Device 'HueDimmerStudy' requested OTA
Debug 2023-11-15 07:49:55Received Zigbee message from 'HueDimmerStudy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Debug 2023-11-15 07:49:55Device 'HueDimmerStudy' requested OTA
Debug 2023-11-15 07:50:05Received Zigbee message from 'HueDimmerStudy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Debug 2023-11-15 07:50:05Device 'HueDimmerStudy' requested OTA
Debug 2023-11-15 07:50:16Received Zigbee message from 'HueDimmerStudy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Debug 2023-11-15 07:50:16Device 'HueDimmerStudy' requested OTA
Debug 2023-11-15 07:50:25Received Zigbee message from 'HueDimmerStudy', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107314102,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Debug 2023-11-15 07:50:25Device 'HueDimmerStudy' requested OTA

Environment details are:
HA Core: 2023.11.2
Supervisor: 2023.11.3
OS: 11.1
Hardware: HomeAssistant Yellow
Zigbee2MQTT Version: 1.33.2-1

BarsMonster, 2025-05-06:
I have the same issue with update loop on the same switch. Is it possible to somehow manually update it to latest version?

OneCreek, 2025-05-20:
same here...
I get:
```
Device '0x00178801xxx' was updated from '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"id":"0x00178801xxx","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"status":"ok","transaction":"en0hf-1"}'
```

OneCreek, 2025-07-24:
> same here... 

Update: ["repairing the device should fix it"](https://github.com/Koenkk/zigbee2mqtt/issues/5869#issuecomment-785347503) from #5869 helped solving my issue

DanaGoyette, 2026-11-09:
I collected more data on this issue with a Philips Hue Dimmer (324131092621).

Initial state before OTA:
```
{
  "installed_version": 1124102917,
  "latest_version": 1124103169,
  "state": "available"
}
```

I ran the OTA update. The transfer progressed normally and took a substantial amount of time, reaching 100%.

After the update completed, Z2M showed:
```
App: 4.3 build 0
Stack: 7.4 build 1
Status: Up-to-date
```

I then restarted Zigbee2MQTT without waking the remote. After the restart, the UI still showed:
```
App: 4.3 build 0
Stack: 7.4 build 1
Status: Up-to-date
```

After waking the remote with button presses, MQTT messages contained:

```
{
  "installed_version": 1124103169,
  "latest_version": 1124103169,
  "state": "idle"
}
```

This suggested that the update had succeeded.

I then used "Check for OTA update". Immediately afterward, Z2M reported:

`OTA update available for 'Hue Remote'`

and subsequent MQTT messages changed to:

```
{
  "installed_version": 1124102917,
  "latest_version": 1124103169,
  "state": "available"
}
```

I restarted Zigbee2MQTT again and woke the remote with button presses. The remote continued to report:

```
{
  "installed_version": 1124102917,
  "latest_version": 1124103169,
  "state": "available"
}
```

So the observed sequence is:
```
Before OTA:
  1124102917 (update available)

After OTA:
  1124103169 (up to date)

After Z2M restart:
  1124103169 (up to date)

After waking remote:
  1124103169

After "Check for OTA update":
  1124102917 (update available)

After another Z2M restart and waking remote:
  1124102917 (update available)
```

The confusing part is that the device appears to report `1124103169` after the OTA completes, but after an OTA availability check it reverts back to reporting `1124102917`.

I have tried this update something like 20 times, and every single time, it goes back to claiming the old version number.
Resetting the Hue Dimmer V1 using the pinhole didn't change anything.