Koenkk/zigbee2mqtt #7874: ZNLDP12LM power_outage_memory support doesn't work

dreem2001, 2021-06-25:
## What happened
I have Xiaomi Bulb ZNLDP12LM. I try to configure power outage memory for the device and once I send the payload as described in the documentation the Zigbee2Mqtt returns an error.
MQTT request:
`zigbee2mqtt/CellLight1/set
{"power_outage_memory": "true"}`
Zigbee logs:
`Error No converter available for 'power_outage_memory' (true)`

## What did you expect to happen
Expect that the device applies the power_outage_memory property and will not light after power outage.

## How to reproduce it (minimal and precise)
Send the request in MQTT cleint.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
Adapter firmware version:  revision: 20190425 type:zStack30x

Koenkk, 2021-06-26:
You are on and old z2m version, try updating to the latest first.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

dreem2001, 2021-06-28:
Thanks for the reply.
I switched HA to 11.19.1-1. But it didn't work for me still. 
But it started showing this attribute among available in MQTT
`{"brightness":253,"color_temp":300,"linkquality":44,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}`

I found that there's OTA update, updated the bulbs to 1.13.2 and after it power_outage_memory started working! 
Thanks for the good advice!

closed 2021-06-28 by dreem2001