Koenkk/zigbee2mqtt #7879: Aqara Smart wall switch H1 EU (No neutral. Double Rocker) WS-EUK02 support

k-szabi, 2021-06-25:
Hi, 

I've tried to add a new Aqara Smart wall switch H1 EU (No neutral. Double Rocker) to my zigbee2mqtt network according to the guidance on https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Turning on off from home assistant is working fine, but is I turning on manually I got status only from the left side switch and not from the right side switch. I got the following error during the configuration:

```
Zigbee2MQTT:error 2021-06-25 16:45:14: Failed to configure '0x54ef4410000f4bce', attempt 3 (Error: ConfigureReporting 0x54ef4410000f4bce/2 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'SOFTWARE_FAILURE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:221:23)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:488:26)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

This is the content of my external converter file:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.switch.l2aeu1'],
        model: 'WS-EUK02',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch H1 (no neutral, double rocker)',
        extend: extend.switch(),
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint2);
        },
};
module.exports = definition;
```
Do you have any idea, what is missing, or wrong with this config?

Br,
Szabolcs

tavalin, 2021-06-25:
There is already an issue open for this - please see [here](https://github.com/Koenkk/zigbee2mqtt/issues/6524)

k-szabi, 2021-06-26:
Hi @tavalin,
Thanks. I haven't found it before.

Koenkk, 2021-06-26:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/6524

closed 2021-06-26 by Koenkk