Koenkk/zigbee2mqtt #7881: Support new manufId for Tuya TS0202

1iverea9er, 2021-06-25:
Device:
https://www.zigbee2mqtt.io/devices/TS0202.html

Work with following external converter^
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'}],
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper()],
};

module.exports = definition;`

..
database.db entry:
`{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138951741ff7f","nwkAddr":15886,"manufId":4417,"manufName":"_TZ3000_mmtwjmaq","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_mmtwjmaq","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00193ae944","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1624646544567}`
..

Koenkk, 2021-06-26:
Added, thanks!

closed 2021-06-26 by Koenkk

vgdh, 2022-02-25:
Also works with another Tuya motion sensor with some changes

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_otvn3lne'}],
zigbeeModel: ['TS0202'],
model: 'TS0202',
vendor: 'TuYa',
description: 'Motion sensor',
fromZigbee: [fz.ias_occupancy_alarm_1],
toZigbee: [],
exposes: [e.occupancy(), e.battery_low()],
};

module.exports = definition;
```