Koenkk/zigbee2mqtt #7886: OTA fails for ikea tradfri

new-kirte, 2021-06-26:
## What happened
[image]

## What did you expect to happen
Firmware updated

## How to reproduce it (minimal and precise)
Click on Update device firmware

## Debug info
Zigbee2MQTT version:
Adapter hardware:  Sonoff bridge (tried both)
Adapter firmware version: 
Sonoff ZHABridge
Tasmota
Program Version	9.5.0(zbbridge)
Build Date & Time	2021-06-17T08:27:09
Core/SDK Version	2_7_4_9/2.2.2-dev(38a443e)
Uptime	0T02:12:56

Log file:

Debug Received Zigbee message from 'IKEA TRADFRI USB Charger', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":536974883,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Debug Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4354,"fileVersion":536974883}'
Debug getNewImage for '0x000d6ffffe9a89ed', meta {"fileVersion":587662897,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.14.024/bin/10037603-3.1-TRADFRI-signal-repeater-2.3.070.ota.ota.signed"}
Debug getNewImage for '0x000d6ffffe9a89ed', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4476,"imageType":4354,"fileVersion":587662897,"zigbeeStackVersion":2,"otaHeaderString":"GBL GBL_tradfri_zigbee_repeater\u0000","totalImageSize":201736}
Debug Got new image for '0x000d6ffffe9a89ed'
Debug Starting upgrade
Debug Received Zigbee message from 'IKEA TRADFRI USB Charger', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":536974883,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Debug OTA update at 0%, remaining Infinity seconds
Info Update of 'IKEA TRADFRI USB Charger' at 0.00%
Info MQTT publish: topic 'zigbee2mqtt/IKEA TRADFRI USB Charger', payload '{"device":{"applicationVersion":32,"dateCode":"20180828","friendlyName":"IKEA TRADFRI USB Charger","hardwareVersion":1,"ieeeAddr":"0x000d6ffffe9a89ed","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1746","networkAddress":39443,"powerSource":"Mains (single phase)","softwareBuildID":"2.0.019","stackVersion":98,"type":"Router","zclVersion":3},"linkquality":128,"update":{"progress":0,"state":"updating"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'IKEA TRADFRI USB Charger' at 0.00%","meta":{"device":"IKEA TRADFRI USB Charger","progress":0,"status":"update_progress"},"type":"ota_update"}'
Debug OTA update at 1.88%, remaining 2600.3060673684213 seconds
Info Update of 'IKEA TRADFRI USB Charger' at 1.88%, ≈ 43 minutes remaining
Info MQTT publish: topic 'zigbee2mqtt/IKEA TRADFRI USB Charger', payload '{"device":{"applicationVersion":32,"dateCode":"20180828","friendlyName":"IKEA TRADFRI USB Charger","hardwareVersion":1,"ieeeAddr":"0x000d6ffffe9a89ed","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1746","networkAddress":39443,"powerSource":"Mains (single phase)","softwareBuildID":"2.0.019","stackVersion":98,"type":"Router","zclVersion":3},"linkquality":128,"update":{"progress":1.88,"remaining":2600,"state":"updating"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'IKEA TRADFRI USB Charger' at 1.88%, ≈ 43 minutes remaining","meta":{"device":"IKEA TRADFRI USB Charger","progress":1.88,"status":"update_progress"},"type":"ota_update"}'
Debug Got upgrade end request for '0x000d6ffffe9a89ed': {"status":149,"manufacturerCode":4476,"imageType":4354,"fileVersion":587662897}
Debug Update failed with reason: 'aborted by device'
Info MQTT publish: topic 'zigbee2mqtt/IKEA TRADFRI USB Charger', payload '{"device":{"applicationVersion":32,"dateCode":"20180828","friendlyName":"IKEA TRADFRI USB Charger","hardwareVersion":1,"ieeeAddr":"0x000d6ffffe9a89ed","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1746","networkAddress":39443,"powerSource":"Mains (single phase)","softwareBuildID":"2.0.019","stackVersion":98,"type":"Router","zclVersion":3},"linkquality":128,"update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'IKEA TRADFRI USB Charger' failed (Update failed with reason: 'aborted by device')","meta":{"device":"IKEA TRADFRI USB Charger","status":"update_failed"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"IKEA TRADFRI USB Charger"},"error":"Update of 'IKEA TRADFRI USB Charger' failed (Update failed with reason: 'aborted by device')","status":"error","transaction":"3l7oq-12"}'
Error Update of 'IKEA TRADFRI USB Charger' failed (Update failed with reason: 'aborted by device')

Tonio16, 2021-06-27:
Hello
Did you looked at the documentation?

> Using the IKEA TRADFRI test server #
The current published firmwares (as of 2020-12-09) by IKEA are corrupted which makes the devices reject the update. The firmwares published on the IKEA test server are not corrupted. You can instruct Zigbee2MQTT to use the test server by adding the following to your configuration.yaml.
WARNING: Use at your own risk!
advanced:
  ikea_ota_use_test_url: true

MattWestb, 2021-06-28:
Dont using the test feed for updating controller devices then IKEA have not certified not good working updates that can making problem.

Wot i knowing is all normal light working OK in the test feed but using the normal for updating remotes until IKEA have making then working OK.

new-kirte, 2021-06-29:
The OTA works with Conbee II and breaks with Zonoff, so it seems that the issue is definitely Sonoff

nonflammable, 2021-06-29:
[FEATURE REQUEST]
Add "Downgrade" option in OTA tab or "Update from file"

MattWestb, 2021-06-29:
Most (or perhaps all) IKEA controller device is flagged as not accepting downgrading the firmware but most (perhaps all) lights is possible downgrading with deCONNZ or ZHA.
Download the latest Zigbee certificate files for you device and look or the part for the OTA cluster.
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=ikea

If having  one J-Link flasher its always possible "extracting" the code and flashing it to the device but its out of the scope here ;-))

closed 2021-08-06 by github-actions[bot]

psi-4ward, 2021-09-17:
Seems IKEA has closed the access: http://fw.test.ota.homesmart.ikea.net/feed/version_info.json gives me a 403 Forbidden

MattWestb, 2021-10-01:
The test feed was down for some days (its one internal IKEA resource and is can being down then they like ti) but is up but only with one old set with the GW version 1.11.47 that is over one year old but hopefully they have getting all things OK and releasing all (current) controllers firmware in the test feed.

I have no hope getting updates for the old motion sensor and also the new one and the round gyro dimmer then its first gen or early second gen devices (all having getting Zigbee 3 updates but not the old motion sensor).

For information: IKEA is using Light Link protocol for forming the network that is also possible using in Zigbee 3 but no open system have implanting Touch Link commands for joining devices so its not possible for us to using.

In Light Link the (distributed network security = no coordinator) its one controlling device that  forming the network and allocating network addressed and range and the same for groups (if having one GW its the GW that is the master and can "giving" addresses and groups to other controlling devices in the network).
The controlling devices can joining router devices and giving them addresses and groups in there range and can controlling them and also sharing the address space with other controllers joined to there address range.
That is completely different and more complex then doing "radical changes" for not braking all devices in the systems that is up and running so IKEA need very much testing before relenting the new batch of controlling firmware.

IKEA first priority is there Light link system with GW and standalone (without GW) but they is very aware of the "open system" then they was fixing the first gen lights and controllers so they is working with other like Philips HUE (and Philips was not doing the same). But they still cant braking all there systems that all customer is having up and running.

If all is going well we is getting one working update for our devices this year 2 (nearly all device have getting it the last year but not the old motion....).