Koenkk/zigbee2mqtt #7891: Aqara Opple Switch, switches off ALL devices - How to disable broadcast?

jilles, 2021-06-27:
# Support request
## What happened
I paired the Aqara Opple switch 1 band, just like four other switches. But this one somehow disables (and enables) ALL other devices in the network. 

## What did you expect to happen
I expected it to work the same as all the other Aqara Opple switches; pair it and then use it to switch the lights I want (via Domoticz). This works perfectly fine for all others.

## How to reproduce it (minimal and precise)
- Pair device (first time OR after full removal)
- Use buttons (and experience unexpected behaviour)

## Debug Info
Zigbee2MQTT version: 1.18.3 commit: f2e39af1
Adapter hardware: CC26X2R1 (ZZH!)
Coordinator type: zStack3x0
Adapter firmware version: 20201026

SolovevVyacheslav, 2021-07-05:
Try to change operation mode to "event" on the exposes tab. Press the back button before changing the mode to wake up device.
[image]

Sundown89, 2021-07-14:
I had the same issue and I found out that this device was somehow binded to several devices when it was freshly paired. I tried to unbind the Opple from these devices and this solved the issue!

srtxema, 2021-07-18:
how you unbind for the devices? i obtain this error where i unbind by zigbee2mqtt web interface:
Failed to unbind cluster 'genLevelCtrl' from 'Boton_Aqara_4_Cuadruple' to 'Coordinator' (Error: Unbind 0x04cf8cdf3c7948a7/1 genLevelCtrl from '0x00212effff066630/1' failed (waiting for response TIMEOUT))

srtxema, 2021-07-18:
> Try to change operation mode to "event" on the exposes tab. Press the back button before changing the mode to wake up device.
> [image]

i obtain this error
Error Publish 'set' 'operation_mode' to 'Boton_Aqara_6_Sextuple' failed: 'Error: Write 0x04cf8cdf3c7d414e/1 aqaraOpple({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (no response received)'

Koenkk, 2021-07-19:
@srtxema you need to click a button on the device right before sending the command to wake it up.

srtxema, 2021-07-21:
thanks a lot! Solve!

rocket59, 2021-08-22:
Same thing just happened to me with my opple switch. Re-pairing by restricting it to only one of the routers (not permit join all) solved the issue.

closed 2021-09-29 by github-actions[bot]