Koenkk/zigbee2mqtt #7892: suddenly: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) (not upgrade process)

kpcz, 2021-06-27:
## What happened

Was adding a device via Homeassistant to zigbee2mqtt, everything was working fine, but at one moment all zigbee devices hang, giving me the error:

`Publish 'set' 'state' to '0xbc33acfffefe0e5c' failed: 'Error: Command 0xbc33acfffefe0e5c/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

Tried to restart mq2zigbee addon, or the whole HA setup, it gives logs that

```
Zigbee2MQTT:info  2021-06-27 16:12:56: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2021-06-27 16:12:56: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:error 2021-06-27 16:13:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-06-27 16:13:15: Failed to start zigbee
Zigbee2MQTT:error 2021-06-27 16:13:15: Exiting...
Zigbee2MQTT:error 2021-06-27 16:13:15: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

```

## Debug info
Zigbee2MQTT version:  1.18.1-1
Adapter hardware: usb-Silicon_Labs_slae.sh_cc2652rb_stick
Adapter firmware version: how can I check?
zigbee devices: ~52 devices

Please help me to resolve this issue, as I am afraid that I will do something wrong, that I will need to manually re-join all 52 devices, as it is really painfull process (did it already with less devices when migrated to cc2652rb_stick ) 

As I read in other posts, the issue began with zigbee2mqtt upgrade to 1.18, but I did not do any update, it was done already previously.

kpcz, 2021-06-27:
Replugging the adapter helped, addon started to work .  Could it be that cc2652rb is not enought powerfull for ~52 devices? I also have one CC2530 router, could this router cause issues?

arkk4, 2021-06-27:
I have a similar problem, coordinator CC2530+2592 (EBYTE-18-TBH-27)
Once a day (sometimes once every two) I catch first
error 
```
2021-06-27 20:09:39: Publish 'set' 'state' to 'Repeater_3_7f' failed: 'Error: Comman
d 0x00124b0022663995 / 1 genOnOff.off ({}, {"sendWhenActive": false, "timeout": 10000, "disableResponse": false, "disableRecovery": false, "disableDefaultResponse": false, "direction": 0, "srcEndpoint ": null," reservedBits ": 0," manufacturerCode ": null," transactionSequenceNumber ": null," writeUndiv ": false}) failed (SRSP - AF - dataRequest after 6000ms) '
```
all devices do not respond to commands, but periodic reports can sometimes still be sent, when the service is restarted, I receive
```
Jun 27 20:49:20 gwshev3093 npm [8682]: # 033 [31mZigbee2MQTT: error # 033 [39m 2021-06-27 20:49:20: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Jun 27 20:49:20 gwshev3093 npm [8682]: at ZStackAdapter. <anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:93:31)
Jun 27 20:49:20 gwshev3093 npm [8682]: at Generator.throw (<anonymous>)
Jun 27 20:49:20 gwshev3093 npm [8682]: at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
```
Service version 1.19.1-Dev on orange Pi zero
5 devices connected, of which 4 routers and one end device

Restarting the service and the coordinator helps, but unfortunately the hosts (and this is manifested in three of my three installations) are far from me
``
homeassistant: true
permit_join: false
mqtt:
  base_topic: my/topic 
  server: myserver
  user: user
  password: password 
serial:
  port: /dev/ttyUSB0
retain: true
advanced:
  log_directory: data/log/
  log_file: latest.log
  log_level: info
  channel: 20
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
frontend:
  port: 8080
  host: 0.0.0.0
  auth_token: mytoken
experimental:
  transmit_power: 27
devices:
  '0x00124b0022094383':
    friendly_name: '0x00124b0022094383'
  '0x00124b001f4f57cd':
    friendly_name: '0x00124b001f4f57cd'
  '0x00124b001ec8410e':
    friendly_name: domofon_r3
  '0x00124b0022663995':
    friendly_name: Repeater_3_7f
  '0x00124b00214379b9':
    friendly_name: Repeater_3_3f
  '0x00124b0022662a9c':
    friendly_name: podval_3-1
  '0x00124b00220943b8':
    friendly_name: podval_3
device_options:
  legacy: false
``

Koenkk, 2021-06-28:
@kpcz make sure you are on the latest firmware and z2m version first (1.19.1). 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0. Firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin . Your stick should be able to handle that network.

@arkk4 in your case the cc2530 is likely not powerful enough, try either the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing or upgrade to a more powerful adapter: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

kpcz, 2021-06-28:
Thanks. indeed I am not on latest addon version. But when I add repo https://github.com/zigbee2mqtt/hassio-zigbee2mqtt I now see two Home Assistant Add-on: Zigbee2mqtt addons, one installed with 1.18.1 and one that is not installed with 1.19.1. What is the correct procedure to migrate from old repo to the new one? Or how its done correctly, so I do not need to re-pair all my zigbee devices. Thanks.

kpcz, 2021-06-29:
> Thanks. indeed I am not on latest addon version. But when I add repo https://github.com/zigbee2mqtt/hassio-zigbee2mqtt I now see two Home Assistant Add-on: Zigbee2mqtt addons, one installed with [...]

To answer my own question, this is [how ](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/52#issuecomment-792873888) to change repo in Hassio

closed 2021-06-29 by kpcz