Koenkk/zigbee2mqtt #7895: Incorrect device type

Anonymenet77, 2021-06-28:
Hello,

Device type incorrectly identified TS0216 not a router.

Best regards,

[image]

Koenkk, 2021-06-28:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-28 by Koenkk

Koenkk, 2021-07-03:
@rindlerblabla that depends on the device itself, how z2m recognises it is only a cosmetic thing, it doesn't have any influence on the actual behaviour of the device.