Koenkk/zigbee2mqtt #7897: Help to Add device Vimar 14593

alesoft73, 2021-06-28:
Hi,
I'm trying to integrate this device.
I was able to integrate and manage the on / off status switch
however, the device should also have the powermetering reading status.

This is the link:
https://www.vimar.com/en/int/catalog/product/index/code/14593

I have use a "zigbee-herdsman-converters"

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
zigbeeModel: ['Mains_Power_Outlet_v1.0'],
model: '14593.0',
vendor: 'Vimar',
description: 'Connected actuator 16A',
extend: extend.switch(),
configure: async (device, coordinatorEndpoint, logger) => {
const endpoint = device.getEndpoint(10);
await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
},
},
];

Switch on/off work... but powermeter ??

this is a db:

{"id":17,"type":"Router","ieeeAddr":"0xf4ce3615c55c9503","nwkAddr":3980,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"Mains_Power_Outlet_v1. 0","epList":[10,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":9,"inlusterList":[0,3,6,5,4],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"Mains_Power_Outlet_v1.0","manufacturerName":"Vimar","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200501","swBuildId":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":33,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[{"cluster":33,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00158d00032013ce","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200501","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1681128303},"lastSeen":1624687590521}

This is a debug:

debug 2021-06-25 21:59:50: Received Zigbee message from '0xf4ce3615c55c9503', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 10 with groupID null

Many thanks.

alesoft73, 2021-08-05:
I have a problem do add a power consumation, many thanks for help

alesoft73, 2021-09-06:
Anyone have any ideas?
Thanks for your help.

Koenkk, 2021-09-06:
I'm not sure if it supports electrical measurements (at least not via the standard way) because the electrical measurement (2820) or metering cluster (1794) is not in the `inClusterList`.

You can try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [{
    zigbeeModel: ['Mains_Power_Outlet_v1.0'],
    model: '14593.0',
    vendor: 'Vimar',
    description: 'Connected actuator 16A',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);

        try {
            await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement']);
            await reporting.activePower(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint);
            logger.info('setup electrical measurement');
        } catch (e) {
            logger.error('eletrical measurement setup failed ' + e.message);
        }

        try {
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.instantaneousDemand(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            logger.info('setup metering');
        } catch (e) {
            logger.error('metering setup failed ' + e.message);
        }
    },
    exposes: [e.power(), e.current(), e.voltage(), e.switch()],
}];
```

If you could provide the logging after pressing the configure button via the frontend (yellow button) we can check if it supports it.

alesoft73, 2021-09-06:
Ok i test, many thanks!

closed 2021-12-06 by github-actions[bot]

alesoft73, 2021-12-14:
Do you have any idea @Koenkk :-)

Koenkk, 2021-12-14:
@alesoft73 since this device does not support power metering via the standard way, you can try to sniff traffic with the original gateway and try to reverse engineer how power measurements are send to the gateway.

massiant70, 2022-01-10:
So do you confirm there is no chance to get full support in Z2M for Vimar 14593 ???

alesoft73, 2022-01-11:
> @alesoft73 since this device does not support power metering via the standard way, you can try to sniff traffic with the original gateway and try to reverse engineer how power measurements are send to the gateway.

This thread might help ??

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3195

(delete message if is a problem ;))

alesoft73, 2022-01-23:
> @alesoft73 since this device does not support power metering via the standard way, you can try to sniff traffic with the original gateway and try to reverse engineer how power measurements are send to the gateway.

I can't sniff because i don't have the gateway... Vimar sells only bluetooth gateway :-(

Many thanks another for your attention.

Raycharges, 2022-01-31:
I'm interested in this topic aswell, sadly I'm a beginner and cannot support at all :(

Cheers

sebenik, 2022-04-21:
Would it be possible to only include partial support for this device so we could at least use on/off feature even without the power meter?