Koenkk/zigbee2mqtt #7901: TS0207 repeater detected as water sensor

jypsie, 2021-06-29:
I have a Zigbee repeater that is correctly discovered as a **TS0207_repeater**, working fine.
I ordered 2 more and they are discovered as **TS0207** being water leak detectors. This results in the following errors

2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.water_leak }}'
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.water_leak }})
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery_low }}'
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.battery_low }})
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.tamper }}'
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.tamper }})
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.battery }}'
2021-06-29 15:48:23 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.linkquality }}'

What can be done to discover this correctly? Can I change the model (being TS0207 now) to TS0207_repeater? How?
How can I get rid of the erros as posted above.

Koenkk, 2021-06-29:
Could you provide the data/database.db entries of these device?

jypsie, 2021-06-29:
can't find a good viewer, but these 2 lines are the 2 repeaters that show up as water sensors

{"id":16,"type":"Router","ieeeAddr":"0x5c0272fffef1978a","nwkAddr":56121,"manufId":4098,"manufName":"_TZ3000_5k5vh43t","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_5k5vh43t","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1624996575712}

and 

{"id":17,"type":"Router","ieeeAddr":"0x5c0272fffeeffad1","nwkAddr":53344,"manufId":4098,"manufName":"_TZ3000_5k5vh43t","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":22,"appVersion":64,"modelId":"TS0207","manufacturerName":"_TZ3000_5k5vh43t","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1624996461472}

jypsie, 2021-06-29:
This repeater is correctly identified:

{"id":6,"type":"Router","ieeeAddr":"0x086bd7fffeeab44e","nwkAddr":63269,"manufId":4098,"manufName":"_TZ3000_m0vaazab","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_m0vaazab","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1624996512155}

jypsie, 2021-06-29:
[image]
 
The first one is ok.....

jypsie, 2021-06-30:
furthermore 8 and 9 don't seem to work as repeater....

Koenkk, 2021-06-30:
Should be fixed now .

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jypsie, 2021-07-01:
Looks good, will confirm in a couple of hours!

Big thanks you for this great support!!!!

jypsie, 2021-07-02:
confirmed! Is working correctly now. Thanks again!

closed 2021-07-02 by Koenkk

DaPeace, 2024-10-08:
I have the same problem, Repeater TS0207 is recognized as TS0207_water_leak_detector.
Manufacture _TZ3000_wmlc9p9z
Adress 0x0cae5ffffe86e1bf
Network Adress 0x2582 / 9602

Is it possible to add these too? Thank you!

Koenkk, 2024-10-10:
Added `_TZ3000_wmlc9p9z`

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.