Koenkk/zigbee2mqtt #7902: Philips Hue Buckram White ambiance double (5047230P6) and quadruple (5047430P6)

SHxKM, 2021-06-29:
## Information about the device + link
5047230P6 - [Link](https://www.philips-hue.com/en-hk/p/hue-white-ambiance-buckram-double-spotlight/5047230P6)
5047430P6 - [Link](https://www.philips-hue.com/en-hk/p/hue-white-ambiance-buckram-quadruple-spotlight/5047430P6)

## More information

I've tried adding custom handlers for these devices (I've successfully added handlers for 2 other Philips devices that were also unsupported), but they still show up as unsupported when I restart Z2M (after adding the `js` file and modifying `configuration.yaml`). 

## example

Here's an example of `buckram_double.js`:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const hueExtend = {
  light_onoff_brightness: (options = {}) => ({
    ...extend.light_onoff_brightness(options),
    toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp(options),
    toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_color: (options = {}) => ({
    ...extend.light_onoff_brightness_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_color({ supportsHS: true, ...options }).toZigbee
      .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp_color: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options })
      .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
};

const definition = {
  zigbeeModel: ['5047230P6_01', '5047230P6_02'],
  model: '5047230P6',
  vendor: 'Philips',
  description: 'Hue White ambiance Buckram double spotlight',
  meta: { turnsOffAtBrightness1: true },
  extend: hueExtend.light_onoff_brightness_colortemp(),
  ota: ota.zigbeeOTA,
};

module.exports = definition;
```

I've tried following the lines of the buckram single, [which is already supported](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js#L1747-L1755).

## data/database.db entry of the device

The `5047230P6` has two entries (perhaps that's the problem?):

```
{
   "id":9,
   "type":"Router",
   "ieeeAddr":"0x0017880108XXXXXX",
   "nwkAddr":35797,
   "manufId":4107,
   "manufName":"Philips",
   "powerSource":"Mains (single phase)",
   "modelId":"5047230P6",
   "epList":[
      11,
      242
   ],
   "endpoints":{
      "11":{
         "profId":260,
         "epId":11,
         "devId":268,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            8,
            4096,
            64514,
            768
         ],
         "outClusterList":[
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"5047230P6",
                  "manufacturerName":"Philips",
                  "powerSource":1,
                  "zclVersion":2,
                  "appVersion":2,
                  "stackVersion":1,
                  "hwVersion":3,
                  "dateCode":"20200327",
                  "swBuildId":"1.65.11_hB798F2B"
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":2,
   "stackVersion":1,
   "hwVersion":3,
   "dateCode":"20200327",
   "swBuildId":"1.65.11_hB798F2B",
   "zclVersion":2,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1624980227086
}
```

2nd entry:

```
{
   "id":10,
   "type":"Router",
   "ieeeAddr":"0x0017880108XXXXXX",
   "nwkAddr":9650,
   "manufId":4107,
   "manufName":"Philips",
   "powerSource":"Mains (single phase)",
   "modelId":"5047230P6",
   "epList":[
      11,
      242
   ],
   "endpoints":{
      "11":{
         "profId":260,
         "epId":11,
         "devId":268,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            8,
            4096,
            64514,
            768
         ],
         "outClusterList":[
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"5047230P6",
                  "manufacturerName":"Philips",
                  "powerSource":1,
                  "zclVersion":2,
                  "appVersion":2,
                  "stackVersion":1,
                  "hwVersion":3,
                  "dateCode":"20200327",
                  "swBuildId":"1.65.11_hB798F2B"
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":2,
   "stackVersion":1,
   "hwVersion":3,
   "dateCode":"20200327",
   "swBuildId":"1.65.11_hB798F2B",
   "zclVersion":2,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1624980227547
}
```

Worth noting that the `ieeeAddr` of each is different, is that what's causing the trouble?

SHxKM, 2021-06-29:
This may or may not be off-topic, but I may have been doing some things wrong...

Let's take the already-supported 2-spots Adore:

```js
{
    zigbeeModel: ['3417931P6'],
    model: '3417931P6',
    vendor: 'Philips',
    description: 'Hue white ambiance Adore GU10 with Bluetooth (2 spots)',
    meta: {turnsOffAtBrightness1: true},
    extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}),
    ota: ota.zigbeeOTA,
}
```

I was pretty sure by some examples that in order to control each light separately, the `zigbeeModel` array had to be of the same length as the # of actual lights (in the above case, 2). This is how I successfully paired (?, not sure anymore) the unsupported [Centris 3-spot cross ceiling light](https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-centris-3-spot-cross-ceiling-light/5060831P7) (notice the suffixes of `01`, `02`, etc..):

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const hueExtend = {
  light_onoff_brightness: (options = {}) => ({
    ...extend.light_onoff_brightness(options),
    toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp(options),
    toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_color: (options = {}) => ({
    ...extend.light_onoff_brightness_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_color({ supportsHS: true, ...options }).toZigbee
      .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp_color: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options })
      .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
};

const definition = {
  zigbeeModel: ['5060830P7_01', '5060830P7_02', '5060830P7_03', '5060830P7_04'],
  model: '5060830P7',
  vendor: 'Philips',
  description: 'Hue White & Color ambience Centris ceiling light (3 spots)',
  meta: { turnsOffAtBrightness1: true },
  extend: hueExtend.light_onoff_brightness_colortemp_color(),
  ota: ota.zigbeeOTA,
};

module.exports = definition;
```

So now I'm unsure: am I doing things wrong?

SHxKM, 2021-07-02:
Maybe the each one of the Buckram’s spotlights is just supposed to be added as separate *device*? @Koenkk do you have any idea of a device that works this way?

Koenkk, 2021-07-02:
@SHxKM it looks that each light is just a separate Zigbee devices (since it appears as 2 in the database). You should use the `5047230P6` as the `zigbeeModel`.

SHxKM, 2021-07-02:
> @SHxKM it looks that each light is just a separate Zigbee devices (since it appears as 2 in the database). You should use the `5047230P6` as the `zigbeeModel`.

OK thanks. Will make a PR shortly after I get them working. I wonder though @Koenkk because the Pillar single is already mapped in Z2M, is there a declarative way to adjust its mapping to also include the double’s zigbeeModel? I guess it doesn’t really work that way.

Another question is: does that mean that I would’ve gotten the same mapping in the native Philips Hue app? (So the Buckram double would’ve appeared as two separate lights). Because for example with the Centris 3 Spotlight I successfully added it with 1 model and 4 lights (1 main + 3 spots). (Not 4 separate devices. Only 4 separate entities)

SHxKM, 2021-07-02:
> @SHxKM it looks that each light is just a separate Zigbee devices (since it appears as 2 in the database). You should use the `5047230P6` as the `zigbeeModel`.

@Koenkk Actually I'm a bit confused now. As I wrote earlier in the thread I have successfully added support for the Centris 3 spotlight, with this code:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const hueExtend = {
  light_onoff_brightness: (options = {}) => ({
    ...extend.light_onoff_brightness(options),
    toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp(options),
    toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_color: (options = {}) => ({
    ...extend.light_onoff_brightness_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_color({ supportsHS: true, ...options }).toZigbee
      .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp_color: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options })
      .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
};

const definition = {
  zigbeeModel: ['5060830P7_01', '5060830P7_02', '5060830P7_03', '5060830P7_04'],
  model: '5060830P7',
  vendor: 'Philips',
  description: 'Hue White & Color ambience Centris ceiling light (3 spots)',
  meta: { turnsOffAtBrightness1: true },
  extend: hueExtend.light_onoff_brightness_colortemp_color(),
  ota: ota.zigbeeOTA,
};

module.exports = definition;
```

As you can see, it's with the `_01`, `_02` convention.

But the 4 lights for this device **also** appear as four separate entries in the DB, here's two of them:

```
{
   "id":14,
   "type":"Router",
   "ieeeAddr":"0x0017880109XXXXXX",
   "nwkAddr":40533,
   "manufId":4107,
   "manufName":"Philips",
   "powerSource":"Mains (single phase)",
   "modelId":"5060830P7_03",
   "epList":[
      11,
      242
   ],
   "endpoints":{
      "11":{
         "profId":260,
         "epId":11,
[... 154 lines cut ...]
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 2,
  "stackVersion": 1,
  "hwVersion": 3,
  "dateCode": "20200327",
  "swBuildId": "1.65.11_hB798F2B",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {
    "configured": 2
  },
  "lastSeen": 1625222218072
}
```

So is there any kind of convention here? Because this contradicts what is written here (emphasis mine):

> it looks that each light is just a separate Zigbee devices (**since it appears as 2 in the database**). You should use the 5047230P6 as the zigbeeModel.

The above worked for the Centris, but its light also appear as 4 separate "devices" in Zigbee2MQTT; is that not what's supposed to happen? and if that's OK, then why doesn't the same convention work for the Buckram double/quadruple? they also have 2/4 entries in the DB.

Koenkk, 2021-07-03:
> then why doesn't the same convention work for the Buckram double/quadruple? they also have 2/4 entries in the DB.

This depends on how Philips implemented this device, this is not something that comes from z2m.

SHxKM, 2021-07-03:
> > then why doesn't the same convention work for the Buckram double/quadruple? they also have 2/4 entries in the DB.
> 
> This depends on how Philips implemented this device, this is not something that comes from z2m.

@Koenkk Alright, thanks. So just so I understand for future reference and better understanding: The only real way to know would be trial and error? there is no definitive way to know how a device should be mapped given X,Y,Z properties?

Koenkk, 2021-07-04:
Yes exactly, it's just a matter how the device identifies itself.

closed 2021-08-11 by github-actions[bot]