Koenkk/zigbee2mqtt #7903: Hive SLR2b thermostat discovery issues, no status in HA

scott-ish, 2021-06-29:
Hi,

## What happened
Trying to pair and use a Hive SLR2b in HA using the official zigbee2mqtt plugin.

## What did you expect to happen
The pairing was successful and discovery added it to the HA entities. I added a thermostat card and linked to the heating and water new entities. I expected to see the state of the devices on the cards (temp, mode etc) nothing is shown. The temp control does work and sends to zigbee2mqtt and to the device.
[image]

## How to reproduce it (minimal and precise)
Pair a SLR2b, use HA discovery, add thermostat card.

## Debug info
Zigbee2MQTT version: offical HA add-on 1.19.1-1
Adapter hardware: CC2652R (zzh)
Adapter firmware version: zStack3x0 20210120

Digging in a bit, I can see in MQTT a 'climate' topic with sub topics for 'climate_heat' and 'climate_water' and further sub config topics. Those make a few references to the topics 'zigbee2mqtt/Heating/heat' and 'zigbee2mqtt/Heating/water' respectively for states. These don't exist in MQTT, only a top level 'zigbee2mqtt/Heating'. This does contain all the relevant data, so something seems off. I assume the discovery sets that config with those topics? Zigbee2mqtt only publishes state info to the main topic not the sub /heat or /water topics, is that expected?
Clearly if I publish the HA config topics pointing at the 'zigbee2mqtt/Heating' topic instead of the sub topics things work for a bit, but I guess the config is then overwritten by zigbee2mqtt at somepoint again as it eventually reverts.

New to HA and zigbee2mqtt, am I missing something on how to set this up?

Cheers,
Scott

peterguy04, 2021-07-02:
i see the same problem with dockered z2m and conbee2

peterguy04, 2021-08-08:
(reviving thread)
heres the states HA, notice NULL and unknown 
[image]

yet the info is all there just not available for individual endpoints as observed by OP
[image]

robertalexa, 2021-08-08:
@peterguy04 Can you go to Configuration -> Devices -> SLR -> MQTT Info.

From that list find the one that is climate.slr. Expand the payload and send a screenshot please
You are looking for sth that looks like this
[image]

peterguy04, 2021-08-08:
heres payload for Heat 
[image]

and heres water 
[image]

robertalexa, 2021-08-08:
Sorry have been AFK.

Let's use temperature (occupied_heating_setpoint_water) as our example.

[image]

So both command topic and state topic are `zigbee2mqtt/SLR/water` which is the correct one.

@peterguy04 also mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/8184#issuecomment-894800176 that 
>there is no response when listening to the individual endpoints as you asked,
>the only response i get is the one above from zigbee2mqtt/SLR

I know for sure that @Koenkk added the water endpoint, I saw it in an issue somewhere but can't find it anymore. But it feels like the mqtt topic used to publish does not follow that principle.

@Koenkk would you be kind to assist? Or maybe even @sjorge can point me in the right direction? I do not own a dual channel receiver as I am blind here, just trying to assist with the issue given that I was looking into the Hive receivers and thermostats.

sjorge, 2021-08-08:
@robertalexa I'm guessing there is no `.withEndpoint('water')` and `.withEndpoint('heating')` in the expose so it can differentiate between both endpoints?

robertalexa, 2021-08-08:
@sjorge There is. It was there before my PR, and @peterguy04 is using external converter from my PR.

OP is using SLR2b and @peterguy04 is using SLR2. Both are dual channel, just aesthetically different.

[image]

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2866/files

robertalexa, 2021-08-08:
I know @Koenkk merged this https://github.com/Koenkk/zigbee2mqtt/issues/7470 earlier this year, which is somewhat related, but it doesn't look like it creates the current issue. Even with _water at the end, the topic still points to /water which seems to be not populated

sjorge, 2021-08-08:
Does it work if you only expose one of them?

robertalexa, 2021-08-08:
@peterguy04 Can you try Jorge's suggestion? I personally do not think that will make a difference given the topics used in the discovery and the fact that z2m is not pushing information into that topic.

@sjorge On the single channel receiver where you do not need to specify the endpoint (SLR1 and SLR1b) this works fine.

sjorge, 2021-08-08:
Yeah I guess the ha extension could be ignoring the endpoint info for climate too, or if it is not 2 entries might confuse the HA side of things. We should be able to rule out the later.

peterguy04, 2021-08-08:
>@peterguy04 Can you try Jorge's suggestion? I personally do not think that will make a difference given the topics used in the discovery and the fact that z2m is not pushing information into that topic.

sure i can test which ever, but how do i go about exposing only one?

so just for me to understand whats happening, 
HA is publishing the commands to z2m on the correct endpoints via `zigbee2mqtt/SLR/water` which is fine, but as z2m is only publishing the states back to `zigbee2mqtt/SLR` and HA is looking in `zigbee2mqtt/SLR/water` this is where the issue is? or am i way off the mark here haha

robertalexa, 2021-08-08:
```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SLR2A'],
    model: 'SLR2A',
    vendor: 'Hive',
    description: 'Dual channel heating and hot water thermostat',
    fromZigbee: [fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation, tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule, tz.thermostat_temperature_setpoint_hold, tz.thermostat_temperature_setpoint_hold_duration],
    endpoint: (device) => {
        return {'heat': 5, 'water': 6};
    },
    meta: {disableDefaultResponse: true, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const heatEndpoint = device.getEndpoint(5);
        const waterEndpoint = device.getEndpoint(6);
        const binds = [
            'genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat',
        ];
        await reporting.bind(heatEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 5});
        await reporting.thermostatRunningState(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatSystemMode(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatOccupiedHeatingSetpoint(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 5});
        await reporting.thermostatTemperatureSetpointHold(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatTemperatureSetpointHoldDuration(heatEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.bind(waterEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatRunningState(waterEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatSystemMode(waterEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatOccupiedHeatingSetpoint(waterEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 5});
        await reporting.thermostatTemperatureSetpointHold(waterEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatTemperatureSetpointHoldDuration(waterEndpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
    },
    exposes: [
        exposes.climate().withSetpoint('occupied_heating_setpoint', 5, 32, 0.5).withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat']).withEndpoint('heat'),
        exposes.numeric('temperature_setpoint_hold_duration', ea.ALL).withValueMin(0).withValueMax(65535)
            .withDescription('Period in minutes for which the setpoint hold will be active. 65535 = attribute not' +
                ' used. 0 to 360 to match the remote display').withEndpoint('heat'),
        exposes.numeric('temperature_setpoint_hold_duration', ea.ALL).withValueMin(0).withValueMax(65535)
            .withDescription('Period in minutes for which the setpoint hold will be active. 65535 = attribute not' +
                ' used. 0 to 360 to match the remote display').withEndpoint('heat')],
};

module.exports = definition;

```
Try this converter. This will only create exposes (check Frontend to confirm) for heat only.

Test the heat climate in HA and report back.

> so just for me to understand whats happening,
> HA is publishing the commands to z2m on the correct endpoints via `zigbee2mqtt/SLR/water` which is fine, but as z2m is only publishing the states back [...]

At a fist glance yes. z2m on the discovery sets the topics to /water, so that is where HA is looking, but it isn't pushing information there.

The alternative is to keep the data where it is and change the topics AND attributes.

But test the above for Jorge and then we go from there.

@sjorge Can you point me towards where the topics gets "set" as part of the discovery? Thanks

peterguy04, 2021-08-08:
ok, so same issue with only single EP exposed

robertalexa, 2021-08-08:
And MQTT Info is exactly the same as it was when you had both endpoints exposed right?

peterguy04, 2021-08-08:
[image]

Koenkk, 2021-08-09:
@peterguy04 since the temperature is not being displayed here, can you verify that zigbee2mqtt publishes a JSON object to `zigbee2mqtt/SLR/heat` with the property `occupied_heating_setpoint_heat`?

robertalexa, 2021-08-09:
@Koenkk No information is published at all on /heat or /water. The only info published by z2m is in `zigbee2mqtt/SLR`

[image]

Koenkk, 2021-08-09:
Does it work after removing this line: https://github.com/Koenkk/zigbee2mqtt/blob/70891eec39018303a43349c875ad04d44b3b43cf/lib/extension/homeassistant.js#L245 ?

peterguy04, 2021-08-09:
@Koenkk yep that did it 
[image]

Koenkk, 2021-08-09:
@peterguy04 does everything work as expected now? (also controlling)

peterguy04, 2021-08-09:
@Koenkk yes the modes are functional and selecting in HA is reported in z2m also now .

only issue we have with it is when we request heat from it its not firing up because its not setting  `temperature_setpoint_hold` to true so when we select temp its jumping back to 1c, but that maybe unrelated and specific to our hive climate devices? but when i pop into z2m and set that true the temp from HA works as expected

robertalexa, 2021-08-09:
@peterguy04 If you request HEAT manually you need to set the setpoint hold to false first. It is the receiver's internal logic that tries to take control. I have noticed that in my testing to. If you want them to work in 1 swift movement you have to do fast consecutive mqtt commands (e.g. multiple params in one call)

robertalexa, 2021-08-09:
Actually no, that is not right. To set heat you have to:
1. choose system_mode heat
2. set hold to true
3. set temperature.

Are you saying you have followed these steps in HA and you are seeing problems?

peterguy04, 2021-08-09:
well in HA you cant do step 2 from that climate entity card  but i remember now you exposed a HA switch for hold . so can achieve it that way . for me its fine as i am happy with mqtt commands, just thinking for other users that arnt so comfortable with commands

robertalexa, 2021-08-10:
> @Koenkk yes the modes are functional and selecting in HA is reported in z2m also now .

Hey  @Koenkk , do you need me to make a PR to remove that if statement? Or is this something you plan on looking at? Thanks
[image]

Koenkk, 2021-08-12:
@robertalexa removed it now. Since the Hive SLR2b/SLR2 is the only thermostat having multiple endpoints I don't expect any side effects. Assuming this can be closed now.

closed 2021-08-12 by Koenkk