Koenkk/zigbee2mqtt #7906: Add new manufacture of TS0505B

hungpr0, 2021-06-30:
## Information about the device + link
Zigbee 3.0 downlight 10W/12W RGBCW with blue ear
https://www.aliexpress.com/item/1005002519334985.html?spm=a2g0o.productlist.0.0.6b907a4cE368J2&algo_pvid=2cdc3e12-c999-4235-968e-34408fb7e460&algo_exp_id=2cdc3e12-c999-4235-968e-34408fb7e460-11
https://www.aliexpress.com/item/1005002510329829.html?spm=a2g0o.productlist.0.0.6b907a4cE368J2&algo_pvid=2cdc3e12-c999-4235-968e-34408fb7e460&algo_exp_id=2cdc3e12-c999-4235-968e-34408fb7e460-2

## data/database.db entry of the device
`{"id":11,"type":"Router","ieeeAddr":"0xa4c138e05f0cb940","nwkAddr":9301,"manufId":4417,"manufName":"_TZ3210_5snkkrxw","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_5snkkrxw","powerSource":1,"zclVersion":3,"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1625029537471}r`

Koenkk, 2021-06-30:
Done, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-30 by Koenkk

ravaz, 2022-01-29:
Hello,

I got a similar problem. I bought the following lamp:

[https://www.aliexpress.com/item/1005003717039854.html?spm=a2g0o.9042311.0.0.18414c4dI5p8Lf](url)

The manufacturer is reported as "**_TZ3210_leyz4rju**", would it be possible to add it as well?