Koenkk/zigbee2mqtt #7907: zigbee2mqtt fails to start with for no specific reason

drivingmx, 2021-06-30:
## What happened
I am running zigbee2mqtt in docker.
Today i recognized zigbee2mqtt isnt working anymore beecause the container isnt able to start anymore.
As far as i can tell the last update was almost 3 weeks ago and it was running till now.
I updated the FW of my zzh coordinator (CC2652R - FW "CC2652R_coordinator_20210120").This didnt fix the problem.
Also I was trying some restarts -  didnt help.

I am a little confused because from my point of view there is no change in the system that could be the reasn for zigbee2mqtt to stop working.

Log from container:
`Using '/app/data' as data directory

Zigbee2MQTT:info  2021-06-30 09:19:30: Logging to console and directory: '/app/data/log/2021-06-30.09-19-28' filename: log.txt

Zigbee2MQTT:info  2021-06-30 09:19:30: Starting Zigbee2MQTT version 1.19.1 (commit #9bd4693)

Zigbee2MQTT:info  2021-06-30 09:19:30: Starting zigbee-herdsman (0.13.111)

Zigbee2MQTT:error 2021-06-30 09:19:32: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2021-06-30 09:19:32: Failed to start zigbee

Zigbee2MQTT:error 2021-06-30 09:19:32: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions

Zigbee2MQTT:error 2021-06-30 09:19:32: Exiting...

Zigbee2MQTT:error 2021-06-30 09:19:32: SyntaxError: Unexpected end of JSON input

    at JSON.parse (<anonymous>)

    at AdapterBackup.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/adapter-backup.js:67:31)

    at Generator.next (<anonymous>)

    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/adapter-backup.js:24:58)`

## What did you expect to happen

I am expecting the container to start as usual.

## How to reproduce it (minimal and precise)

Every restartof the container.

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

sjorge, 2021-06-30:
I think the adaptor backup is complete so it can't be parsed as json. There should probably be a nicer error for that I guess.

drivingmx, 2021-06-30:
What kind  of solution do you suggest? Should I delete the corrupt backup file?

sjorge, 2021-06-30:
That would probably make z2m start, but not sure if the coordinator still knows about the network.

~ sjorge

drivingmx, 2021-07-01:
I tried that. 
After deleting the file coordinator_backup.json z2m mentioned, that my panID is already in use.
After changing the panID in my configuration.yaml z2m startet up.
I had to repair all of my devices.

Not the best solution but a solution.

closed 2021-07-01 by drivingmx