Koenkk/zigbee2mqtt #7908: ajouter un appareil TRADFRIbulbE27WWclear250lm -Ampoule LED IKEA

VASNIER, 2021-06-30:
Hello
Je souhaiterai pouvoir intégrer un nouvelle ampoule LED IKEA (TRADFRIbulbE27WWclear250lm) en Zigbee2MQTT dans Gladys.

Merci pour votre aide

{
  "name": "0x680ae2fffe4c9018",
  "external_id": "zigbee2mqtt:0x680ae2fffe4c9018",
  "model": "TRADFRIbulbE27WWclear250lm",
  "features": [],
  "should_poll": false,
  "service_id": "8f5634d8-e97e-43f8-806f-bd8ed12a4811",
  "supported": false
}

toxictaste1981, 2021-06-30:
opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/ikea.js

add this

{
        zigbeeModel: ['TRADFRI bulb E27 WW clear 250lm', 'TRADFRI bulb E26 WW clear 250lm'],
        model: '1934G3',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 WW clear 250 lumen, dimmable',
        extend: tradfriExtend.light_onoff_brightness(),
    },

after this
{
        zigbeeModel: ['TRADFRI bulb E27 WW clear 250lm', 'TRADFRI bulb E26 WW clear 250lm'],
        model: 'LED1842G3',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 WW clear 250 lumen, dimmable',
        extend: tradfriExtend.light_onoff_brightness(),
    },

toxictaste1981, 2021-06-30:
change in file /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices
`
    {
        zigbeeModel: ['TRADFRI bulb E27 WW clear 250lm', 'TRADFRI bulb E26 WW clear 250lm'],
        model: 'LED1842G3',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 WW clear 250 lumen, dimmable',
        extend: tradfriExtend.light_onoff_brightness(),
    },
`
to
`
	{
        zigbeeModel: ['TRADFRI bulb E27 WW clear 250lm', 'TRADFRI bulb E26 WW clear 250lm','TRADFRIbulbE27WWclear250lm'],
        model: 'LED1842G3/1934G3',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 WW clear 250 lumen, dimmable',
        extend: tradfriExtend.light_onoff_brightness(),
    },
`
after that it run for me

VASNIER, 2021-07-01:
Hello
Je n'ai pas de fichier "opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/ikea.js"
Où devrait-il se trouver?

closed 2021-08-08 by github-actions[bot]