Koenkk/zigbee2mqtt #7915: NAS-PD07 PIR, temperature & humidity sensor

mrTuomoK, 2021-07-01:
## Information about the device + link

Motion sensor with at least temperature sensor (might have humidity too)
[https://www.aliexpress.com/item/1005002339649035.html](Aliexpress)
..

## data/database.db entry of the device

`
{"id":11,"type":"EndDevice","ieeeAddr":"0x842e14fffee35609","nwkAddr":10890,"manufId":4098,"manufName":"_TZE200_7hfcudw5","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_7hfcudw5","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4134","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1624352191152}
`

Anyone with the same device and not able to get it working?

closed 2021-08-08 by github-actions[bot]

Leodjmao, 2021-09-01:
I have the same motion sensor and work to write converter.

mrTuomoK, 2021-09-01:
> I have the same motion sensor and work to write converter.

Nice, I haven't had time at all on this, keep us posted on your progress 👍

Leodjmao, 2021-09-01:
I'm à new user but as soon as i get something, i make a pull request to z2m.

Leodjmao, 2021-09-02:
I find that for this moment.

[image]