Koenkk/zigbee2mqtt #7918: Groups are not identified unique across multiple instances

nicholsongreg, 2021-07-01:
I'm running two different Zigbee networks in my house, each hosted with the latest zigbee2mqtt stable.

On the upgrade to latest (1.20.0), I've noticed that the second upgraded system overwrote the Home assistant group from the first system, even though they are reporting on different mqtt queues.

Example:  System 1, group 1 is Kitchen Table.  This works in Home Assistant as expected.  Upgrade System 2 to 1.20.0, now, Kitchen Table is Bonus Room (but the entity reported is the Kitchen Table Entity).  System 2, Group 1 is... Bonus Room.

The groups should be differentiated in the Home Assistant autodiscovery with a unique identifier so that the group numbers do not collide.

I'm currently renumbering the groups on one system to prevent the collision.

Koenkk, 2021-07-02:
This happens because the used MQTT discovery topics collide, I've added the hashed version of the `mqtt.base_topic` to this topic now (these should be different between different z2m instances). 

Please confirm if this issue is fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

nicholsongreg, 2021-07-03:
Something is still not quite right.  

Latest-dev, updated 11:00am CST.

Not seeing the new topics in mqtt explorer.  Logs show :

> Zigbee2MQTT:info  2021-07-03 11:03:40: MQTT publish: topic 'homeassistant/light/12210510398101101501091131161164550_1/light/config', payload 'null'
zigbee2mqtt    | Zigbee2MQTT:info  2021-07-03 11:03:40: MQTT publish: topic 'homeassistant/light/12210510398101101501091131161164550_50/light/config', payload 'null'
zigbee2mqtt    | Zigbee2MQTT:info  2021-07-03 11:03:40: MQTT publish: topic 'homeassistant/light/12210510398101101501091131161164550_51/light/config', payload 'null'
zigbee2mqtt    | Zigbee2MQTT:info  2021-07-03 11:03:40: MQTT publish: topic 'homeassistant/light/12210510398101101501091131161164550_52/light/config', payload 'null'
zigbee2mqtt    | Zigbee2MQTT:info  2021-07-03 11:03:40: MQTT publish: topic 'homeassistant/light/12210510398101101501091131161164550_53/light/config', payload 'null'

I did renumber, but created a test group with group number 1 to test the collision.  It doesn't show up, period.  In theory, I think your idea is correct - I've got two different topics, and get a different hash, but both show payload 'null'.

Koenkk, 2021-07-04:
Could you provide the z2m debug log from starting till seeing these messages?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

nicholsongreg, 2021-07-06:
https://gist.github.com/nicholsongreg/e8701bee249ea9704cc140db04bb1f32

Koenkk, 2021-07-06:
Thanks, fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

nicholsongreg, 2021-07-06:
So, Better.

I'm seeing the separate queues in MQTT now.

homeassistant/light/1221051039810110150109113116116459911111098101101_1/light/config

`{"availability":[{"topic":"zigbee2mqtt-conbee/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt-conbee/Dining Table/set","device":{"identifiers":["zigbee2mqtt_1"],"name":"Dining Table","sw_version":"Zigbee2MQTT 1.20.0-dev"},"json_attributes_topic":"zigbee2mqtt-conbee/Dining Table","name":"Dining Table","schema":"json","state_topic":"zigbee2mqtt-conbee/Dining Table","supported_color_modes":["color_temp"],"unique_id":"1_light_zigbee2mqtt-conbee"}`

homeassistant/light/12210510398101101501091131161164550_1/light/config has a payload of :

`{"availability":[{"topic":"zigbee2mqtt-2/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt-2/test_group/set","device":{"identifiers":["zigbee2mqtt_1"],"name":"test_group","sw_version":"Zigbee2MQTT 1.20.0-dev"},"json_attributes_topic":"zigbee2mqtt-2/test_group","name":"test_group","schema":"json","state_topic":"zigbee2mqtt-2/test_group","supported_color_modes":["xy","color_temp"],"unique_id":"1_light_zigbee2mqtt-2"}`

When I look in HA, I do get both devices, however, they are still grouped together, instead of as their own device.  So, instead of Dining table as a device with Dining Table as an entity, I also have test_group as an entity inside of Dining Table.

I think it's from the "device":{"identifiers":["zigbee2mqtt_1"] being the same.  I realize this is getting more finicky, but it looks like that needs to be uniqued as well.

And to be clear, THIS is working, it's just confusing to find your entities when they have been grouped different.

Koenkk, 2021-07-07:
Thanks for the good description, fixed this issue now (encoded topic will also be included in the identifier now)

nicholsongreg, 2021-07-07:
I think that fixed it.  Thank you!

closed 2021-07-07 by nicholsongreg