Koenkk/zigbee2mqtt #7920: Can't calibrate Moes Curtain Switch Module (identified as TUYA TS130F)

marcospg75, 2021-07-02:
Hello,
I have 7 of this modules (Moes Curtain Switch) 

https://es.aliexpress.com/item/1005002352672340.html?spm=a2g0s.9042311.0.0.4f1e63c0H7AFg7

Pairing process is doing well, but it's identified as TUYA TS120F
https://www.zigbee2mqtt.io/devices/TS130F.html

The issue is calibrating the device... I allways get this error:

Publish 'set' 'calibration' to 'toldo_del_balcon' failed: 'Error: Write 0x60a423fffeff2b8e/1 closuresWindowCovering({"tuyaCalibration":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

This is de definition of the device:
{
  "backlight_mode": null,
  "calibration": null,
  "linkquality": 108,
  "motor_reversal": "OFF",
  "moving": "STOP",
  "position": 90,
  "position": 100,
  "state": null
}

Any idea on how to solve this? (I have a lot of devices running well, but I'm newbie on debuging this issues :(  )

Thanks in advance!!!!

Akkalabeth, 2021-07-08:
Same issue here. 
Hoping for a solution soon, otherwise I'm thinking, if I would get a 2 gang switch, I can just workaround the whole calibration issue, right? Connect L1 to the "Open" wire, L2 to the "Close", then using "hold"/"release" actions on a remote to trigger the open/stop/close. Although I guess I have to make sure the both switches can never be "On" (so power to both the "Open" and "Close", I guess the motor wouldn't like that)

Thanks!

SkyByte, 2021-07-09:
Same issue here! (see also #7128)

The 'MOES Zigbee + RF Curtain Switch Module' cannot being calibrated in Z2M. For this switch (the version including 433MHz RF!) Z2M sends an 'UNSUPPORTED_ATTRIBUTE'. That's too bad, because without calibration this switch is absolutely useless, unless your curtain opens/closes in only 10 sec. (=preset).  ;-)

Hoping this issue will be fixed soon. Thank you!

`
Error:
Publish 'set' 'calibration' to '0x60a423fffeff2bc7' failed: 'Error: Write 0x60a423fffeff2bc7/1 closuresWindowCovering({"tuyaCalibration":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')' Info: MQTT publish: topic 'tele/zigbee/bridge/log', payload '{"message":"Publish 'set' 'calibration' to '0x60a423fffeff2bc7' failed: 'Error: Write 0x60a423fffeff2bc7/1 closuresWindowCovering({\"tuyaCalibration\":0}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0x60a423fffeff2bc7"},"type":"zigbee_publish_error"}'
`

vampywiz17, 2021-07-10:
> Same issue here.
> Hoping for a solution soon, otherwise I'm thinking, if I would get a 2 gang switch, I can just workaround the whole calibration issue, right? Connect L1 to the "Open" wire, L2 to [...]

The two device basically same, the only difference, that the "curtain module" is able to meassure the electricity, it is the method that is sense the two endpoint of the curtain. (and possible to make a percent based control)

If you replace it to normal two gang switch, you able to open and close the curtain, but can not do a percent control.

SkyByte, 2021-07-10:
> The two device basically same, the only difference, that the "curtain module" is able to meassure the electricity, it is the method that is sense the two endpoint of the curtain. (and possible to make a percent based control)
> 

@vampywiz17 
I am almost sure that not the curtain switch module does meassure the electricity for the endpoints, but the motor itself. 
The module is stupid and switches on the OPEN/CLOSE connection for a determinated time, and the motor has to stop itself at the endpoints. The problem here is the preset time of 10 seconds (=100%). If you set the curtain from 0% to e.g. 60%, the 'CLOSE'-relay of the module will switch on for about 6 seconds. Just simple.

In calibration mode, you just set the amount of time to open/close the curtain.It would be redundant to include a current measure in the module. If the preset switch time of the module would be e.g. 30 or more seconds, all would be fine (except the percent control). The motor stops by itself at the endpoints.

A common 2 gang switch module is on principle the same, but has a different programming and poses the danger of switching on both relays at same time. That could destroy the curtain motor. If your programming / control management ensures that this will not happen, you abolutely can use such a module. You are even more flexible.

vampywiz17, 2021-07-10:
@SkyByte The true is that i use Shelly 2.5 to move my curtain. In Shelly, the calibration is use the measuring. I set curtain the two  endpoint (it full manually, it need a special skrewdriver to set it) and after shelly measure the electricity and counts the time. If reach the 0W (the curtain reach the endpoint) it finish the counting. See the detailed article: https://www.facebook.com/notes/shelly-support-group-english-version/shelly-225-roller-shutter-calibration-tips/2064850700280950/
(How does calibration work?)

I don't know that how work the tuya zigbee module (i never use it) but if the calibration is automatically with tuya app, it need the measuring ability, because it dont know, where is the endpoint. (and try to move the curtain forever..)

SkyByte, 2021-07-13:
I just get the Moes **WiFi** + RF curtain module (not the Zigbee-, but the WiFi version). The preset time span of this module for activating the relays to open/close the curtain is exactly 60 seconds.
Don't know the intention to preset the default time span of the zigbee version to only 10 seconds. In most cases this will be too short!

SkyByte, 2021-07-17:
Today, my wireless Tuya Zigbee Gateway arrived. I added the Curtain Switch Module to the Smart Life app and was able to enter the amount of seconds for switching length to an individual value. In my case i successfully changed it to 60 seconds.
After that, i did not disconnect the module from Tuya, but i joined the module to Z2M pressing the joining button for about 10 seconds as usual. The module is then automatically shown as 'offline' in the Tuya Smart Life app.

And what a surprise!! The setting has **not** been lost. Sure, the calibration command still doesn't work in Z2M, but the time period is still at 60 seconds, even if you delete this device in Tuya Smart Life app later.

So i would recommend to buy or borrow a Tuya Zigbee Gateway to set the amount of seconds once to a sufficient value, until this issue will be fixed in Z2M.

aelias-eu, 2021-08-11:
Duplicate of #7128 
Should be resolved

closed 2021-08-11 by Koenkk

BubbaCZ, 2022-10-11:
> > Today, my wireless Tuya Zigbee Gateway arrived. I added the Curtain Switch Module to the Smart Life app and was able to enter the amount of seconds for switching length to an individual value. In my case i successfully changed it to 60 seconds. After that, i did not delete the module from Tuya, but i joined the module to Z2M pressing the joining button for about 10 seconds as usual. The module is then automatically shown as 'offline' in the Tuya Smart Life app.
> > And what a surprise!! The setting has **not** been lost. Sure, the calibration command still doesn't work in Z2M, but the time period is still at 60 seconds, even if you later delete this device in Tuya Smart Life app or power off.
> > So i would recommend to buy or borrow a Tuya Zigbee Gateway to set the amount of seconds once to a sufficient value, until this issue will be fixed in Z2M.
> 
> Hi, I'm having the same issue and also decided to by a Tuya Zigbee Gateway. I just add the devices and they work. The problem is that I don't find where I can change the Time duration parameter. Can you please Help me? Thanks in advance. Mario

Can you specify exact model of the wireless Zigbee gateway that will allow you to extend the running time, I have normal Tuya Hub already and when adding curtains to Smart life, there is not such option available.