Koenkk/zigbee2mqtt #7921: SmartThings multipurpose sensor IM6001-MPP01 is reporting every few seconds

Fabiancrg, 2021-07-02:
## What happened
Sensor is sending data all the time, does not look like it's going to sleep so the battery is empty after a month.
## What did you expect to happen
Sensor only send data when an event occurs and at regular interval (every 5, 10 or 30 minutes) for battery and temperature.

## How to reproduce it (minimal and precise)
Just add the sensor, you will see a lot of messages in the log

## Debug info
Zigbee2MQTT version: 1.19.1-dev commit: 2afd9ab6
Adapter hardware: CC26X2R1
Adapter firmware version: 20200805

asjmcguire, 2021-07-08:
I have this problem too, and I am unable to change any of the reporting times. With v1.20.0 any attempt to change the reporting intervals or even to disable them completely is met with an error message about a timeout.

```
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0x286d970001134176/1 manuSpecificSamsungAccelerometer from '0x00212effff0741a6/1' failed (waiting for response TIMEOUT)'
```

Koenkk, 2021-07-08:
@asjmcguire make sure to wakeup the device right before sending this command by clicking a button on it.

Fabiancrg, 2021-07-22:
@asjmcguire did you succeed to configure your device(s) ?

Fabiancrg, 2021-07-22:
@Koenkk I receive the following message for example when I am in debug mode:

``
juil. 22 15:00:32 NUC npm[300931]: Zigbee2MQTT:debug 2021-07-22 15:00:32: Received Zigbee message from 'IM6001', type 'attributeReport', cluster 'manuSpecificSamsungAccelerometer', data '{"x_axis":2,"y_axis":-45,"z_axis":-1028}' from endpoint 1 with groupID 0
``

But when I try to change the reporting interval values by publishing 

{"id":"0x286d97000106d50f/1","cluster":"manuSpecificSamsungAccelerometer","attribute":"acceleration","minimum_report_interval":300,"maximum_report_interval":3600,"reportable_change":5}

to zigbee2mqtt/bridge/request/device/configure_reporting, I get this error:

``
juil. 22 15:00:43 NUC npm[300931]: Zigbee2MQTT:debug 2021-07-22 15:00:43: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure_reporting' with data '{"id":"0x286d97000106d50f/1","cluster":"manuSpecificSamsungAccelerometer","attribute":"acceleration","minimum_report_interval":300,"maximum_report_interval":3600,"reportable_change":5}'
juil. 22 15:00:44 NUC npm[300931]: Zigbee2MQTT:error 2021-07-22 15:00:44: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x286d97000106d50f/1 manuSpecificSamsungAccelerometer([{"attribute":"acceleration","minimumReportInterval":300,"maximumReportInterval":3600,"reportableChange":5}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4362,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
juil. 22 15:00:44 NUC npm[300931]: Zigbee2MQTT:info  2021-07-22 15:00:44: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0x286d97000106d50f/1 manuSpecificSamsungAccelerometer([{\"attribute\":\"acceleration\",\"minimumReportInterval\":300,\"maximumReportInterval\":3600,\"reportableChange\":5}], {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4362,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')","status":"error"}'
``

Is there omething wrong in the command I am using ?

asjmcguire, 2021-07-22:
@Fabiancrg nope, the only value I am successfully able to change is the Temperature Reporting time. Anything to do with "manuSpecificSamsungAccelerometer" comes back with an error message - either a time out, or unsupported attribute.

Fabiancrg, 2021-07-22:
Indeed, I did the same with the temperature and it worked:

Published this:
``
{"id":"0x286d97000106d50f/1","cluster":"msTemperatureMeasurement","attribute":"measuredValue","minimum_report_interval":300,"maximum_report_interval":3600,"reportable_change":5}
``

and got this:

``
juil. 22 15:40:11 NUC npm[300931]: Zigbee2MQTT:debug 2021-07-22 15:40:11: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure_reporting' with data '{"id":"0x286d97000106d50f/1","cluster":"msTemperatureMeasurement","attribute":"measuredValue","minimum_report_interval":300,"maximum_report_interval":3600,"reportable_change":5}'
juil. 22 15:40:15 NUC npm[300931]: Zigbee2MQTT:info  2021-07-22 15:40:15: Configured reporting for '0x286d97000106d50f/1', 'msTemperatureMeasurement.measuredValue'
juil. 22 15:40:15 NUC npm[300931]: Zigbee2MQTT:info  2021-07-22 15:40:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"measuredValue","cluster":"msTemperatureMeasurement","id":"0x286d97000106d50f/1","maximum_report_interval":3600,"minimum_report_interval":300,"reportable_change":5},"status":"ok"}'
``

Koenkk, 2021-07-22:
I've increased the reporting interval now. Try updating to the latest-dev, wakeup the device and right after that click the configure button in the frontend (yellow refresh icon).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Fabiancrg, 2021-07-23:
@Koenkk it's working now with this new dev release.
Thank you for your help !

closed 2021-07-23 by Fabiancrg

hellcry37, 2021-08-07:
this is most definitive not fixed
```txt
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:35:13.656Z","linkquality":57,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:35:16.757Z","linkquality":57,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Test Vibration Sensor', payload '{"action":null,"angle":50,"angle_x":0,"angle_x_absolute":90,"angle_y":0,"angle_y_absolute":90,"angle_z":90,"battery":100,"last_seen":"2021-08-07T15:35:44.281Z","linkquality":63,"sensitivity":"medium","strength":9,"voltage":3065}'
info MQTT publish: topic 'zigbee2mqtt/Test Vibration Sensor', payload '{"action":null,"angle":50,"angle_x":0,"angle_x_absolute":90,"angle_y":0,"angle_y_absolute":90,"angle_z":90,"battery":100,"last_seen":"2021-08-07T15:35:45.508Z","linkquality":48,"sensitivity":"medium","strength":9,"voltage":3065}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:36:14.329Z","linkquality":60,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:36:17.428Z","linkquality":57,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Living TV Motion', payload '{"battery":100,"illuminance":3847,"illuminance_lux":2,"last_seen":"2021-08-07T15:37:00.934Z","led_indication":false,"linkquality":141,"motion_sensitivity":"medium","occupancy":false,"occupancy_timeout":20,"temperature":26.2,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:37:15.000Z","linkquality":63,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:37:18.102Z","linkquality":60,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:38:15.673Z","linkquality":60,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
info MQTT publish: topic 'zigbee2mqtt/Entrance Door Multisensor', payload '{"battery":98.5,"battery_low":false,"contact":true,"last_seen":"2021-08-07T15:38:18.773Z","linkquality":60,"moving":false,"tamper":false,"temperature":27.3,"x_axis":5,"y_axis":-1071,"z_axis":4}'
```
Debug info:
Zigbee2MQTT version: 1.21.0 commit: 70891ee
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210708

asjmcguire, 2021-08-07:
For me - one of mine is fixed: the max reporting time has been increased to 3600 and the min change has been increased to 5. The other one though is still sitting at a max reporting interval of 60 and a min change of 1. I did not reconfigure the one that has fixed itself, and the one that is not fixed I did try to press the yellow refresh button and wake the device up (both by opening and closing the door, and when that failed - by pressing the recessed button every second).

Fabiancrg, 2021-08-07:
I have 3 sensors and 2 of them were fixed after the upgrade+restart; the last one was still working but was reporting only if the door was open or closed.
The last one was fixed by removing the battery and reinserting it, just after that I pressed the reconfigure button from the Z2M frontend.

hellcry37, 2021-08-07:
already did reconfigure with sensor waked up successfully (no difference) and restarted z2m (no joy)

dcyonce, 2022-05-29:
This still seems to be an issue. 
These sensors still report every 10 seconds with NO changes.

Zigbee2MQTT:info  2022-05-28 21:24:20: MQTT publish: topic 'zigbee2mqtt/Dirty Sensor', payload '{"battery":93,"battery_low":null,"contact":null,"last_seen":"2022-05-29T02:24:20.766Z","linkquality":135,"moving":false,"tamper":null,"temperature":29.22,"voltage":2600,"x_axis":-973,"y_axis":20,"z_axis":-28}'
Zigbee2MQTT:info  2022-05-28 21:24:30: MQTT publish: topic 'zigbee2mqtt/Dirty Sensor', payload '{"battery":93,"battery_low":null,"contact":null,"last_seen":"2022-05-29T02:24:30.768Z","linkquality":147,"moving":false,"tamper":null,"temperature":29.22,"voltage":2600,"x_axis":-973,"y_axis":20,"z_axis":-28}'
Zigbee2MQTT:info  2022-05-28 21:24:54: MQTT publish: topic 'zigbee2mqtt/Dirty Sensor', payload '{"battery":93,"battery_low":null,"contact":null,"last_seen":"2022-05-29T02:24:54.395Z","linkquality":129,"moving":false,"tamper":null,"temperature":28.21,"voltage":2600,"x_axis":-973,"y_axis":20,"z_axis":-28}'
Zigbee2MQTT:info  2022-05-28 21:25:20: MQTT publish: topic 'zigbee2mqtt/Dirty Sensor', payload '{"battery":93,"battery_low":null,"contact":null,"last_seen":"2022-05-29T02:25:20.814Z","linkquality":153,"moving":false,"tamper":null,"temperature":28.21,"voltage":2600,"x_axis":-973,"y_axis":20,"z_axis":-28}'

asjmcguire, 2022-05-29:
@dcyonce remove the sensor and then readd it, that's how I fixed mine.