Koenkk/zigbee2mqtt #7923: Migrating from CC2531 to Zzh Issues with starting Z2M

miguelarios, 2021-07-02:
The discussion is below, but after flashing my Zzh! with the firmware appropriately using the method using python cc2538-bsl

Then I configured my configuration.yaml to have

```
homeassistant: true

permit_join: true

frontend:
  port: 8081

mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.0.0.2'
  user: '!secret user'
  password: '!secret password'
  client_id: 'zigbee2mqtt'
serial:
  port: /dev/ttyUSB0 #<- this i have changed from using serial/by-id AND just the serial usb location such as this

devices: devices.yaml

advanced:
  network_key: '!secret network_key'
  last_seen: epoch
  availability_timeout: 60
  rtscts: false
  pan_id: 0x1a64

```
Now I am just getting errors in the log like:

```
info  2021-07-02 14:04:03: Logging to console and directory: '/app/data/log/2021-07-02.14-04-03' filename: log.txt
info  2021-07-02 14:04:03: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
info  2021-07-02 14:04:03: Starting zigbee-herdsman (0.13.117)
error 2021-07-02 14:04:03: Error while starting zigbee-herdsman
```
Is there any way to debug further?

### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/7566

<div type='discussions-op-text'>

<sup>Originally posted by **miguelarios** May 22, 2021</sup>
I have read the documentation that if I change hardware devices that are different base hardware I need to repair my devices. What I’m curious is since I run in a container do I need to just start with a fresh install of the container and delete my devices database?

Do I need to do an unpairing or sorts?

What happens if all I change is the configuration.yaml file and don’t delete the devices in the database? Does this mess with the migration?</div>

miguelarios, 2021-07-02:
I already ran the script located [here](https://gist.github.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23) to check if I flashed the zzh correct and I get this response so i do not think it is my coordinator.

```
Got 7 bytes in response to PING command: b'\xfe\x02a\x01Y\x06='
PASS|OK
```

I also have simply just renamed my old config file to something else to momentarily run my new config file that I shared up there ☝🏽 . I was able to just quickly undo the config file naming to turn Z2M back on with my old coordinator.

I am curious what I am doing wrong or what is going on, and why i cannot switch coordinators.

miguelarios, 2021-07-02:
I have done some more searching on passed issues and I think issue #7433 is similar if not the same thing that is happening to me.

Koenkk, 2021-07-03:
can you try with: `pan_id: 0x1a65`?

ArJay60, 2021-07-04:
Do you probably get elifcycle errors? Best to be seen when running 'npm start' from /opt/zigbee2mqtt after stopping zigbee. If do unplug and replug Zzh stick and run 'npm start' again. If elifcycle error persist then move stick away with USB extensions cable ( probably interference issue)

miguelarios, 2021-07-08:
> can you try with: `pan_id: 0x1a65`?

OK so i tried like and still no cigar...

```
homeassistant: true

permit_join: true 

frontend:
  port: 8081

mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.0.0.2'
  user: '!secret user'
  password: '!secret password'
  client_id: 'zigbee2mqtt'
serial:
  port: /dev/ttyUSB0

devices: devices.yaml

advanced:
  network_key: '!secret network_key'
  last_seen: epoch
  availability_timeout: 60
  rtscts: false
  pan_id: 0x1a65
  log_level: debug
```

```
debug 2021-07-07 22:12:14: Loaded state from file /app/data/state.json
info  2021-07-07 22:12:14: Logging to console and directory: '/app/data/log/2021-07-07.22-12-13' filename: log.txt
debug 2021-07-07 22:12:14: Removing old log directory '/app/data/log/2021-07-02.13-52-18'
info  2021-07-07 22:12:14: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
info  2021-07-07 22:12:14: Starting zigbee-herdsman (0.13.117)
debug 2021-07-07 22:12:14: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6757},"serialPort":{"path":"/dev/ttyUSB0","rtscts":false}}'
error 2021-07-07 22:12:14: Error while starting zigbee-herdsman
```

miguelarios, 2021-07-08:
> Do you probably get elifcycle errors? Best to be seen when running 'npm start' from /opt/zigbee2mqtt after stopping zigbee. If do unplug and replug Zzh stick and run 'npm start' again. If elifcycle error persist then move stick away with USB extensions cable ( probably interference issue)

@ArJay60 how would I go about doing this from docker? Do i just run the command when doing a docker exec or what? I just did a quick and dirty way by restarting my container with the pan id that @Koenkk recommened minus the USB extension I have been using but still same issue.

miguelarios, 2021-07-08:
OK so I added the environment variable for debugging zigbee herdsman and I see this in the log

```
^[[34mZigbee2MQTT:debug^[[39m 2021-07-07 22:54:51: Loaded state from file /app/data/state.json
^[[32mZigbee2MQTT:info ^[[39m 2021-07-07 22:54:51: Logging to console and directory: '/app/data/log/2021-07-07.22-54-51' filename: log.txt
^[[34mZigbee2MQTT:debug^[[39m 2021-07-07 22:54:51: Removing old log directory '/app/data/log/2021-07-02.15-00-19'
^[[32mZigbee2MQTT:info ^[[39m 2021-07-07 22:54:51: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
^[[32mZigbee2MQTT:info ^[[39m 2021-07-07 22:54:51: Starting zigbee-herdsman (0.13.117)
^[[34mZigbee2MQTT:debug^[[39m 2021-07-07 22:54:51: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedP>
^[[31mZigbee2MQTT:error^[[39m 2021-07-07 22:54:51: Error while starting zigbee-herdsman
^[[31mZigbee2MQTT:error^[[39m 2021-07-07 22:54:51: Failed to start zigbee
^[[31mZigbee2MQTT:error^[[39m 2021-07-07 22:54:51: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
^[[31mZigbee2MQTT:error^[[39m 2021-07-07 22:54:51: Exiting...
^[[31mZigbee2MQTT:error^[[39m 2021-07-07 22:54:51: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
```

I see that there are issues with opening the serial port. I will try again using the serial by-id perhaps.

miguelarios, 2021-07-08:
Yea it doesn't work still. 😢

miguelarios, 2021-07-08:
Oh man, I think I forgot a critical thing. Each time I have restarted the container I haven't passed the new USB prior to starting the container 🤦🏽. It is late for me now so i will try again tomorrow night, but now I think I know what is going on. I don't have time to repair all the devices once it works.

closed 2021-08-17 by github-actions[bot]