Koenkk/zigbee2mqtt #7925: TS004F Switch Buttons not working, only Battery Status

J-117, 2021-07-02:
TS004F will only Report when you press the upper left button, then it reports Battery Status and Link Strength,

Other Buttons will not Report any State (Logs in Debug Mode)

Using Standard Configuration just adding the Device through learn, also tryed multiple devices -> same Problem

heroiamarelo1, 2021-07-09:
I'm having the same issue.
Zigbee Dongle USB - Sonoff CC2531

[image]
[image]
[image]

J-117, 2021-07-10:
Yes i get the exact same Messages, maybe its the Dongle? Is your Dongle already flashed, mine should be as i bought it as flashed, maybe someone has another Dongle to try,

Kind Regards,

J-117, 2021-07-25:
Tested with a Conbee 2, and it works, all Buttons send their respective individual Payload, seems to be a Problem with the Dongle

heroiamarelo1, 2021-07-30:
Hello. Sorry for the offtopic but how did you do it? I added my switches to conbee II and conbee assumes the switches are lights. Two of the commands work to power on and off the lights (no idea why!) the other two buttons do nothing. I can't sign any commands because they are considered lights instead of switches, even in home assistant.

closed 2021-09-06 by github-actions[bot]