Koenkk/zigbee2mqtt #7926: Error Failed to call 'Receive' 'onZigbeeEvent'

a-bailey, 2021-07-02:
## What happened
When I control my curtain actuactors I receive an error message in the zigbee2mqtt log which says:

`Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))`

This didn't happen before the 1.20 update, it seems to work fine but I'd like to get the error fixed the configuration for the device is as follows

## What did you expect to happen
No error in the log

## How to reproduce it (minimal and precise)

This is the device I'm trying to control which gives me the error, unfortunately I can't provide more steps to reproduce the error maybe someone can nudge me inn the ridht direction.

```
{
        zigbeeModel: ['LXN59-CS27LX1.0'],
        model: 'LXN59-CS27LX1.0',
        vendor: '3A Smart Home DE',
        description: 'Smart Zigbee 3.0 curtain controller',
        fromZigbee: [fz.cover_position_tilt, fz.LXN59_cover_state_via_onoff, fz.LXN59_tuya_cover_options],
        toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration],
        whiteLabel: [{vendor: '3A Smart Home DE', model: 'LXN59-CS27LX1.0', description: 'Smart Zigbee 3.0 curtain controller'}],
        meta: {configureKey: 2, disableDefaultResponse: true},
        exposes: [e.cover_position(), exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['closuresWindowCovering'])
            await reporting.currentPositionLiftPercentage(endpoint1);
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
```

## Debug info
Zigbee2MQTT version: 1.2.0
Adapter hardware:  CC1352P-2
Adapter firmware version: latest

Koenkk, 2021-07-03:
Please provide some more context of this logging by providing the debug log.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

a-bailey, 2021-07-06:
A bit more information I got from the debug log:

```
Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))

Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":12,"tuyaMovingState":1}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))

Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))

Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":1}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))

Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":10,"tuyaMovingState":2}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))

Debug Received Zigbee message from 'buero_rollladen_links', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":10,"tuyaMovingState":2}' from endpoint 1 with groupID 0
Error Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:117:42) at Array.filter (<anonymous>) at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:116:65) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44))
```

Koenkk, 2021-07-06:
I see this device is not supported by z2m out-of-the-box, are you using an external converter?

The error is that these 2 converters do not exist: `fz.LXN59_cover_state_via_onoff, fz.LXN59_tuya_cover_options`

a-bailey, 2021-07-06:
Ah you are right sorry, forgot to include the configuration after updating. Will close this issue thanks for the help.

closed 2021-07-06 by a-bailey