Koenkk/zigbee2mqtt #7928: Can't rejoin Aqara WSDCGQ11LM sensors after removing them first

jilles, 2021-07-03:
## What happened
All my sensors stopped showing humidity correctly in Domoticz (Domoticz interpreted Humidity as Temperature and said it was 0 degrees celsius). So I decided to remove them all together: all devices from Domoticz and from zigbee2mqtt (configuration.yaml, database.db, state.json). 

But now it seems that I can't rejoin them again. I've rebooted and updated everything. Replaced batteries. Tried joining in Domoticz and in zigbee2mqtt, but to no avail. 

I was able to join (first time, so no rejoin) an Aqara vibration sensor.

## What did you expect to happen
That it would rejoin and I could use the sensors again.

## How to reproduce it (minimal and precise)
In my case: remove the device from Domoticz and zigbee2mqtt and then try to rejoin.

## Debug info
Zigbee2MQTT version: 1.20.0 commit: 41b67fdd
Adapter hardware: CC26X2R1 (ZZH!)
Adapter firmware version: 20201026

mhzawadi, 2021-07-03:
I think I have a similar issue, I had 2 sonoff door sensors and some Ikea bulbs that dont now rejoin to my conbee2.

I too have tied lots to make it work 😢

jilles, 2021-07-04:
I ran into a Reddit saying that you just have to keep trying. First reset, then try and force an update from the device. After some minutes trying I got three rejoined. Didn't have the patience to do the others. So basically it looks like my problem is somewhat solved. I do still wonder why I never had this issue before. I have joined all off them (7) at least 3 to 4 times before. And that would work within the 1 or 2 tries. Would love to learn what is what.

closed 2021-08-11 by github-actions[bot]