Koenkk/zigbee2mqtt #7939: Linkind 5-Key Remote Control (ZBT-RGBWSwitch-D0801)

dm6tt, 2021-07-04:
## Information about the device + link
Manufacturer: Linkind
ID: ZS230002-IE-V1A02
Description: 5-Key Remote Control
Zigbee model:  ZBT-RGBWSwitch-D0801
## data/database.db entry of the device
{"id":15,"type":"EndDevice","ieeeAddr":"0xccccccfffeae0173","nwkAddr":65023,"manufId":4456,"manufName":"lk","powerSource":"Battery","modelId":"ZBT-RGBWSwitch-D0801","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,1,3,2821,4096,64769],"outClusterList":[3,4,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-RGBWSwitch-D0801","manufacturerName":"lk","powerSource":3,"zclVersion":3,"appVersion":43,"stackVersion":6,"hwVersion":2,"dateCode":"20200706-95","swBuildId":"2.27"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[]}},"appVersion":43,"stackVersion":6,"hwVersion":2,"dateCode":"20200706-95","swBuildId":"2.27","zclVersion":3,"interviewCompleted":true,"meta":{}}

I got the device paired, but no response in the log when pressing a key.
```
Jul  4 13:51:35 nuc zigbee2mqtt[909]: #033[32mzigbee2mqtt:info #033[39m 2021-07-04 13:51:35: Device '0xccccccfffeae0173' joined
Jul  4 13:51:35 nuc zigbee2mqtt[909]: #033[32mzigbee2mqtt:info #033[39m 2021-07-04 13:51:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xccccccfffeae0173"}}'
Jul  4 13:51:35 nuc zigbee2mqtt[909]: #033[32mzigbee2mqtt:info #033[39m 2021-07-04 13:51:35: Starting interview of '0xccccccfffeae0173'
Jul  4 13:51:35 nuc zigbee2mqtt[909]: #033[32mzigbee2mqtt:info #033[39m 2021-07-04 13:51:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xccccccfffeae0173"}}'
Jul  4 13:51:37 nuc zigbee2mqtt[909]: #033[34mzigbee2mqtt:debug#033[39m 2021-07-04 13:51:37: Device '0xccccccfffeae0173' announced itself
Jul  4 13:51:47 nuc zigbee2mqtt[909]: #033[34mzigbee2mqtt:debug#033[39m 2021-07-04 13:51:47: Received Zigbee message from '0xccccccfffeae0173', type 'readResponse', cluster 'genBasic', data '{"modelId":"ZBT-RGBWSwitch-D0801","manufacturerName":"lk","powerSource":3}' from endpoint 1 with groupID 0
Jul  4 13:51:47 nuc zigbee2mqtt[909]: #033[33mzigbee2mqtt:warn #033[39m 2021-07-04 13:51:47: Received message from unsupported device with Zigbee model 'ZBT-RGBWSwitch-D0801'
Jul  4 13:51:47 nuc zigbee2mqtt[909]: #033[33mzigbee2mqtt:warn #033[39m 2021-07-04 13:51:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

```

I tried to follow the 'How to support new devices page', but got no reaction of adding a .js file:

confiuguration.yaml
```
advanced:
  log_level: debug
  last_seen: epoch
external_convertes:
  - ZBT-RGBWSwitch-D0801.js
```

ZBT-RGBWSwitch-D0801.js
```
automation@nuc:/opt/IOTstack/volumes/zigbee2mqtt/data$ cat ZBT-RGBWSwitch-D0801.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZBT-RGBWSwitch-D0801'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZS230002-IE-V1A02', // Vendor model number, look on the device for a model number
    vendor: 'Linkind', // Vendor of the device (only used for documentation and startup logging)
    description: '5-Key Remote Control', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

```

Has someone experience with that device or help me with the next steps?

Thanks
Marcus

nurikk, 2021-07-04:
you need to add this line at the very bottom of your converter file

`module.exports = definition;`

dm6tt, 2021-07-04:
Thanks for the reply.
I just checked it, it is there. Only missed the last line when copying.

pumabaer, 2021-08-08:
have you found a solution to the problem?
I have the same problem that I get no response to a key press.

closed 2021-08-16 by github-actions[bot]