Koenkk/zigbee2mqtt #7941: Moes BRT-100-TRV not reporting after OTA update

wizu, 2021-07-04:
## What happened
Z2M showed me, that my TRV have an OTA update available, so I've updated it's firmware.
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#moes-brt-100-trv
After successful upgrade I couldn't control it any more. I've removed it from Z2M and added again serveral times, but it didn't help.
Not it only reports:
{
    "child_lock": "UNLOCK",
    "eco_mode": "OFF",
    "linkquality": 34,
    "update": {
        "state": "idle"
    }
}
and there is nothing abnormal in the logs, about this device.
Changing child lock from Z2M works, but nothing more.

## Debug info
Zigbee2MQTT version: 1.20.0 commit: 41b67fd
Adapter hardware: CC2538 + CC2592
Adapter firmware version: zStack30x 20200211

CVKBaca, 2021-07-10:
I have the same problem. TVR does not respond after firmware upgrade but nothing abnormal in the logs..

Zigbee2mqtt version: 1.20.0-1
adapter: CC2652RB development stick with zigbee2mqtt firmware

vladi1234, 2021-07-12:
Hi, whether thermostat itself works? And what firmware version now shows thermostat?

wizu, 2021-07-12:
Termostat works ok and show firmware U1, or UI

vladi1234, 2021-07-12:
Does z2m pairing work correctly?

wizu, 2021-07-12:
Yes, I have tried that several times and it worked every time, but no difference.

vladi1234, 2021-07-12:
I need more information.

[image]

I need result.

CVKBaca, 2021-07-12:
sorry I dont see fw version :(
[image: image.png]

wizu, 2021-07-12:
`Read result of 'genBasic': {"appVersion":87,"zclVersion":3,"stackVersion":0,"hwVersion":1,"manufacturerName":"_TZE200_b6wax7g0","modelId":"TS0601","dateCode":"","powerSource":3}`

vladi1234, 2021-07-12:
Read result of 'genBasic': {"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"manufacturerName":"_TZE200_b6wax7g0","modelId":"TS0601","dateCode":"","powerSource":3}

On my thermostat "appVersion" is: 64. I did not update.

CVKBaca, 2021-07-12:
[image]

vladi1234, 2021-07-12:
Moes BRT-100-TRV thermostat does not have a firmware update yet.
I think the thermostat got the wrong firmware update from Tuya.

Look here:
https://github.com/Koenkk/zigbee-OTA/commit/8c9017844d110ea8d499da0158e0e4c85d8aa1d5

wizu, 2021-07-12:
Is there a way to get it back to old firmware? @Koenkk

vladi1234, 2021-07-12:
Tuya and Moes have the same "modelId": "TS0601".
That is not right.
We have to speak to @Koenkk that he differentiates according to "manufacturerName": "_ TZE200_b6wax7g0" and not according to "modelId": "TS0601".

vladi1234, 2021-07-13:
Tuya and Moes also have the same "manufacturer code": 4098,

vladi1234, 2021-07-13:
```
{
	"id": 17,
	"type": "EndDevice",
	"ieeeAddr": "0xcc86ecfffebxxxxx",
	"nwkAddr": 53108,
	"manufId": 4098,
	"manufName": "_TZE200_b6wax7g0",
	"powerSource": "Battery",
	"modelId": "TS0601",
	"epList": [
		1
	],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 81,
			"inClusterList": [
				0,
				4,
				5,
				61184
			],
			"outClusterList": [
				25,
				10
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"65503": "�f|(\u0012�f|(\u0012",
						"65506": 31,
						"65508": 0,
						"modelId": "TS0601",
						"manufacturerName": "_TZE200_b6wax7g0",
						"powerSource": 3,
						"zclVersion": 3,
						"stackVersion": 0,
						"hwVersion": 1,
						"dateCode": "",
						"appVersion": 64
					}
				},
				"genDeviceTempCfg": {
					"attributes": {}
				},
				"genLevelCtrl": {
					"attributes": {}
				},
				"msTemperatureMeasurement": {
					"attributes": {}
				}
			},
			"binds": [
				{
					"cluster": 0,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0022812735",
					"endpointID": 1
				},
				{
					"cluster": 61184,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0022812735",
					"endpointID": 1
				}
			],
			"configuredReportings": [],
			"meta": {}
		}
	},
	"appVersion": 64,
	"stackVersion": 0,
	"hwVersion": 1,
	"dateCode": "",
	"zclVersion": 3,
	"interviewCompleted": true,
	"meta": {},
	"lastSeen": 1626155319566
}
```

wizu, 2021-08-13:
problem still exists

CVKBaca, 2021-08-17:
I confirm the same problem. I also found out that after the OTA firmware update, it is no longer possible to set the device to pairing mode. I tried to reset the device manually via the AE function and entering a value "01", but even that didn't help.

vladi1234, 2021-09-03:
> Is it possible to force an update manually for HY-368 (white moes clone) by means of zigbee2mqtt?
> 
> Here is binary and I know that it works well with my TRVs.
> https://github.com/Koenkk/zigbee-OTA/tree/master/images/Tuya
> 
> At the moment it looks that OTA update is locked in zigbee2mqtt

What "manufacturerName:" does "HY-368" TVR have?
Do you mean this "1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin" file?

pawol, 2021-09-03:
Do you mean this "1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin should be fine for 
this TRV:
https://pl.gearbest.com/smart-home-controls/pp_009269115611.html

At the moment I have 3 pieces of TRV updated and 1 not updated. I would like to do it but OTA is locked now

vladi1234, 2021-09-03:
I still need to know which 
`"manufacturerName:"` 
has "HY-368" TVR?

bruvv, 2021-09-03:
[image]

wizu, 2021-09-03:
Model Zigbee
TS0601
zigbee_manufacturer
_TZE200_cwnjrr72

pawol, 2021-09-03:
Since Augusto OTA update worked with it:
[image]

bruvv, 2021-09-03:
Well this is what it is showing on my end in the OTA field:
[image]

vladi1234, 2021-09-04:
> Model Zigbee
> TS0601
> zigbee_manufacturer
> _TZE200_cwnjrr72
> 
> And I also have _TZE200_ckud7u2l
> 
> Thats two of my HY-368

@wizu are you sure that the _TZE200_cwnjrr72 and _TZE200_ckud7u2l are compatible?

wizu, 2021-09-04:
I bought 10 of _TZE200_cwnjrr72 a week ago and they look the same. But had no chance to update it so I don't know if they will work with this firmware - I think they will.
I will try to update one of them when the OTA will be available.

wizu, 2021-09-04:
@vladi1234 so yes, let's assume that they are compatible and I will check that on my own TRV.

MattWestb, 2021-09-04:
Keep in mind that most tuya TRVs is coming with 2 different Zigbee modules TZE200 and TYST11 and can having the same tuya MCU and if flashing the wrong version is very likely not working or braking the zigbee module or changing the comport so it cant communicating with the tuya MCU.

Also the Zigbee OTA file is only updating the Zigbee module on TS0601 device and not the tuya MCU that must being done with tuya (DP) commands.

So updating the Zigbee module is not helping with problems in the tuya MCU that is the TRV controller only fixing "zigbee problems" like bad cluster setting or reporting settings from zigbee cluster (that is barley not used on this devices) then the tuya MCU is only using the Zigbee module as one "tunneling" device for its communication with tuya cloud.

artkrz, 2021-09-04:
I think z2m should have an option to mark device with "ignore fw update checks" or something similar. If device works well enough it doesn't have to be updated.

vladi1234, 2021-09-05:
> Ich denke, z2m sollte eine Option haben, um das Gerät mit "Ignore fw update checks" oder ähnlichem zu markieren. Wenn das Gerät gut genug funktioniert, muss es nicht aktualisiert werden.

[image]

You can switch off here.

If the device works well enough, it doesn't need to be updated. ------> Then just don't do it!

vladi1234, 2021-09-06:
First part of my OTA update PR has already been merged with z2m, but the second part of PR has to wait another 2 months, then it will be merged.
Reason see here:
https://github.com/Koenkk/zigbee-OTA/pull/50#issuecomment-913736281

wizu, 2021-09-06:
Great, thanks @vladi1234

closed 2021-09-06 by wizu

CVKBaca, 2021-09-08:
is there any chance of repairing malfunction devices (_ TZE200_b6wax7g0) in the future with correct firmware via OTA update?

vladi1234, 2021-09-09:
The possibility exists only if we have OTA update file.

muddy79, 2021-09-11:
> First part of my OTA update PR has already been merged with z2m, but the second part of PR has to wait another 2 months, then it will be merged.
> Reason see here:
> [Koenkk/zigbee-OTA#50 (comment)](https://github.com/Koenkk/zigbee-OTA/pull/50#issuecomment-913736281)

Well, I understand the reason, but it means there will be people like me who are unable to update their MOES TS0601 devices (_TZE200_ckud7u2l), which is crucial to get them working properly: without update, the device goes to sleep state and it's not possible to use it via HA. I've just installed 6 of them, the heating season is about to start at my area, and I just found out the OTA update is not possible... :( Any suggestions?

MattWestb, 2021-09-11:
OTA can checking the version of Z2M and if its old its blocking the update and if in pace with Z2M is allowing the update but its need some more code being added and all use dont need waiting so long.

Safs, 2021-09-23:
Is any news about TS0601 update?

jmarfer, 2021-10-03:
Good Morning. I cannot use this valve for my automations because it does not show system mode (to be able to switch HVAC between heat, auto and off). Would this be contemplated in the next update? When is it expected?

Thanks

vladi1234, 2021-10-03:
Moes BRT-100-TRV thermostat does not have to have a 'system mode' either.
It works very well in any automation scenario.

1. For this you have to use the 'manual' position.

2. Set 'current_heating_setpoint' that what you need room target temperature.

3. Set 'eco_temperature' to '0', or '5' in any case less than the target room temperature.

4. To switch off the heating you have to use 'eco-mode_on'
and to switch on 'eco-mode_off'.

**eco-mode_on = off**
**eco-mode_off = auto**
**boost_heating_on = heat**

In any case, I am very satisfied with it.

jmarfer, 2021-10-03:
> boost_heating_on

Thanks for the quick reply.

Actually I need to create a thermostat and dispense with the valve temperature sensor, using an external one. I only need a valve opening and closing switch. With another valve I have succeeded but with this I am not capable.

Following your indications, when I put "boost_heating_on = heat" the valve changes but in a while it returns to its original position.

Is there a way to get a "clean" valve open / close switch to use in a climate?

Thanks again

vladi1234, 2021-10-03:
@jmarfer Then try this:

1. For this you have to use the 'manual' position.

2. Set 'current_heating_setpoint' 35°C room target temperature.

3. Set 'eco_temperature' to '0', or '5' in any case less than the target room temperature.

4. To switch off the heating you have to use 'eco-mode_on' and to switch on 'eco-mode_off'.

**eco-mode_on = off**
**eco-mode_off = on**

jmarfer, 2021-10-03:
> @jmarfer Then try this:
>
> 1. For this you have to use the 'manual' position.
> 2. Set 'current_heating_setpoint' 35°C room target temperature.
> 3. Set 'eco_temperature' to '0', or '5' in any case less [...]

Perfect @vladi1234, now it works for me. There has been a time when the 'current_heating_setpoint' dropped to 35 and I did not understand why. I will follow up.

One last thing. By putting the switch indicated in the climate, it works the other way around, that is, when it has to heat the valve it closes and when it has to go off the valve opens. Can I reverse this?

vladi1234, 2021-10-03:
@jmarfer If you write automation scenario, it doesn't really matter 'on' or 'off'.
I don't think you can optically swap things around.

sokol07, 2021-10-05:
Any information about fix of this issue?
I wanted to implement these TRVs in whole house, fortunately I bought only one but at the moment after a while the device is loosing any communication and I can't update the firmware (as I don't have a standalone tuya gateway)...

vladi1234, 2021-10-05:
> Any information about fix of this issue? I wanted to implement these TRVs in whole house, fortunately I bought only one but at the moment after a while the device is loosing any communication and I can't update the firmware (as I don't have a standalone tuya gateway)...

@sokol07 Which TRV do you mean?

sokol07, 2021-10-05:
Good question, I've bought it as "GTZ04 (Moes TS0601)" but it is listed in Z2M as `_TZE200_cwnjrr72`.
The OTA update is, as mentioned earlier, impossible due to the "No image available for imageType '5634'", which I guess means that the OTA is still disabled for this model?

vladi1234, 2021-10-05:
> Good question, I've bought it as "GTZ04 (Moes TS0601)" but it is listed in Z2M as `_TZE200_cwnjrr72`. The OTA update is, as mentioned earlier, impossible due to the "No image available for imageType '5634'", which I guess means that the OTA is still disabled for this model?

OTA update is currently deactivated for this TRV.

https://github.com/Koenkk/zigbee2mqtt/issues/7941#issuecomment-913752102

sokol07, 2021-10-05:
Yeah, I've read the topic, I now about the deactivation, I was just curious if you have any forecast about fixing this issue - is it "while" like 2 weeks or more lika 2 months?

vladi1234, 2021-10-07:
On November 1st, PR will be merged.

vladi1234, 2021-11-01:
@wizu  worked for you with OTA update?

wizu, 2021-11-01:
@vladi1234 yes, I have updated one of my TRV and it looks like it is working ok. Tommorow I will try to update the rest of them

vladi1234, 2021-11-01:
@wizu Super! Thanks!

oopiicaa, 2021-11-01:
I can confirm too, upgrade went through okay, everyrhing seems to work.

Coollision, 2021-11-10:
Hi sorry to poke a closed thread, but I recently acquired a Moes thermostat (HY368 _TZE200_cwnjrr72).
 first thing I did was an ota update, and now it only reports its temp (also acts) once an hour even when heating so yeah small room == major overheating. is there a way arround this or a fix?

amarce, 2021-11-11:
Hi,

The update is not appearing for me, I keep getting the error "No image available for imageType '5634'"

Mine is reported as _TZE200_b6wax7g0 and the firmware build date appears empty under the OTA section. 

I'm using the version of zigbee2mqtt that comes with the last home assistant, should I update it or something?

Thanks a lot!

vladi1234, 2021-11-11:
>"No image available for imageType '5634'"

No image available Is correct, because Moes BRT-100-TRV "_TZE200_b6wax7g0" thermostat doesn't have its own firmware update yet.

"appVersion" 64 is the last current version.

amarce, 2021-11-11:
Hi Vladi,

Understood.

Thanks a lot for the reply!

romek92, 2021-11-14:
Update _TZE200_cwnjrr72 by OTA is safe now?

henkiejan1, 2023-09-26:
Hi, i got yesterday this TRV and Zigbee2MQTT report also there is a update available from version 67 to 192. And more importantly it will not brick the device? Is there someone who could say it's safe to upgrade? I upgraded my Tuya TS011F powerplugs also from 67 to 192 without anyproblems. But a thermostat is not a power plug. ;)

sgilyin, 2023-10-05:
> Hi, i got yesterday this TRV and Zigbee2MQTT report also there is a update available from version 67 to 192. And more importantly it will not brick the device? Is there someone who could say it's [...]

DO NOT INSTALL THIS!!!
Today my thermostat was broken after this ota update ;-(

[6 more replies]