Koenkk/zigbee2mqtt #7942: Aqara T1 without neutral (SSM-U02) not able to join the network

jd1900, 2021-07-04:
## What happened
I can't make the Aqara T1 without neutral (SSM-U02 - https://www.zigbee2mqtt.io/devices/SSM-U02.html) join. I first press 8 seconds the device button to be in pairing mode.
It "seems" to be joining, I can even control it for a few seconds, and then it is disconnected. 
According to the logs, the pairing is succesful, however, after a few seconds I can't control it anymore.
The blue light of the device keeps blinking all the time quickly. According to the documentation it means that "Zigbee network is connecting".
When I "Disable join" devices, the light changes from blue blinking to red blinking slowly (Zigbee network is not connected).

I've tried several times, and even with the device a few cm. away from the coordinator. Always the same.

I've tried removing the device, keeping it, reconfiguring...

The best I can get is to control it (turn on or off) right after pairing. 

According to the logs, the device seems to be keeping trying to connect announcing itself all the time.

Manually turning on or off when the device is in red works perfectly. 
When the device is in blue, it works, but after a few seconds it turns off back again.
Same when controlling remotely with Z2M when it allows me (for a few seconds).

## What did you expect to happen
Join properly.

## How to reproduce it (minimal and precise)
Enable joining devices.
Press 8 seconds the device button.

## Debug info
Zigbee2MQTT version: 1.20.0 commit: 1.20.0
Adapter hardware: zzh
Coordinator type: zStack3x0
Coordinator revision: 20210120
Frontend version: 0.4.29

jd1900, 2021-08-04:
No one can help me with this?

closed 2021-09-12 by github-actions[bot]

jd1900, 2021-12-09:
No one?
I've seen that the DLKZMK12LM has recently been added (https://github.com/Koenkk/zigbee-herdsman-converters/pull/3368) by @DJTerentjev 

But even in the latest version I still have the same problem, even pairing a few cm away from the coordinator.

DJTerentjev, 2021-12-09:
Hello! Could you describe your setup.  What type of bulb you have connected? Feells like the device doesn't have enough power supply. I have same relay and it's working well. If you suspect the problem with initial pairing, try to remove device, **restart z2m** and pair device again.

jd1900, 2021-12-09:
> Hello! Could you describe your setup. What type of bulb you have connected? Feels like the device doesn't have enough power supply. I have same relay and it's working well. If you suspect the problem with initial pairing, try to remove device, **restart z2m** and pair device again.

For the pairing/testing I'm using a GU10 led bulb. 
As described, the Aqara is working fine manually. Manually pressing the Aqara button turns on/off the bulb properly. And it "pairs" for a few seconds and I can control it from Zigbee2Mqtt for a few seconds, until it disconnect. But zigbee2mqtt doesn't realize of the "disconection".
Therefore, I have to "force remove" the device to be able to try the pairing again.

DJTerentjev, 2021-12-09:
GU10 led bulb (especially if it is only bulb) may be not enough. Relay without neutral has limitation with minimum connected load. Try to use conventional incandescent lamp firs for testing.

julian-ladjani, 2021-12-10:
> 
I have exactly same problem, I use Conbee II usb dongle.

jd1900, 2021-12-10:
> I have exactly same problem, I use Conbee II usb dongle. Their are 3 bulbs connected on it (common E27 led bulbs)

So it seems there must be a problem with the device integration in Z2M, right?