Koenkk/zigbee2mqtt #7943: States not changing with new device

dbroome1978, 2021-07-05:
# Bug Report
## What happened
I can view but not change the devices state.

https://www.moeshouse.com/collections/tuya-zigbee-switch/products/zigbee-wall-touch-smart-light-switch-with-neutral-wire-no-neutral-wire-no-capacitor-needed-smart-life-tuya-2-3-way-muilti-control-association-hub-required-4-gang-white-us

It looks similar (very) to https://www.zigbee2mqtt.io/devices/TS0601_switch.html
So i borrowed the configs from here 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/75a13064874174606c7434d8afa380467d46d686/devices/tuya.js
https://github.com/Koenkk/zigbee-herdsman-converters/blob/f22b305fa7324498bc38246db0ce27bf7235dc79/devices/moes.js

So mine looks like this
https://paste.ubuntu.com/p/2hqPV4Trh9/
I am able to see now the states all 4 switches but I cannot change them

mosquitto_pub -t "zigbee2mqtt/0xbc33acfffeb12f9a/l1/set" -h 192.168.1.45 -m  {\"state_l1\":\"ON\"}
mosquitto_pub -t "zigbee2mqtt/0xbc33acfffeb12f9a/l1/set" -h 192.168.1.45 -m  {\"state\":\"ON\"}

I get this error
Error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'includes' of undefined at /app/lib/extension/publish.js:187:60 at Array.find (<anonymous>) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:187:42) at Controller.callExtensionMethod (/app/lib/controller.js:386:44) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:95:5))

Below is the log.  I've included when the state changes from physically changing the switch and when I try it over the server
https://paste.ubuntu.com/p/mRyNkKQSzY/
## Debug Info
Zigbee2MQTT version: 1.19.1 commit: 9bd4693
Adapter hardware: CC2652R
Adapter firmware version: zzh (newest)

dbroome1978, 2021-07-06:
I reposted this on the forum as well.  The edit to my post here is simply cleaning it up and making it more readable

Koenkk, 2021-07-06:
Try with: 
```
mosquitto_pub -t "zigbee2mqtt/0xbc33acfffeb12f9a/l1/set" -h 192.168.1.45 -m '{"state":"ON"}'
```

dbroome1978, 2021-07-06:
I get back the same error

> Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'includes' of undefined at /app/lib/extension/publish.js:187:60 at Array.find (<anonymous>) at [...]

Koenkk, 2021-07-06:
Could you post some more logging of this? (what happens before?)

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

dbroome1978, 2021-07-06:
Here you go 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6772111/log.txt)
If you need the raw output from docker, I can do that as well.
I'm not sure if there is a way to have it only report logs for the specific device
The device ID is 0xbc33acfffeb12f9a

dbroome1978, 2021-07-06:
Here is an extra one with a physical button press as well as using mqtt
[deug with button press vs mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/6772351/deug.with.button.press.vs.mqtt.txt)

Koenkk, 2021-07-07:
The issue is in your external converter, it uses `fz.tuya_switch2`, this converter does not exist anymore. You can probably just remove it from `fromZigbee` since `fz.tuya_switch` should be good.

Let me know if that works, then I can add support for this device out-of-the-box.

dbroome1978, 2021-07-07:
I receive the same error with or without fz.tuya_switch2
I did reconfigure after making the change.  Do I need to do a full remove and pair?

Koenkk, 2021-07-07:
@dbroome1978 try to make `fromZigbee` empty, then add converters one by one until you see the errors.

dbroome1978, 2021-07-07:
Ok - i removed all and have it no converters listed.  Now this is odd.  If I try to change the state once, I get "No converter available" but when I try a second time, i get the old error

> Debug Received MQTT message on 'zigbee2mqtt/0xbc33acfffeb12f9a/l1/set' with data '{"state_l1":"OFF"}'
> Error No converter available for 'state' (undefined)
> 

> Debug Received MQTT message on 'zigbee2mqtt/0xbc33acfffeb12f9a/l1/set' with data '{"state_l1":"OFF"}'
> Error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'includes' of undefined at /app/lib/extension/publish.js:187:60 at Array.find (<anonymous>) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:187:42) at Controller.callExtensionMethod (/app/lib/controller.js:386:44) at processTicksAndRejections (internal/process/task_queues.js:95:5))
>

dbroome1978, 2021-07-07:
I believe it's working and will double check when I get home.
The 'tz' line had been commented out.  Changing that and changing the formatting to match 'fz' lines got it working.

closed 2021-07-07 by dbroome1978

dbroome1978, 2021-11-01:
I lost my container and am setting up from scratch.  Just an FYI that this does not yet seem to be supported out of the box.  The converter does work, though