Koenkk/zigbee2mqtt #7951: Ikea FYRTUR cover entity not discovered via MQTT discovery

JackEker, 2021-07-05:
# Bug Report
## What happened
I have had to re-pair a FYRTUR Ikea blind. The initial pairing (~8 months ago) worked perfectly and showed up as a cover entity within HA. Re-pairing required resetting the blind and repeater. The cover entity is not being discovered no matter how I attempt to pair the blind. The entities discovered are:
- binary_sensor.<id>_update_available
- sensor.<id>_battery
- sensor.<id>_linkquality
- sensor.<id>_update_state
 
They are all initially disabled. Because of this I can't add the cover entity to any room in order to control it via the HA UI. The only way I've found to control the blind is to go into the Zigbee2MQTT UI select the blind device and enter the 'Exposes' tab. 

This initially didn't work with the device on the firware it shipped with so I OTA updated both the blind and repeater to the latest firmware. Still no joy. 

I have tried to re-pair in many different ways including:
- Pairing the repeater (hold pin in hole for ~10 seconds)
- Waiting a few minutes to ensure it's online
- Hold the repeater against the blind battery cover
- Hold both blind buttons for 5 seconds to pair

The network map shows the blind as an endpoint connected to via repeater.

I have used [Manual Home Assistant Configuration](https://www.zigbee2mqtt.io/devices/E1757.html#manual-home-assistant-configuration) to add only the 'cover' entity which allows me to have a seperate cover entity that cannot be added to a room but does show up on the main HA UI and can be controlled correctly. So it all works in practice but the Zigbee2MQTT device discovery has missed the cover device.

Way less important for me but could be related - I cannot get the IKEA TRADFRI open/close remote to pair. With the repeater and blind paired with Zigbee2MQTT:
- Hold remote next to repeater and hold remote reset button for ~10 seconds
- Wait for lights to go out
- Hold remote next to blind battery cover and hold remote reset button for ~10 seconds
- Wait for lights to go out

## What did you expect to happen
Pairing the repeater and blind would add a cover device that could be assigned to a room and operated from the main HA UI.

## How to reproduce it (minimal and precise)
Reset both repeater and blind.
Pair repeater.
Pair blind with repeater next to blind battery cover.

## Debug Info
Home Assistant OS Version:
6.1

Core Version:
core-2021.6.6

Adapter:
CC2652R Stick [Electrolama zig-a-zig-ah! (zzh!)]

Zigbee2MQTT Version:
1.18.1 commit: 1.18.1

Coordinator Type:
zStack3x0

Coordinator Revision:
20201026

FYRTUR Firmware Version:
2.3.073

TRADFRI Signal Repeater Firmware Version:
2.3.070

AitorDB, 2021-07-05:
I was facing the same issue than you with an old edge build and same adapter/coordinator/home assistant/etc. Updating to the latest edge version has solved the problem for me

JackEker, 2021-07-05:
@AitorDB I must be overlooking something as I thought I'd updated everything! What's the edge version - I'm not sure what edge refers to?

AitorDB, 2021-07-05:
@JackEker if you check the addon store there should be a Zigbee2MQTT Edge version

JackEker, 2021-07-05:
Ahhh, thanks for the quick reply! I'm currently using the mainline version of the add-on not the edge version. Does that potentially indicate a change has broken this in the mainline version and has since been fixed in the edge development build...?

Koenkk, 2021-07-05:
You need to update to z2m 1.19.0.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

JackEker, 2021-07-05:
Ohhh I hadn't seen that as I didn't realise 1.19 was a thing! Awesome I'll get on it - is it possible to migrate all devices, config etc. without having to re-pair things?

Koenkk, 2021-07-05:
Yes, just uninstall the old one and install the new addon

JackEker, 2021-07-07:
Thanks, that has worked and fixed my issue! 

As a note to upgrade from the old version and keep all devices, settings etc., I had to play around a bit with the following steps:
1. Add new repository to add-on store
2. Create new directory /config/zigbee2mqtt
3. Copy 'database.db', 'devices.yaml', 'groups.yaml' from /share/zigbee2mqtt to /config/zigbee2mqtt
4. Make a note of the existing config used by the v1.18.x Zigbee2MQTT addon
5. Uninstall v1.18.x of Zigbee2MQTT add-on
6. Remove old repository from add-on store
6. Install new v.1.20.x of Zigbee2MQTT add-on
7. Migrate any custom settings from config noted in step 3. into new Zigbee2MQTT configuration
8. Start the Zigbee2MQTT addon

closed 2021-07-07 by JackEker