Koenkk/zigbee2mqtt #7960: Problem with home assistant zigbee2mqtt and CC2531

MarcelJ28, 2021-07-06:
I use zigbe2mqtt with a CC2531 adapter with home assistant. Until today everything worked fine. Yesterday I added some extra devices and that also worked fine. This morning however nothing worked and I get errors like this in my home assistant logs;

21-07-06 07:20:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE serial hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC0X00124B001CD4B491-if00
21-07-06 07:20:32 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE usb hardware /dev/bus/usb/001/005
21-07-06 07:20:34 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-07-06 07:20:34 INFO (MainThread) [supervisor.host] Host information reload completed
21-07-06 07:20:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/001/006
21-07-06 07:20:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC0X00124B001CD4B491-if00
21-07-06 07:20:39 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.7:8099 ssl:default [Connect call failed ('172.30.33.7', 8099)]
21-07-06 07:20:42 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE serial hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC0X00124B001CD4B491-if00
21-07-06 07:20:42 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE usb hardware /dev/bus/usb/001/006
21-07-06 07:20:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/001/007
21-07-06 07:20:44 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC0X00124B001CD4B491-if00

So the USB is removed and added although I do nothing.

I already set back a backup from yesterday morning (before I added the devices) but that also does not work, same problem. 

The logs of zb2mqtt are;

at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.19.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.19.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-07-06T06_34_12_738Z-debug.log
[07:34:13] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.19.1 start /app
> node index.js
Zigbee2MQTT:error 2021-07-06 07:34:18: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-07-06 07:34:18: Failed to start zigbee
Zigbee2MQTT:error 2021-07-06 07:34:18: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-07-06 07:34:18: Exiting...
Zigbee2MQTT:error 2021-07-06 07:34:18: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.19.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.19.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-07-06T06_34_18_463Z-debug.log
[07:34:18] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.19.1 start /app
> node index.js
Zigbee2MQTT:error 2021-07-06 07:34:24: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-07-06 07:34:24: Failed to start zigbee
Zigbee2MQTT:error 2021-07-06 07:34:24: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-07-06 07:34:24: Exiting...
Zigbee2MQTT:error 2021-07-06 07:34:24: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.19.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.19.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-07-06T06_34_24_449Z-debug.log
[07:34:24] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.19.1 start /app
> node index.js
Zigbee2MQTT:error 2021-07-06 07:34:30: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-07-06 07:34:30: Failed to start zigbee
Zigbee2MQTT:error 2021-07-06 07:34:30: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-07-06 07:34:30: Exiting...
Zigbee2MQTT:error 2021-07-06 07:34:30: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.19.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.19.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-07-06T06_34_30_505Z-debug.log
[07:34:31] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.19.1 start /app
> node index.js

What could be the problem???

Sportslite, 2021-07-08:
Hi,
did you check this: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
I had nearly the same problem and solved it at the end by buying a new usb stick. If you are able to flash the firmware, try that before.

MarcelJ28, 2021-07-09:
> Hi,
> did you check this: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
> I had nearly the same problem and solved it at the end by buying a new usb stick. If you are able to flash the firmware, try that before.

Yes I did and nothing helped. But I already ordered a new CC2652RB . Because I also think that is the problem and I am not able to flash nor reset because the stick is totally sealed. And I want to let my network of devices grow some more so I think buying a new stick is the best solution. Thanks for your reply.

closed 2021-08-16 by github-actions[bot]