Koenkk/zigbee2mqtt #7962: New group implementation error: Covers / Blinds not working in all cases

jd1900, 2021-07-06:
## Problem
I'm trying the new group feature using blinds / covers.

However, I have two problems:
- I can use the Open / Close / Stop controls.
- I can't controll through Google Assistant, maybe because it uses does controls.

Individually they work fine. They are Legrand covers.

However, I can set position to 0 (closed), 100 (opened) or any other number in the group, it works.

I think of a workaround: change the open/close commands internally to use 0/100 positions.
However, it would still miss the stop command.

## Debug info
Zigbee2MQTT version: 1.20.0 commit: 1.20.0
Adapter hardware: zzh
Coordinator type: zStack3x0
Coordinator revision: 20210120
Frontend version: 0.4.29

sjorge, 2021-07-06:
I'm guessing the group extension just needs to be updated to know about states that are not ON, OFF or TOGGLE as I believe there is some logic around that to make groups appear correctly.

It should probably treat OPEN like ON, CLOSE like OFF and STOP similar to TOGGLE.

Based on the stack though, it looks like the group extension is also picking the wrong converter and is using the one for lights.

@Koenkk probably know exactly what is going wrong with the converter selection.

Koenkk, 2021-07-06:
Duplicate of #7922

closed 2021-07-06 by Koenkk

jd1900, 2021-07-06:
> Duplicate of #7922

In that issue you say it is fixed and that it may be a problem of not only having covers.
However, I only have covers and still have the problem.

Koenkk, 2021-07-06:
@jd1900 then the issue is not fixed yet, please post your data/databse.db entry of your group in #7922

jd1900, 2021-07-06:
> @jd1900 then the issue is not fixed yet, please post your data/databse.db entry of your group in #7922

Where can I send it to you?
And..., in which version did you solve it?
I have the latest from this repo: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/tree/master/zigbee2mqtt
Maybe not the same as this one?

Koenkk, 2021-07-07:
@jd1900 you can put it there (just comment). You need to be on the  latest-dev. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)