Koenkk/zigbee2mqtt #7975: Schneider Electric PowerTag

OsteHovel, 2021-07-07:
## Information about the device + link
Schneider Electric PowerTag Monoconnect 63A 1P+N top position

This is energy sensor utilizing Zigbee 3 Green Power

https://www.se.com/ww/en/product/A9MEM1521/energy-sensor%2C-powertag-monoconnect-63a-1p%2Bn-top-position/

## data/database.db entry of the device
This gets added during pairing, but it actually does not really pair completely more info below
```JSON
{
  "id": 3,
  "type": "GreenPower",
  "ieeeAddr": "0x00000000e20765cb",
  "nwkAddr": 26059,
  "manufId": null,
  "modelId": "GreenPower_254",
  "epList": [
    242
  ],
  "endpoints": {
    "242": {
      "epId": 242,
      "inClusterList": [],
      "outClusterList": [],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1625428765211
}
```
| Info | Info |
| --- | --- |
| OS | Windows |
| Zigbee2mqtt | Latest on master branch |
| Zigbee-herdsman | Latest on master branch |
| Cordinator | CC2531 |
| Channel | 15  |
| Network key | 0c9f45818f14a953d0993f2304015e10 |

I have tried with both using a [Philips Hue Light strip](https://www.zigbee2mqtt.io/devices/9290022890.html) and using a [IKEA E1746 TRADFRI signal repeater](https://www.zigbee2mqtt.io/devices/E1746.html)

Utilizing the Philips Hue Light strip the light strip does not reply to the channel request sent out by the PowerTag.
By utilizing a IKEA signal repeater that actually replies to the channel requests and gets it on the correct channel and sending out a commissioning notification, see [Channel request and configuration.pcapng](https://cloud.ostsoft.com/s/MsAyT9MmgBYejBz).

I had to do some changes to Zigbee-herdsman to get it actually decode the packet correctly
buffaloZcl.ts - Added a check for key present during payload
```JS
private readGdpFrame(options: TsType.Options): Gdp | {raw: Buffer} | Record<string, never> {
    // Commisioning
    if (options.payload.commandID === 224) {
        const deviceID = this.readUInt8();
        const options1 = this.readUInt8();
        const extendedOptions = this.readUInt8();

        const gdpKeyPresent = (extendedOptions & 0x20) != 0;
        if (gdpKeyPresent) {
            return {
                deviceID: deviceID,
                options: options1,
                extendedOptions: extendedOptions,
                securityKey: this.readBuffer(16),
                keyMic: this.readUInt32(),
                outgoingCounter: this.readUInt32(),
            };
        } else {
            return {
                deviceID: deviceID,
                options: options1,
                extendedOptions: extendedOptions,
                securityKey: null,
                keyMic: null,
                outgoingCounter: this.readUInt32(),
            };
        }
    } else if (this.position != this.buffer.length) {
        return {raw: this.buffer.slice(this.position)};
    } else {
        return {};
    }
}
```

greenPower.ts
```JS
public async onZclGreenPowerData(dataPayload: AdapterEvents.ZclDataPayload): Promise<void> {
    if (dataPayload.frame.Payload.commandID === 224 && typeof dataPayload.address === 'number') {

        const keyPresent = (dataPayload.frame.Payload.extendedOptions & 0x20) != 0;
        if (keyPresent) {
            const key = this.encryptSecurityKey(
                dataPayload.frame.Payload.srcID, dataPayload.frame.Payload.commandFrame.securityKey
            );

            const payload = {
                options: 0b1_1_100_10_1_0_10_0_1_000,
                srcID: dataPayload.frame.Payload.srcID,
                sinkGroupID: this.adapter.greenPowerGroup,
                deviceID: dataPayload.frame.Payload.commandFrame.deviceID,
                frameCounter: dataPayload.frame.Payload.commandFrame.outgoingCounter,
                gpdKey: [...key],
            };

            const frame = Zcl.ZclFrame.create(
                Zcl.FrameType.SPECIFIC, Zcl.Direction.SERVER_TO_CLIENT, true,
                null, ZclTransactionSequenceNumber.next(), 'pairing', 33, payload
            );

            await this.adapter.sendZclFrameToAll(242, frame, 242);
        } else {
            // Key not present security
            const key = Buffer.from([0xc9, 0x79, 0xd5, 0x22, 0x9c, 0xb9, 0xe9,
                0xd2, 0x8d, 0x5d, 0xa6, 0x99, 0xf8, 0xca, 0x02, 0x97]);

            const options = 0b1_1_010_11_1_1_10_0_1_000;
            const payload = {
                options: options,
                srcID: dataPayload.frame.Payload.srcID,
                sinkGroupID: this.adapter.greenPowerGroup,
                deviceID: dataPayload.frame.Payload.commandFrame.deviceID,
                frameCounter: dataPayload.frame.Payload.commandFrame.outgoingCounter,
                gpdKey: [...key],
            };

            const frame = Zcl.ZclFrame.create(
                Zcl.FrameType.SPECIFIC, Zcl.Direction.SERVER_TO_CLIENT, true,
                null, ZclTransactionSequenceNumber.next(), 'pairing', 33, payload
            );

            await this.adapter.sendZclFrameToAll(242, frame, 242);
        }

        const eventData: GreenPowerDeviceJoinedPayload = {
            sourceID: dataPayload.frame.Payload.srcID,
            deviceID: dataPayload.frame.Payload.commandFrame.deviceID,
            networkAddress: dataPayload.address,
        };

        this.emit(GreenPowerEvents.deviceJoined, eventData);
    }
}
```
I have flipped around with the options to try to get any response.

[PowerTag Commissioning.pcapng](https://cloud.ostsoft.com/s/MsAyT9MmgBYejBz) is a captured when the PowerTag is being commissioned by a [Schnieder Electric Acti 9 Smartlink SI B](https://www.se.com/ww/en/product/A9XMZA08/acti9-smartlink-si-b---modbus-tcp-ip-and-wireless-communication-module/)

[Zigbee2mqtt pairing.pcapng](https://cloud.ostsoft.com/s/MsAyT9MmgBYejBz) is a capture of what happening when I am trying to pair with zigbee2mqtt
I am thinking it is because the IKEA Signal Repeater (and for that matter the Philips Hue) is repeating the Green Power using protocol version 2 instead of 3, or I am just formatting the packets wrong.

OsteHovel, 2021-07-08:
I realized when I read this post again that I should include some screen captures so people do not need to click the links and download the capture files to see what I am talking about

So the PowerTag send out this Commission packet after getting the Channel Configuration
[image]

# Using Acti 9 Smartlink SI B

Then using Acti 9 Smartlink SI B it will reply Commission Reply
[image]

And then the device replies Success and everyone is happy
[image]

# Using Zigbee2mqtt

Using zigbee2mqtt the Commission packet from the PowerTag gets translated/packet into a Zigbee packet:
[image]

And I try to reply to that packet:
[image]

And nothing happens, it just continuing to send out Commission packet like nothing happened.
It kind of make sense to me as I cannot see any packet getting translated to Green Power and sent out by the proxy(Signal Repeater or Hue light strip)
Its like it is just ignored

NicolaiPetri, 2021-07-08:
I will follow here.
I have the almost same devices, just 3p+0 model and I tried adding a Hue White bulb to my network but I get absolutely no pairing from the device. I'm using latest release from July 1st. 
Do I need to upgrade to latest unstable version ? Or what can I do to get at least pairing to work ?
I don't mind testing stuff if you need it.

NicolaiPetri, 2021-07-08:
**After enabling further debugging this is also present in the logs (snippet), I'm not sure it's related but endpoint 242 seems to be right:**

Zigbee2MQTT:debug 2021-07-08 17:51:21: Received Zigbee message from 'Repeater1', type 'raw', cluster 'greenPower', data '{"data":[9,73,6,8,117,30,14,0,0,0,0,243,1,29],"type":"Buffer"}' from endpoint 242 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,39,68,129,0,0,33,0,7,0,242,242,1,60,0,8,12,202,0,0,19,17,72,4,8,8,0,0,0,0,87,0,0,0,227,1,94,117,30,207,117,30,29,64] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 39 - 2 - 4 - 129 - [0,0,33,0,7,0,242,242,1,60,0,8,12,202,0,0,19,17,72,4,8,8,0,0,0,0,87,0,0,0,227,1,94,117,30,207,117,30,29] - 64 +12ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":7,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":60,"securityuse":0,"timestamp":13241352,"transseqnumber":0,"len":19,"data":{"type":"Buffer","data":[17,72,4,8,8,0,0,0,0,87,0,0,0,227,1,94,117,30,207]}} +23ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":72,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":2056,"srcID":0,"frameCounter":87,"commandID":227,"payloadSize":1,"commandFrame":{"raw":{"type":"Buffer","data":[94,117,30,207]}}},"Command":{"ID":4,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009}],"name":"commisioningNotification"}},"address":7,"endpoint":242,"linkquality":60,"groupID":0,"wasBroadcast":true,"destinationEndpoint":242}' +25ms
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '7', skipping... +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +24ms
Zigbee2MQTT:debug 2021-07-08 17:51:21: No converter available for 'E1746' with cluster 'greenPower' and type 'raw' and data '{"data":[9,73,6,8,117,30,14,0,0,0,0,243,1,29],"type":"Buffer"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,50,68,129,0,0,33,0,121,11,242,242,1,60,0,7,50,203,0,0,30,17,74,4,8,8,121,11,2,226,93,0] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,50,68,129,0,0,33,0,121,11,242,242,1,60,0,7,50,203,0,0,30,17,74,4,8,8,121,11,2,226,93,0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,224,12,254,231,131,121,18,0,0,3,94,16,93,67,117,30,208,117,30,29,106] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,50,68,129,0,0,33,0,121,11,242,242,1,60,0,7,50,203,0,0,30,17,74,4,8,8,121,11,2,226,93,0,0,0,224,12,254,231,131,121,18,0,0,3,94,16,93,67,117,30,208,117,30,29,106] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 50 - 2 - 4 - 129 - [0,0,33,0,121,11,242,242,1,60,0,7,50,203,0,0,30,17,74,4,8,8,121,11,2,226,93,0,0,0,224,12,254,231,131,121,18,0,0,3,94,16,93,67,117,30,208,117,30,29] - 106 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":2937,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":60,"securityuse":0,"timestamp":13316615,"transseqnumber":0,"len":30,"data":{"type":"Buffer","data":[17,74,4,8,8,121,11,2,226,93,0,0,0,224,12,254,231,131,121,18,0,0,3,94,16,93,67,117,30,208]}} +1s
  zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":33,"data":{"type":"Buffer","data":[17,74,4,8,8,121,11,2,226,93,0,0,0,224,12,254,231,131,121,18,0,0,3,94,16,93,67,117,30,208]},"address":2937,"endpoint":242,"linkquality":60,"groupID":0,"wasBroadcast":true,"destinationEndpoint":242}' +1s
  zigbee-herdsman:controller:log 'raw' data is from unknown device with address '2937', skipping... +1ms

rindlerblabla, 2021-07-23:
Any updates on this? Don't have those devices but would be interested in getting them.

rindlerblabla, 2021-08-23:
I am still interested in thise devices but don't have any myself...

OsteHovel, 2021-08-23:
I ended up purchasing the gateway from Schneider Electric, but it would still be nice to be able to add to them to Zigbee2mqtt.
I am at a standstill on what to do or what I can test to make it work.
I have 10 of them

rindlerblabla, 2021-08-23:
Cool! That's beyond my knowledge, but have you tried sniffing the network when you connect the power tag to the Schneider gateway?

OsteHovel, 2021-08-24:
> 
> 
> Cool! That's beyond my knowledge, but have you tried sniffing the network when you connect the power tag to the Schneider gateway?

Yepp, I actually borrowed a gateway before I purchased it, to try to avoid using the gateway at all.
The network capture is linked on the bottom of the first post else you find it [here](https://cloud.ostsoft.com/s/MsAyT9MmgBYejBz)

rindlerblabla, 2021-08-24:
Do you think it would do any difference if you paired it through some other green power supported device?

OsteHovel, 2021-09-24:
I did try with both a Philips Hue Light Strip and a Ikea Signal Repeater.

rindlerblabla, 2021-09-24:
Sorry for tagging, but @sjorge or @Koenkk could you help out in some way how to pair these green power sensors? Guess that most info is in the second post in this thread.

sjorge, 2021-09-24:
> Sorry for tagging, but @sjorge or @Koenkk could you help out in some way how to pair these green power sensors? Guess that most info is in the second post in this thread.

Best wait for Koen, I have no experience with GP devices. A bit surprising an energy meter is using GP too. I'd have not expected that as GP is rather limited, but that is mostly because they are supposed to harvest energy from the user actuating the buttons, at least how I understood it.

Koenkk, 2021-09-24:
Can you check wether this device can be paired to a Hue bridge?

filips, 2021-11-30:
I managed to get some progress on pairing the Schneider PowerTag devices.

I am still working on cleaning up the code I have, but the main element necessary to complete the pairing is implementing the GP Commisioning Reply (0xF0) command. We probably also need to respond to the GP Channel Request (0xE3). Currently I had an IKEA Float lamp sporadically shouting out "Channel 16" to the device, which is not at all what I'm using, so need to figure out how to get that one to be quiet during commisioning.

Here you see what I'm sniffing after the commisioning flow done by zigbee2mqtt:

[image]

I still need to fix up the GP Pairing request to make the GPP forward the GP frames to me, but that should be relatively simple.

Baftalie, 2022-01-17:
@filips Any update on pairing powertag devices to zigbee2mqtt?

Twiistz, 2022-01-30:
Any update on this?

filips, 2022-03-01:
I'm sorry, been really busy with work lately. My current WIP code is very rough and hard-coded, so I don't know it will serve much purpose sharing it, but I will gladly elaborate on my findings.

Essentially what needs to be done is the specific handling of more Green Power commands. Currently in greenPower.ts in onZclGreenPowerData only the GP Commissioning command is handled.

This works for some very simple Green Power devices, where you simply need to send a pairing packet to a GP Proxy, instructing it to forward the one-way data. For an RX enabled device like Schneider PowerTag, the flow is somewhat more complicated. 

The commands that need responses are, to my understanding, these:

```js
switch(dataPayload.frame.Payload.commandID) {
    case undefined:
        // GP Undefined Command
        break;
    case 0xE0:
        // GP Commissioning
        // You (Sink) should respond with GP Commissioning Reply
        break
    case 0xE2:
        // GP Success
        // You should send a pairing command to set up GP Proxy to forward messages
        break;
    case 0xE3:
        // GP Channel Request
        // You should reply with a GP Channel Reply 
        break
    case 0xA1:
        // GP Manufacturer-specific Attribute Reporting
        // This is the actual data received
        break;
    default:
        // Unhandled GP Command
}
```

So in this case, the flow will be something like this:

- GP Device sends GP Channel Request to get Zigbee chanel
- GP Sink (you) send a GP Channel Reply to let the GP Device know which channel is used
- GP Device sends GP Commissioning 
- GP Sink (you) send GP Commissioning Reply, instructing GP Device of all settings
- GP Device will send GP Success if successful
- GP Sink (you) will now send a pairing packet, configuring GP Proxy to mediate the data sent from GP Device over the Zigbee Network

Inrego, 2022-03-15:
I recently started getting problems with wiser2mqtt, unable to authenticate to it. So I'm back to hoping for progress on this.
@Koenkk can this issue be reopened? I'm not sure why it was closed in the first case.

Inrego, 2022-03-18:
Hmm.. Even though these powertags are paired to their own bridge, they are still joining my Z2M network and spamming unsupported messages:
[image]
Even though I remove it and block it from re-joining, it somehow joins again after a little while.
While joined, it's spamming errors in the Z2M log about every second.

So all in all, looking forward to having these devices properly joined on my Z2M network instead :)

Inrego, 2022-03-21:
https://zigbeealliance.org/wp-content/uploads/2019/11/docs-09-5499-26-batt-zigbee-green-power-specification.pdf

Here's the link to the GP specification. In case it's needed by anyone who didn't find it already.
I currently run Z2M in docker, so I'm not sure if/how I could go about setting up a developing/testing environment. I don't have extra zigbee coordinators, and I don't want to mess up my current zigbee network with 100+ devices in it.

Also I don't have much knowledge on the technical side of Zigbee, but that wouldn't hold me back from getting into it, as that's just something I can learn :)

But I am still rooting for @filips

filips, 2022-03-23:
I have had some progress with the PowerTags.

One slightly odd thing is that Schneider has implemented a form of ACK'ing for received Manufacturer-specific Attribute Reports. If this ACK is not sent, one will only see some boring model number reports - not the actual readings.

Just moments ago I got everything plumbed through, and got this reading in the Z2M web UI:

[image]

There are a number of outstanding items still remaining, including:
- Cleanup of hardcoded options fields/command structures/etc.
- Investigation into using encrypted connection between GP Proxy and the powertags. In my initial attempts I did not manage to pair the GP Device to a GP Proxy in a way that allowed GP Response packets (in this case the strange ACK) to be encrypted by the proxy before sending it to the powertag.
- Implementation of remaining reported values

I'm not quite sure where it would make most sense to put the code that sends the manufacturer specific ACK frame out.. Right now I just have it in greenPower.ts, but that's definitely not right.

Inrego, 2022-03-24:
Maybe @Koenkk can weigh in on how to structure the code?

smulle48, 2022-04-02:
I'm looking forward for the development, as I'm planing to buy one for my BEV Charger 😁

Koenkk, 2022-04-02:
@filips where did you get stuck?

smulle48, 2022-04-04:
This one looks quite interesting.

Is says Harmony Hub in the description. Maybe that will connect easier
[image]
[image]

Baftalie, 2022-04-04:
Harmony Hub is also made by schneider electric and afaik all powertag devices work with it (never tried it though)

smulle48, 2022-04-04:
> Harmony Hub is also made by schneider electric and afaik all powertag devices work with it (never tried it though)

It only says Harmony Hub for the models with cable :-)

Baftalie, 2022-04-04:
Thats interesting, any idea what’s different about those? At least from the outside I cannot see any difference.

smulle48, 2022-04-04:
> Thats interesting, any idea what’s different about those? At least from the outside I cannot see any difference.
> 
> 

I have no idea, but I'll try to order one :-)

filips, 2022-04-05:
The PowerTag with the external power connections is the one I have been using for development. AFAIK nothing is different between that one and the models with built-in contacts. 

Having it externally powered can make sense for development purposes, as toggling power after communication times out is the way to initiate the commissioning flow.

smulle48, 2022-04-05:
> The PowerTag with the external power connections is the one I have been using for development. AFAIK nothing is different between that one and the models with built-in contacts.
>
>
>
> Having it [...]

Just found out that Harmony Hub referees to Schneider's own data hub called Harmony Hub 😁

verydrunk, 2022-04-05:
R9 is for the residential user, A9 (Acti9) is for industrial and commercial buildings and is mostly used with modbus tcp ip for that buildings "superior control system" (is it the right English word for it?)

smulle48, 2022-04-09:
Looking forward to testing it :-)

NicolaiPetri, 2022-04-09:
Do anyone know how to re-pair a device that is already connected to a Scheider gateway ?

verydrunk, 2022-04-09:
I guess u go in eSetup and remove it from the gateway to realise it

filips, 2022-04-09:
Either that, or you can turn off the gateway and wait until the led on the powertag blinks quickly red (~2 times per second). Then power cycling the powertag initiates commissioning

smulle48, 2022-04-12:
@filips is there a specific way the power has to run through the powertag? Can't see any direction markings on it

smulle48, 2022-04-12:
> > @filips is there a specific way the power has to run through the powertag? Can't see any direction markings on it
> 
> I don't believe so, as long as N and the phases are connected correctly.

I found out it has to go from "top to bottom"
[image]

Btw. how often does it report? and is there a way (as for now) to change that time?

filips, 2022-04-12:
Ah ok, that's also the way I have mine wired.

It reports around every five seconds by default. I've seen some GP Write Attribute commands from the wiser gateway that may configure this, but it is not something I've looked into.

filips, 2022-04-19:
By the way, this is the command i suspect could have something to do with the reporting interval:

[image]

avsantos76, 2022-05-16:
Maybe this could help in the solution, after all the diference is the number o phases that each device can support:
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/A9MEM1570.md

edenhome, 2022-05-21:
I have been testing Powertag with direct link to zigbee2mqtt, model R9M70 works perfect, model R9M41links but don’t show any values

smulle48, 2022-05-21:
> I have been testing Powertag with direct link to zigbee2mqtt, model R9M70 works perfect, model R9M41links but don’t show any values

Try to post your debug log when adding it to the network. Maybe we can figure out something to make it work 😃

edenhome, 2022-05-21:
> 
I assume this it what you mean,, I very unfamiliar with zigbee2mqtt

[image]

smulle48, 2022-05-22:
> I assume this it what you mean,, I very unfamiliar with zigbee2mqtt
> 
> [image]

@filips is this because of an different modelID ?

filips, 2022-05-23:
The modelID should not have much influence.. I have successfully paired a R9M42, so I would expect the R9M41 to be the same, as it it just the mirrored version as far as I can tell.

When looking at the debug output I do notice something weird: the encryption key provided by the powertag is pure 0's. I haven't seen that before, and it makes me wonder whether there is some issue with the Proxy used. I usually allow pairing only to one router in the Z2M web interface, so I can be sure which one I'm pairing to. Could you confirm how you pair the powertags, and mybe try to do it the same way for both, only allowing pairing to a specific router?

edenhome, 2022-05-24:
> The modelID should not have much influence.. I have successfully paired a R9M42, so I would expect the R9M41 to be the same, as it it just the mirrored version as far as I can tell.
> 
> When looking at the debug output I do notice something weird: the encryption key provided by the powertag is pure 0's. I haven't seen that before, and it makes me wonder whether there is some issue with the Proxy used. I usually allow pairing only to one router in the Z2M web interface, so I can be sure which one I'm pairing to. Could you confirm how you pair the powertags, and maybe try to do it the same way for both, only allowing pairing to a specific router?

Agree R9M42 and R9M41 should be the same.
It was located next to the sonoff cordinator when i first tried,, now it has been removed from the switchboard.
I tried to pair with just one router,,, tried both a HUE and a Nordtronic, result always the same

when conecting power is blink red, then after paring it will keep blink orange,,, and after a while it go back to fast red blink,,, same as it never recived an ack

Inrego, 2022-05-27:
I'm in the process of changing over to a slae.sh dongle. The PowerTags are working great! I'm having some issues with some light bulbs on this dongle, but that's unrelated to this.

Telk0, 2022-05-29:
Which powertag you have? i have R9M22 1P+N and its not putting out any data.

Inrego, 2022-06-17:
I ended up having just the PowerTags paired with my slae.sh dongle, and then the rest of my zigbee network on the Conbee (due to some of my lights not working on the slae.sh dongle).

It was working great, for a week or so. Then I stopped getting any data from the PowerTags. I tried resetting them, and they won't even pair with the slae.sh dongle.
I had one of them paired to my conbee for a while (about an hour) and then it stopped working. The other one would pair, but not send any data. It's all a bit unstable, really.

Inrego, 2022-06-17:
Now they're both paired to my conbee dongle, but getting no data from any of them

smulle48, 2022-06-26:
@filips I've had it running for a while, but I'm getting some unexpected behavior.

I've attached the log file from zigbee2mqtt.

Though I get an mitt message about every 5 second, the value from the PowerTag does not update every 5 seconds. I can't seem to find a clear pattern on how/when it update the value.

Is that anything you have experienced?

[log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/8986761/log2.txt)

verydrunk, 2022-07-05:
I would like to beg for some help, I have R9M20/M40 and a sonoff z3.0 usb dongle plus. And I can't even get them to pair. Do I need something else? IKEA repeater, lamp what ever? Please help.
I'm getting desperate as Schneider desided to ruin the little function the old app still had.

filips, 2022-07-05:
> I would like to beg for some help, I have R9M20 and R9M40 and a sonoff z3.0 usb dongle plus. And I can't get none of them to pair. Do I need something else? IKEA repeater, lamp what ever? Please help. I'm getting desperate as Schneider desided to ruin the little function the old app still had.

To pair a green power device you need a device to act as a green power proxy, which is always powered. It can be a repeater, lamp or smart plug.  I have had it working with both IKEA and Philips devices, but when I used an IKEA repeater i stopped getting some updates after a week or so. For the past two months I have been using a Philips Hue smart plug instead, which I have had no issues with.

filips, 2022-07-05:
> @filips I've had it running for a while, but I'm getting some unexpected behavior.
>
> I've attached the log file from zigbee2mqtt.
>
> Though I get an mitt message about every 5 second, the value from [...]

It looks like the values are not changing from message to message, I'm specifically looking at the "energy" field. I have experienced this in some form, where the powertag is just continuously retransmitting the last message, because it is not receiving the ACK package that it needs to get. You will probably be able to see the powertag led flashing red rapidly when this happens. 

What device are you using as the proxy for pairing? I have had good success with philips devices, and somewhat less with IKEA repeaters. I've had something like what you're seeing happen after a week or so with an ikea repeater.

smulle48, 2022-07-05:
> It looks like the values are not changing from message to message, I'm specifically looking at the "energy" field. I have experienced this in some form, where the powertag is just continuously [...]

Unfortunately I only got ikea bulbs.

The weird thing is that if I take power(w) and run it through a Reimann in HA, I get the correct readings

verydrunk, 2022-07-09:
> Info 2022-07-09 12:00:15Device '0x00000000e2032ba2' joined
Info 2022-07-09 12:00:15MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00000000e2032ba2","ieee_address":"0x00000000e2032ba2"},"type":"device_joined"}'
Info 2022-07-09 12:00:15Successfully interviewed '0x00000000e2032ba2', device has successfully been paired
Warning 2022-07-09 12:00:15Device '0x00000000e2032ba2' with Zigbee model 'GreenPower_84' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2022-07-09 12:00:15MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00000000e2032ba2","ieee_address":"0x00000000e2032ba2","status":"successful","supported":false},"type":"device_interview"}'
Debug 2022-07-09 12:00:15Received Zigbee message from '0x00000000e2032ba2', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"applicationInfo":0,"deviceID":94,"extendedOptions":188,"gpdClientClusters":{"data":[],"type":"Buffer"},"gpdCommandIdList":{"data":[],"type":"Buffer"},"gpdServerClusters":{"data":[],"type":"Buffer"},"keyMic":0,"manufacturerID":0,"modelID":0,"numClientClusters":0,"numGdpCommands":0,"numServerClusters":0,"options":216,"outgoingCounter":1420269332,"securityKey":{"data":[86,193,29,233,43,52,15,108,72,216,253,245,131,102,39,143],"type":"Buffer"}},"commandID":224,"frameCounter":7183602,"gppGddLink":157,"gppNwkAddr":35576,"options":2616,"payloadSize":44,"srcID":3791858594}' from endpoint 242 with groupID 0
Warning 2022-07-09 12:00:15Received message from unsupported device with Zigbee model 'GreenPower_84' and manufacturer name 'undefined'
Warning 2022-07-09 12:00:15Please see: 

> https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html``

verydrunk, 2022-07-09:
Still having trouble with R9M20/M40
tried connecting to Hue bulb LWA001, Hue Smart Plug ED8597, Hue GU10 LCT003.
I have a butt load of different hue lights and guess i continue to switch router.
But any help would be appreciated

> Info 2022-07-09 12:00:15Device '0x00000000e2032ba2' joined
> Info 2022-07-09 12:00:15MQTT publish: topic 'zigbee2mqtt/bridge/event', payload [...]

Inrego, 2022-08-08:
> To pair a green power device you need a device to act as a green power proxy, which is always powered. It can be a repeater, lamp or smart plug. I have had it working with both IKEA and Philips [...]

After your suggestion, I finally seem to have a working, stable setup. Here's my setup to actually get it working:
My primary ZigBee network is on a ConBee II (some of my lights were very unstable on the slae.sh dongle).
I added a slae.sh dongle to run a seperate zigbee network for these PowerTags. I added a single Philips Hue bulb to this network (the PowerTags didn't work without this step).

So yeah, I'm running 2 seperate zigbee networks. One for the PowerTags, and one for everything else.
The PowerTags don't work properly on the ConBee II coordinator, and some other devices don't work properly on the slae.sh coordinator. So I guess that's the only way, unfortunately. With that said, it's not that big of a hassle since it just works now.

ciedema, 2022-09-05:
I was able to connect to a Clipsal 4RCBEM2B, which I think is an Australian version of an R9M60

[130 more replies]

closed 2023-05-05 by github-actions[bot]