Koenkk/zigbee2mqtt #7976: IKEA bulbs need 2 x light.turn_X / light.toggle calls from Home Assistant to actually turn on or off

artkrz, 2021-07-07:
## What happened

Since couple of weeks some lights aren't working as expected. I have Ikea bulbs e27 and I have to send turn_on twice from home assistant for state of the bulb to change to ON, same with turn_off and toggle. These bulbs work OK from zigbee2mqtt dashboard. I also have some groups with gu10 bulbs which I had to add to home assistant as mqtt light and these work OK, no problems at all. This problem stared a while ago on 1.19.x, I'm using HA core-2021.6.6

## What did you expect to happen

When I use light.turn_on service call in home assistant bulb should turn on and change state to on in home assistant

## How to reproduce it (minimal and precise)

Send light.turn_on to single bulb (not a group)

## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: Conbee II
Adapter firmware version: {'meta': {'maintrel': 0, 'majorrel': 38, 'minorrel': 88, 'product': 0, 'revision': '0x26580700', 'transportrev': 0}, 'type': 'ConBee2/RaspBee2'

io53, 2021-07-07:
I had this issue, debugged it a bunch. It turned out that all I had to do was change the color temperature once for all lights once. 
Try that, hope it helps!

artkrz, 2021-07-07:
> I had this issue, debugged it a bunch. It turned out that all I had to do was change the color temperature once for all lights once. 
> 
> Try that, hope it helps!

Thanks!

closed 2021-07-07 by Koenkk