Koenkk/zigbee2mqtt #7977: TS0601_thermostat TuYa No converter available for 'get'

WojciechWitos, 2021-07-07:
## What happened
Throught MQTT Explorer function zigbee2mqtt/friendly_name/get doesn't work with any posibble things in documentation. I tried almost every parameter which is possible to get. The problem is that I get on the web page answear is like: "No converter available for 'get' 'window_detection' ()".
Also the problem is that the state of device doesn't update on the server without sending any 'Set', device doesn't send his status even if it has changed. 

## What did you expect to happen
Device after sending him a topic zigbee2mqtt/friendly_name/get a payload for example {"window_detection":""} I expect the value will be published in zigbee2mqtt/friendly_name/get. Also I expect that the device will send his state by hisself once per some peroid of time. 
At the bridge in topic logging error 
{"level":"error","message":"No converter available for 'get' 'window_detection' ()"}
{"level":"debug","message":"Received MQTT message on 'zigbee2mqtt/term2/get' with data '{\"window_detection\":\"\"}'"}

## How to reproduce it (minimal and precise)
Publish at the topic zigbee2mqtt/friendly_name/get any correct payload.

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2531
Adapter firmware version:

WojciechWitos, 2021-07-16:
I explored a bit further. I checked the manufacuterID and it's:  _TZE200_cwnjrr72
On TuYa gateway it was recognised as HY368RT model. Could it be a problem? I couldn't find this model in whiteLabel of Tuya TS0601_thermostat, there is just HY368 but no HY368RT. 

How can I add this thing to support "get" method, when i'm trying to post /zigbe2mqtt/friendlyname/get? Because there this problem occur. Also the same problem appears when i'm trying to 'set' some parameters which aren't avaliable for set method, but this I understand why it doesn't work.

Koenkk, 2021-07-16:
Reading these properties is not supported. Fixed the docs now.

closed 2021-07-16 by Koenkk