Koenkk/zigbee2mqtt #7981: BAC-002-ALZB - full support heating, cooling, ventilation

Prim0z, 2021-07-08:
BAC-002-ALZB thermostat supports heating, cooling and ventilation. With current device support #6761  you can only control cooling. For full support we also need heating and ventilation.

At current stage we are missing mode setpoint (heating/cooling/ventilation) and fan speed setpoint (1/2/3/auto).

Is anyone willing to implement this futures? I don't have deep enough technical knowledge about zigbee2mqtt to do it myself, but I can provide data logs and testing resources. I can also send you a BAC-002-ALZB thermostat for testing and development. No need to return it, you can keep it.

Prim0z, 2021-07-13:
I created configuration to support:
- state: OFF, ON
- system_mode: cool, heat, fan_only
- fan_mode: low, medium, high, auto

If you want to include it in master branch, feel free to do so.

Instructions to include it as custom device:

1. Currently you need a DEV branch.
2. Stop zigbee2mqtt
3. Create file zigbee2mqtt/data/BAC-002-ALZB.js
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse'],
	convert: (model, msg, publish, options, meta) => {
		const dp = msg.data.dp;
		const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
[... 109 lines cut ...]
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
            exposes.climate()
				.withSetpoint('occupied_heating_setpoint', 5, 35, 1, ea.STATE_SET)
				.withSetpoint('occupied_cooling_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
				.withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
				.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
				.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program'])
				.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },
];

```

4. Include external convertor to configuation.yaml
```
external_converters:
  - BAC-002-ALZB.js
```
5. Start zigbee2mqtt

closed 2021-08-21 by github-actions[bot]

mediaworld2007, 2021-12-12:
hi i am trying this procedure to add heating and fan features ..but it's doesn't work .. is there way to see this features integrated with z2m directly ?
@Prim0z  could you help me with manual integration ? i have done all steps ( and i hasn't error ) , but i can't see additional features in zigbee2mqtt edge activated . 

my system :   Versione Zigbee2MQTT 1.21.1-dev commit: 5829663 - HKGK BAC-002-ALZB - Tipo Coordinator zStack3x0
Revisione Coordinator 20210708 Versione frontend 0.6.8

regards 

 Luigi

Bandit13, 2021-12-16:
> I created configuration to support:
> 
> * state: OFF, ON
> * system_mode: cool, heat, fan_only
> * fan_mode: low, medium, high, auto
> 

@Prim0z  Thanks a lot!!! You save my day!

mediaworld2007, 2022-01-06:
> hi i am trying this procedure to add heating and fan features ..but it's doesn't work .. is there way to see this features integrated with z2m directly ? @Prim0z could you help me with manual [...]

I had solved ( issue were space in the file.js ) . But from today it doesn't work anymore... has been any update that have make this method not working ? regards 
Luigi

mediaworld2007, 2022-01-06:
> > I created configuration to support:
> > 
> > * state: OFF, ON
> > * system_mode: cool, heat, fan_only
> > * fan_mode: low, medium, high, auto
> 
> @Prim0z Thanks a lot!!! You save my day!

Hi Bandit, is this working for you at today ?

regards 
Luigi

Bandit13, 2022-01-13:
Yes! It works!

madspoulsen, 2022-06-09:
This should be merged into the release. Natively in Z2M the device is supported, but Fan speed and mode is not exposed.

madspoulsen, 2022-06-12:
hm, I set up the external converter and can now control all objects of the thermostat, but the feedback isn't being read. Values not updating.
I get in logs this:
Debug 2022-06-12 23:01:24No converter available for 'BAC-002-ALZB' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,0,25],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4666}'
Debug 2022-06-12 23:01:24Received Zigbee message from 'Living Room A/C', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,25],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4922}' from endpoint 1 with groupID 0

So it seems that the from zigbee is not converted properly.

Can this be because i'm not in DEV branch?

dilayn, 2022-06-22:
> hm, I set up the external converter and can now control all objects of the thermostat, but the feedback isn't being read. Values not updating. I get in logs this: Debug 2022-06-12 23:01:24No [...]

I have the same issue and I'm in DEV branch, did you manage to fix it?

madspoulsen, 2022-06-22:
Dilayn, I'm using this currently. It supports the feature thermostat to a useable extend, with feedback.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
[... 90 lines cut ...]
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series thermostat',
        fromZigbee: [fz_moes_thermostat_custom],
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
            exposes.climate()
				.withSetpoint('occupied_heating_setpoint', 16, 35, 1, ea.STATE_SET)
				.withSetpoint('occupied_cooling_setpoint', 16, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
				.withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
				.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
				.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program'])
				.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },
];

dilayn, 2022-06-22:
thank you,

try to upload the file as attachment

madspoulsen, 2022-06-22:
[BAC-002-ALZB_converter.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/8957168/BAC-002-ALZB_converter.js.txt)

Ok, here it is, rename to .js

madspoulsen, 2022-06-22:
in the Dashboard thermostat-view, it shows 2 temperature adjustments. For me it is the right one that adjusts the thermostat... :S

dilayn, 2022-06-22:
was hoping this would fix the deadzone functionality, sadly not :')

dilayn, 2022-07-01:
[BAC-002-ALZB.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9030501/BAC-002-ALZB.js.txt)

Use the file above or the code below as converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
[... 108 lines cut ...]
				.withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
				.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
				.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program'])
				.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },
];
```

And use the yaml below for dashboard card. (insert your device's friendly name)

```
type: grid
square: false
columns: 1
cards:
  - type: entities
    entities:
      - entity: switch.<device friendly name>
    title: Sal AC
  - type: thermostat
    entity: climate.<device friendly name>
```

c0dr1ver, 2022-07-01:
> [BAC-002-ALZB.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9030501/BAC-002-ALZB.js.txt)
> 
> Use the file above or the code below as converter:
> 

THX MAN!

frodix, 2023-01-15:
Hello,
I have recently installed this thermostat.
I have followed the steps but I get the following error when starting zigbee2mqtt:

/app/data/extension/externally-loaded.js:183
				.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
				 ^
TypeError: exposes.climate(...).withSetpoint(...).withSetpoint(...).withLocalTemperature(...).withLocalTemperatureCalibration(...).withSystemMode(...).withFanMode(...).withRunningState(...).withPreset(...).withSensor is not a function
    at /app/data/extension/externally-loaded.js:183:6
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

I would appreciate if someone could help me.
Forgive my English, but I use Google translate from Spanish.

simguth, 2023-05-18:
Hi all,

@Rick5o  i have exactly the same issue on my side...
any idea to solve this ?

Rick5o, 2023-06-30:
This is my configuration it was working almost perfect until last update :)

[code]const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
[... 100 lines cut ...]
			exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.climate()
			.withSetpoint('occupied_heating_setpoint', 16, 35, 1, ea.STATE_SET)
			.withLocalTemperature(ea.STATE)
			.withLocalTemperatureCalibration(ea.STATE_SET)
			.withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
			.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
			.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
			.withPreset(['hold', 'program']),
			e.temperature_sensor_select(['IN', 'AL', 'OU']),
			
			exposes.composite('programming_mode', 'programming_mode', ea.STATE)
                .withDescription(
                    'Schedule MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature.',
                )
                .withFeature(e.week())
                .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
                .withFeature(exposes.text('holidays_schedule', ea.STATE_SET)),
			
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
	],
    },
];
[/code]

adminsmarthome-x, 2023-08-24:
> This is my configuration it was working almost perfect until last update :)
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> 
> 
> const fz_moes_thermostat_custom = {
> 	cluster: 'manuSpecificTuya',
> 	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
> 	convert: (model, msg, publish, options, meta) => {
>             const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
>             const dp = dpValue.dp;
>             const value = tuya.getDataValue(dpValue);
> 		let temperature;
> 		/* See tuyaThermostat above for message structure comment */
> 		switch (dp) {
> 		case tuya.dataPoints.moesSchedule:
> 			return {
> 				program: [
> 					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
> 					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
> 					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
> 					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
> 					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
> 					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
> 					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
> 					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
> 					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
> 					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
> 					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
> 					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
> 				],
> 			};
> 		case 1: // system mode (OFF, ON)
> 			if(value == 0) {
> 				return {state: 'OFF'};
> 			} else if(value == 1) {
> 				return {state: 'ON'};
> 			} else {
> 				return {};
> 			}
> 		case 2: // system mode (cooling, heating, fan)
> 			if(value == 0) {
> 				return {system_mode: 'cool'};
> 			} else if(value == 1) {
> 				return {system_mode: 'heat'};
> 			} else if(value == 2) {
> 				return {system_mode: 'fan_only'};
> 			} else {
> 				return {};
> 			}
> 		case 28: // fan mode
> 			if(value == 0) {
> 				return {fan_mode: 'low'};
> 			} else if(value == 1) {
> 				return {fan_mode: 'medium'};
> 			} else if(value == 2) {
> 				return {fan_mode: 'high'};
> 			} else if(value == 3) {
[... 120 lines cut ...]
>                 .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
>                 .withFeature(exposes.text('holidays_schedule', ea.STATE_SET)),
> 			
> 			exposes.switch()
> 				.withState('state', false, 'OFF/ON state'),
> 	],
>     },
> ];
> ```

HI R. 

i am trying your code, but i am having this error : 

"/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)"

any idea? Thanks for your support 
Luigi

simguth, 2023-08-24:
> HI R.
>
> i am trying your code, but i am having this error :
>
> "/app/node_modules/zigbee-herdsman-converters/index.js:79 if (converter.options) { ^ TypeError: Cannot read properties of undefined [...]

Hello

Same issue on my side

MBrognoli, 2023-08-30:
Hello
Same issue on my side.

prsveloso, 2023-09-03:
Change this:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');`

By this:
`const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};`

And this:
`fromZigbee: [fz_moes_thermostat_custom],
		onEvent: tuya.onEventSetLocalTime,
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
			exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),`

By this:
`fromZigbee: [fz_moes_thermostat_custom, fz.legacy.moes_thermostat],
		onEvent: tuya.onEventSetLocalTime,
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.legacy.moes_thermostat_child_lock, tz.legacy.moes_thermostat_current_heating_setpoint, tz.legacy.moes_thermostat_mode,
            tz.legacy.hgkg_thermostat_standby, tz.legacy.moes_thermostat_sensor, tz.legacy.moes_thermostat_calibration,
            tz.legacy.moes_thermostat_deadzone_temperature, tz.legacy.moes_thermostat_max_temperature_limit],
			exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),`

I have a different thermostat (TYBAC-006 - _TZE204_mpbki2zm) but I am using this code. I've made some adjustments (for my mode) and it is working for what I consider important settings.

prsveloso, 2023-09-06:
> ok thanks, it seems to work, but now when I make changes to the target temperature I get this error in Z2M: Publish 'set' 'occupied_heating_setpoint' to 'Termostato_comedor' failed: 'TypeError: tuya.sendDataPointValue is not a function'

Like I told, I'm using a different thermostat model. In this line:
`.withSetpoint('occupied_heating_setpoint', 16, 35, 0.1, ea.STATE_SET)` I'm using `'current_heating_setpoint'` instead of `'occupied_heating_setpoint'`. I don't know if it will work for the topic's model, but you can try and give feedback (if you want) ;)

piposant, 2023-09-07:
> Like I told, I'm using a different thermostat model. In this line: `.withSetpoint('occupied_heating_setpoint', 16, 35, 0.1, ea.STATE_SET)` I'm using `'current_heating_setpoint'` instead of [...]

I tried to replace but it didn't work, checking the yaml, and the error "TypeError: tuya.sendDataPointValue is not a function"
The only place where sendDataPointValue appears in the code is the following:

const tz_temperature_setpoint = {
	key: ['occupied_heating_setpoint'],
	convertSet: async (entity, key, value, meta) => {
		await tuya.sendDataPointValue(entity, 16, value);
	},
};

Koenkk, 2023-09-20:
`_TZE204_mpbki2zm` has been added and will be supported in the next release and dev branch. (https://github.com/Koenkk/zigbee-herdsman-converters/pull/6174)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Rick5o, 2023-10-09:
BAC-002-ALZB is it fixed?

Rick5o, 2023-10-09:
> `_TZE204_mpbki2zm` has been added and will be supported in the next release and dev branch. ([Koenkk/zigbee-herdsman-converters#6174](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6174))
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

BAC-002-ALZB is it fixed?

rkanovsky, 2023-10-10:
It still doesn't work for me. Here is my version of the external converter. The code needs a bit of cleaning up, but it works for me.
https://gist.github.com/rkanovsky/93b2354aafb1c5e384cee7eb12847449

Rick5o, 2023-10-10:
Thank you. This is working  :) 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [{
            modelID: 'TS0601',
            manufacturerName: '_TZE200_dzuqwsyg',
        },
    ],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',
    fromZigbee: [tuya.fz.datapoints, fz.ignore_tuya_set_time],
    toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'fan_mode', 'deadzone_temperature', 'eco_mode', 'max_temperature_limit', 'min_temperature_limit']}],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.switch().withState('state', false, 'OFF/ON state'),
		e.child_lock(),
		e.climate()
		        .withLocalTemperature(ea.STATE)
		        .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
		        .withSystemMode(['cool', 'heat', 'fan_only'])
		        .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
		        .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
		e.min_temperature().withValueMin(5).withValueMax(15),
		e.max_temperature().withValueMin(15).withValueMax(45),
		exposes.switch().withState('eco_mode', false, 'ECO Mode OFF/ON'),
		e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (Default: 20 ºC)').withValueMin(15).withValueMax(30),
		e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (Default: 26 ºC)').withValueMin(15).withValueMax(30),
		e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
		e.binary('valve', ea.STATE, 'CLOSED', 'OPEN').withDescription('3-Way Valve status'),
    ],
    meta: {
        tuyaDatapoints:
        [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [4, 'eco_mode', tuya.valueConverter.onOff],
            [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],

            [19, 'max_temperature', tuya.valueConverter.divideBy10],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.divideBy10],
            [27,  'local_temperature_calibration',  {
                from: (value) => {
                    if (value > 4000) 
                        return value - 4096;
                    return value;
                    },
                to: (value) => {
                    if (value < 0) 
                        return 4096 + value;
                    return value;
                    }
                }],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
            [36, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 0, 'CLOSE': 1})],
			[40, 'child_lock', tuya.valueConverter.lockUnlock],
			[103, 'deadzone_temperature', tuya.valueConverter.raw],
			[104, 'min_temperature_limit', tuya.valueConverter.divideBy10],
			[105, 'max_temperature_limit', tuya.valueConverter.divideBy10],
			/* TODO - help needed
			"101": DP for "Schedule/Manual",
            "106": DP for "Schedule",
            "107": DP for "Schedule",
            "108": DP for "Schedule",
            "109": DP for "Schedule",
            "110": DP for "Schedule",
            "111": DP for "Schedule",
            "112": DP for "Schedule"
            */
        ],
    },
};

module.exports = definition;`

This is clean version of [piposant] work but setpoint not working correctly. I don't know why divideBy10 not working.

piposant, 2023-10-10:
My model is modelID: 'TS0601',            manufacturerName: '_TZE200_dzuqwsyg',
And don't work correctly.

Rick5o, 2023-10-10:
> My model is modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg', And don't work correctly.

It is working correctly from yours code except setting temperature. When I set for example 5degree controller change to 50, when I set 6, controller set 60. This is only thing that doesn't work. I can't understand why it multiply by 10 (and divide 10 not working).

Rick5o, 2023-10-10:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [{
            modelID: 'TS0601',
            manufacturerName: '_TZE200_dzuqwsyg',
        },
    ],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',
    fromZigbee: [tuya.fz.datapoints, fz.ignore_tuya_set_time],
    toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'fan_mode', 'deadzone_temperature', 'eco_mode', 'max_temperature_limit', 'min_temperature_limit']}],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.switch().withState('state', false, 'OFF/ON state'),
		e.child_lock(),
		e.climate()
		        .withLocalTemperature(ea.STATE)
		        .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
		        .withSystemMode(['cool', 'heat', 'fan_only'])
		        .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
		        .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
		e.min_temperature().withValueMin(5).withValueMax(15),
		e.max_temperature().withValueMin(15).withValueMax(45),
		exposes.switch().withState('eco_mode', false, 'ECO Mode OFF/ON'),
		e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (Default: 20 ºC)').withValueMin(15).withValueMax(30),
		e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (Default: 26 ºC)').withValueMin(15).withValueMax(30),
		e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
		e.binary('valve', ea.STATE, 'CLOSED', 'OPEN').withDescription('3-Way Valve status'),
    ],
    meta: {
        tuyaDatapoints:
        [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [4, 'eco_mode', tuya.valueConverter.onOff],
            [16, 'current_heating_setpoint', tuya.valueConverter.raw],

            [19, 'max_temperature', tuya.valueConverter.divideBy10],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.divideBy10],
            [27,  'local_temperature_calibration',  {
                from: (value) => {
                    if (value > 4000) 
                        return value - 4096;
                    return value;
                    },
                to: (value) => {
                    if (value < 0) 
                        return 4096 + value;
                    return value;
                    }
                }],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
            [36, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 0, 'CLOSE': 1})],
			[40, 'child_lock', tuya.valueConverter.lockUnlock],
			[103, 'deadzone_temperature', tuya.valueConverter.raw],
			[104, 'min_temperature_limit', tuya.valueConverter.divideBy10],
			[105, 'max_temperature_limit', tuya.valueConverter.divideBy10],
			/* TODO - help needed
			"101": DP for "Schedule/Manual",
            "106": DP for "Schedule",
            "107": DP for "Schedule",
            "108": DP for "Schedule",
            "109": DP for "Schedule",
            "110": DP for "Schedule",
            "111": DP for "Schedule",
            "112": DP for "Schedule"
            */
        ],
    },
};

module.exports = definition;
```

I make a little progress. Temp/fan now working. 
I have one big issue (same as [rkanovsky](https://github.com/rkanovsky)), when you change state off/on fan and valve it's not running on/open.

Rick5o, 2023-10-11:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [{
            modelID: 'TS0601',
            manufacturerName: '_TZE200_dzuqwsyg',
        },
    ],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',

    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({timeStart: '2000'}),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.child_lock(),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
            .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
            .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
            .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
        e.min_temperature().withValueMin(5).withValueMax(15),
        e.max_temperature().withValueMin(15).withValueMax(45),
        e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('ECO mode ON/OFF'),
        e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (default: 20 ºC)').withValueMin(15).withValueMax(30),
        e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (default: 26 ºC)').withValueMin(15).withValueMax(30),
        e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
        e.binary('valve', ea.STATE, 'CLOSED', 'OPEN').withDescription('3-Way Valve status'),
        e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints:
        [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [4, 'eco_mode', tuya.valueConverter.onOff],
            [16, 'current_heating_setpoint', tuya.valueConverter.raw],
            [19, 'max_temperature', tuya.valueConverter.raw],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.raw],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup(
                {'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
            [36, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 0, 'CLOSE': 1})],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
            [104, 'min_temperature_limit', tuya.valueConverter.raw],
            [105, 'max_temperature_limit', tuya.valueConverter.raw],
            [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [101, 'manual_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

```

I think I make it good enough.  Can someone verify.
@prsveloso thank you.

Rick5o, 2023-10-12:
It's not good enough.  When I switch in Z2M on/off/on to turn on valve and fan I need to set temperature above 29 degree (local temp is 20.5). I don't know why.

piposant, 2023-10-13:
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition =
> {
>     fingerprint: [{
>             modelID: 'TS0601',
>             manufacturerName: '_TZE200_dzuqwsyg',
>         },
>     ],
>     model: 'BAC-002-ALZB',
>     vendor: 'HKGK',
>     description: 'BAC series thermostat',
> 
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEvent({timeStart: '2000'}),
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
>         e.child_lock(),
>         e.climate()
>             .withLocalTemperature(ea.STATE)
>             .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
>             .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
>             .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
>             .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
>         e.min_temperature().withValueMin(5).withValueMax(15),
>         e.max_temperature().withValueMin(15).withValueMax(45),
>         e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('ECO mode ON/OFF'),
>         e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (default: 20 ºC)').withValueMin(15).withValueMax(30),
>         e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (default: 26 ºC)').withValueMin(15).withValueMax(30),
>         e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
>         e.binary('valve', ea.STATE, 'CLOSED', 'OPEN').withDescription('3-Way Valve status'),
>         e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
>         ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
>     ],
>     meta: {
>         tuyaDatapoints:
>         [
>             [1, 'state', tuya.valueConverter.onOff],
>             [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
>             [4, 'eco_mode', tuya.valueConverter.onOff],
>             [16, 'current_heating_setpoint', tuya.valueConverter.raw],
>             [19, 'max_temperature', tuya.valueConverter.raw],
>             [24, 'local_temperature', tuya.valueConverter.divideBy10],
>             [26, 'min_temperature', tuya.valueConverter.raw],
>             [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
>             [28, 'fan_mode', tuya.valueConverterBasic.lookup(
>                 {'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
>             [36, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 0, 'CLOSE': 1})],
>             [40, 'child_lock', tuya.valueConverter.lockUnlock],
>             [103, 'deadzone_temperature', tuya.valueConverter.raw],
>             [104, 'min_temperature_limit', tuya.valueConverter.raw],
>             [105, 'max_temperature_limit', tuya.valueConverter.raw],
>             [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
>             [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
>             [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
>             [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
>             [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
>             [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
>             [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
>             [101, 'manual_mode', tuya.valueConverter.onOff],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```
> 
> I think I make it good enough. Can someone verify. @prsveloso thank you (it's basically yours code).

I think it's working well for me. Only one possible improvement: it would be possible to modify the temperature from 05 to 0.5

Rick5o, 2023-10-13:
From 1 to 0.5?

piposant, 2023-10-17:
> From 1 to 0.5?

Yes, from 1 to 0.5, if I modify the parameter when performing the tests, sometimes the value does not jump but to 00 or 65000

pcrednet, 2023-11-27:
@prsveloso 
I have this thermostat at home and the only problem I see is that home assistant does not show the status of heating or cooling.
It only shows the mode it is in, whether it is cooling or heating.
Is there any way to modify this so that home assistant interprets it correctly?
Thx.

Rick5o, 2023-11-28:
> @prsveloso I have this thermostat at home and the only problem I see is that home assistant does not show the status of heating or cooling. It only shows the mode it is in, whether it is cooling or heating. Is there any way to modify this so that home assistant interprets it correctly? Thx.

Not true at all. You probably using default config not that one in this thread, that show correctly cooling/heating/off status when you change at controller.

dilayn, 2024-03-18:
> It's not good enough. When I switch in Z2M on/off/on to turn on valve and fan I need to set temperature above 29 degree (local temp is 20.5). I don't know why. "deadzone_temperature" is 1. It's turning off properly.
> 
> Do you know how go secret menu and look for all setting. Manual saying push two first buttons when off but it's not working.
> 
> ![image](https://private-user-images.githubusercontent.com/121449694/274483081-75e5417c-030f-4a5d-9c0c-6b7413188b55.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MjI4MTYsIm5iZiI6MTcxMDcyMjUxNiwicGF0aCI6Ii8xMjE0NDk2OTQvMjc0NDgzMDgxLTc1ZTU0MTdjLTAzMGYtNGE1ZC05YzBjLTZiNzQxMzE4OGI1NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxOFQwMDQxNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZmZmNmE1NTQ5YzhjZmY0ZmQ0YzhlZTJkOWQyNjJjZTkwZTM0YzlhZTFmYjJmZWU0NThmNGU3Y2EzYzlhYWZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Ojx_egrpnWd_Y3fjnxYTSuWfkD0wQDq1y_91-vThOK8)

Hi, have you managed to get into the secret menu with all the settings? I'm trying to swich Fan Control from 00 to 01 (Low speed Fan on when room temp is at set temp). No option in the Z2M.

piposant, 2025-04-10:
Hello, I hadn't noticed this, but it doesn't show me the valve status. Do you see it? Open or closed?