Koenkk/zigbee2mqtt #7982: CC2652R1 not starting

agrieco, 2021-07-08:
## What happened
zigbee2mqtt isn't able to start talking to my CC2652R1.  I'm getting "Error: Failed to connect to the adapter".

## What did you expect to happen

## How to reproduce it (minimal and precise)
Run zigbee2mqtt and the above error is seen.

## Debug info
Zigbee2MQTT version:  version 1.20.0 (commit #41b67fdd)
Adapter hardware:  CC2652R1
Adapter firmware version:  CC1352P2_CC2652P_launchpad_coordinator_20210120.hex

/dev/ttyACM0 debug:
```
pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jul  7 11:43 /dev/ttyACM0
pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Jul  7 11:16 usb-Texas_Instruments_XDS110__03.00.00.15__Embed_with_CMSIS-DAP_L1100H3O-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Jul  7 11:16 usb-Texas_Instruments_XDS110__03.00.00.15__Embed_with_CMSIS-DAP_L1100H3O-if03 -> ../../ttyACM1
pi@raspberrypi:/opt/zigbee2mqtt $ test -w /dev/ttyACM0 && echo success || echo failure
success
```

Startup debug:
```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start
npm WARN npm npm does not support Node.js v14.17.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12.
npm WARN npm You can find the latest version at https://nodejs.org/

> zigbee2mqtt@1.20.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2021-07-07 11:43:05: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2021-07-07 11:43:05: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-07-07.11-43-02' filename: log.txt
Zigbee2MQTT:info  2021-07-07 11:43:05: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fdd)
Zigbee2MQTT:info  2021-07-07 11:43:05: Starting zigbee-herdsman (0.13.117)
Zigbee2MQTT:debug 2021-07-07 11:43:05: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2021-07-07 11:43:25: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-07-07 11:43:25: Failed to start zigbee
Zigbee2MQTT:error 2021-07-07 11:43:25: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-07-07 11:43:25: Exiting...
Zigbee2MQTT:error 2021-07-07 11:43:25: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:93:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.20.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.20.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-07-07T11_43_25_720Z-debug.log
```

Startup debugging is here:

Heardsman debug:
```
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +13ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +483ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
```

AbeltjeNL, 2021-07-08:
Same problem here. After numerous retries it will restore itself and will work properly. Sometimes I've got to manually restart the addon to get it working

agrieco, 2021-07-08:
> Same problem here. After numerous retries it will restore itself and will work properly. Sometimes I've got to manually restart the addon to get it working

I've had mine trying to run for about 3 days now...no magical 'restore itself' happening here I'm afraid.

AbeltjeNL, 2021-07-08:
> > Same problem here. After numerous retries it will restore itself and will work properly. Sometimes I've got to manually restart the addon to get it working
> 
> 
> 
> I've had mine trying to run for about 3 days now...no magical 'restore itself' happening here I'm afraid.

Did you try to change the pan id to 0x1a62 or 0x1a63? This will also sometimes do the trick..

agrieco, 2021-07-09:
> Did you try to change the pan id to 0x1a62 or 0x1a63? This will also sometimes do the trick..

Changing the pan id doesnt fix the issue.  thanks

closed 2021-08-16 by github-actions[bot]