Koenkk/zigbee2mqtt #7983: Legrand Home/Away Master Switch can't be paired !

Nasrudin78, 2021-07-08:
## What happened
My Legrand home/away , master switch device is not paired. I have several devices (Aqara, Sonoff...) that are fully integrated with my Zzh! coordinator , also some Legrand cover switches are paired. But Is impossible to with home/away one. 
 
## What did you expect to happen
Pair this device (https://www.zigbee2mqtt.io/devices/064873.html) , since other Legrand devices (covers) are paired correclty

## How to reproduce it (minimal and precise)
In home assistant : disable join
I press the reset button for 8 secs in the home/away switch
In home assistant: permit join
I press one more time reset button for 8 secs till a red flash is displayed
After that it begins to sync (blue led blinks) but then it blinks 2 times in red and finally 1 time in red. 

Note: Original Legrand Gateway is switched off and disconnected from main power.
Note2: I'am using channel 11 in zigbee2mqqt config file

## Debug info
Zigbee2MQTT version:  1.20.0-1
Adapter hardware: zzh! CC2652R running on a Raspberry Pi 4 
Adapter firmware version: flashed with CC2652R_coordinator_20210120.hex

closed 2021-08-16 by github-actions[bot]

MysterePMS, 2021-11-04:
Hello @Nasrudin78 ,
Did you manage to integrate your departure / arrival?

Nasrudin78, 2021-11-04:
No. No way

MysterePMS, 2021-11-04:
As soon as I get there, I'll keep you informed ;-)

Loic691, 2023-01-02:
Did you sucess with it ?

cardentey, 2025-03-07:
Did you sucess with it ? Please