Koenkk/zigbee2mqtt #7985: Fyrtur state not reporting in HA

AshtonKem, 2021-07-08:
## What happened

Sometime around HA 2021.6.6, my Fyrtur blinds stopped reporting their state to HA via MQTT, resulting in my blinds not moving anymore. Both reconfiguring and re-pairing my blinds with Zigbee2mqtt has not solved the issue.

I can manually open and close the blinds via the open and close button in the Zigbee2mqtt dashboard, but it does not work in HA, which makes me think it's a Zigbee2mqtt -> MQTT issue or a MQTT -> HA issue.

### Pairing logs
```
debug Device 'Bedroom Blind Two' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bedroom Blind Two","ieee_address":"0x588e81fffe6402f6"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Bedroom Blind Two"},"type":"device_announced"}'
debug Received Zigbee message from 'Bedroom Blind Two', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587675185,"imageType":4487,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug Check if update available for '0x588e81fffe6402f6' (FYRTUR block-out roller blind)
debug Is new image available for '0x588e81fffe6402f6', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4487,"fileVersion":587675185}', latest meta '{"fileVersion":587675185,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.15.034/bin/10037585-5.1-TRADFRI-connected-blind-2.3.073.ota.ota.signed"}'
info MQTT publish: topic 'zigbee2mqtt/Bedroom Blind Two', payload '{"battery":94,"linkquality":63,"position":0,"update":{"state":"idle"},"update_available":false}'
```

## What did you expect to happen

## How to reproduce it (minimal and precise)

Not clear. Own Fyrtur blinds and pair?

## Debug info
Zigbee2MQTT version:1.18.1-1 
Adapter hardware: zzh
Adapter firmware version: 20210120

AshtonKem, 2021-07-08:
Looks like I found an error log:

```
2021-07-08 10:28:48 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/0x588e81fffe6402f6/set', 'device': {'identifiers': ['zigbee2mqtt_0x588e81fffe6402f6'], 'manufacturer': 'IKEA', 'model': 'FYRTUR roller blind (E1757)', 'name': '0x588e81fffe6402f6', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/0x588e81fffe6402f6', 'name': '0x588e81fffe6402f6', 'position_topic': 'zigbee2mqtt/0x588e81fffe6402f6', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/0x588e81fffe6402f6/set', 'unique_id': '0x588e81fffe6402f6_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
return schema(data)
File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 125, in validate_options
raise vol.Invalid(
voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 179, in async_discover
config = schema(discovery_payload)
File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
return self._exec((Schema(val) for val in self.validators), v)
File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
raise e if self.msg is None else AllInvalid(self.msg, path=path)
File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
v = func(v)
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__
raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
```

Koenkk, 2021-07-08:
Updating to zigbee2mqtt 1.19.0+ will fix it.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-07-08 by Koenkk