Koenkk/zigbee2mqtt #7987: Alternative manufid for newer Trust Remote ZYCT-202

milobt, 2021-07-08:
I have two Trust Remotes ZYCT-202, but one is slightly newer and appears to have a slightly different model number printed on the back of the remote and then is also recognised by a different manufid.
They appear to be exactly the same, the only noticeable different is that the LEDs on the new remote are more green and brighter; the older remote LEDs are a yellow-green and dimmer.

Recognised remote manufid: 4151 - Model number: 71171 ZYCT-202
Unrecognised remote manufid: 4653 - Model number: 70171 ZYCT-202

Please can you add support for the other model.
..

## data/database.db entry of the device
Failed to pair:
Trust 70171 ZYCT-202

{"id":28,"type":"EndDevice","ieeeAddr":"0x00158d0004ff6dbf","nwkAddr":59331,"manufId":4653,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1625777064750}

Successfully Paired: 
Trust 71171 ZYCT-202

{"id":18,"type":"EndDevice","ieeeAddr":"0x00158d0001ccab1b","nwkAddr":27016,"manufId":4151,"manufName":"\u0000\u0000\u0000","powerSource":"Battery","modelId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2080,"inClusterList":[0,4,3,6,8,4096],"outClusterList":[0,4,3,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"\u0000\u0000\u0000","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":1010,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1625640171994}

milobt, 2021-08-08:
@Koenkk please could you add this manufid?

Koenkk, 2021-08-09:
Since the `Trust 70171 ZYCT-202` failed to pair it cannot be added yet. Try pairing this device close to the coordinator and see if you can manage to pair it correctly (interview should succeed).

milobt, 2021-08-13:
So today it has paired and added to my list of devices and is reporting correct json, says 'Support Status': supported, but says 'interview completed': False.
[image]

However, the previsouly paired version in my name: Miles Bedside Remote, is missing some data, but still works fine:
[image]

Koenkk, 2021-08-13:
Can you provide the herdsman debug log when pairing fails?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

milobt, 2021-08-17:
https://pastebin.com/dHyYJLEm

Device: 0x00158d0004ff6dbf
Was called: Nichola Bedside Remote

Koenkk, 2021-08-17:
This is the debug logging, not the herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

milobt, 2021-08-17:
I’m using hassio so added “zigbee_herdsman_debug: true” to my confit which then started logging what seemed to be herdsman logs under the logs tab, but is this not saved to the logs file?
Is it saved anywhere?
What is STDOUT?

Koenkk, 2021-08-18:
You can grab it by sshing into your machine and executing ` docker logs zigbee2mqtt > log.txt 2>&1` where `zigbee2mqtt` needs to replaced with the container id (can be found with `docker ps -a`.

milobt, 2021-09-08:
I can't seem to login to my home assistant via secure ssh.
So can't get get a list of dockers, let alone read from the file in the z2m docker.