Koenkk/zigbee2mqtt #7988: Ecolink Zigbee Alarm Listener FFZB1-SM-ECO

onedr0p, 2021-07-08:
## Information about the device + link

I have tried to pair this device but I am not able to get it to detect smoke.

`e.tamper` and `e.battery_low()` appear to work just fine

```js
// FFZB1-SM-ECO.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FFZB1-SM-ECO'],
    model: 'FFZB1-SM-ECO',
    vendor: 'Ecolink',
    description: 'Ecolink Zigbee Alarm Listener',
    fromZigbee: [fz.ias_smoke_alarm_1],
    toZigbee: [],
    exposes: [e.smoke(), e.battery_low(), e.tamper()],
};
```

https://smile.amazon.com/gp/product/B07HN5CZCV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

## data/database.db entry of the device

```
{"id":31,"type":"EndDevice","ieeeAddr":"0x000d6f000b5c9f23","nwkAddr":31907,"manufId":4130,"manufName":"Ecolink","powerSource":"Battery","modelId":"FFZB1-SM-ECO","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FFZB1-SM-ECO","manufacturerName":"Ecolink","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":3,"swBuildId":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002160067d","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":3,"swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1625777508347}
```

Koenkk, 2021-07-09:
Do you see anything in the debug log when the alarm is triggered?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

onedr0p, 2021-07-09:
I have debug enabled. I will need to verify, but initially I didn't see anything come thru. I'll try again when my house is empty to test. Thanks!

onedr0p, 2021-07-09:
```
 Successfully interviewed '0x000d6f000b5c9f23', device has successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x000d6f000b5c9f23","ieee_address":"0x000d6f000b5c9f23","status":"successful","supported":false},"type":"device_interview"}'
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":36}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":36}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":36}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51905841,"imageType":4,"manufacturerCode":4130}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51905841,"imageType":4,"manufacturerCode":4130}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51905841,"imageType":4,"manufacturerCode":4130}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51905841,"imageType":4,"manufacturerCode":4130}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51905841,"imageType":4,"manufacturerCode":4130}' from endpoint 1 with groupID 0
```

Well the devices LED does flash red when it detects the sound from the smoke alarm, however it looks like nothing gets announced.

The lines with 

```Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":36}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000b5c9f23', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0```

are just the tamper switch being toggled.

Doesn't seem like it announces anything when i

onedr0p, 2021-07-09:
Seems like this not working is a "feature", the folks over at [hubitat](https://community.hubitat.com/t/driver-request-ecolink-firefighter-zigbee/48551) were having the same issues as me.

I will close this issue and return the device.

closed 2021-07-09 by onedr0p