Koenkk/zigbee2mqtt #8011: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)

LucasJanin, 2021-07-10:
## What happened
Zigbee2mtt failed to start

log:
Using '/app/data' as data directory
Zigbee2MQTT:debug 2021-07-10 17:45:09: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-07-10 17:45:09: Logging to console and directory: '/app/data/log/2021-07-10.17-45-02' filename: log.txt
Zigbee2MQTT:debug 2021-07-10 17:45:09: Removing old log directory '/app/data/log/2021-07-10.17-41-59'
Zigbee2MQTT:info  2021-07-10 17:45:09: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
Zigbee2MQTT:info  2021-07-10 17:45:09: Starting zigbee-herdsman (0.13.117)
Zigbee2MQTT:debug 2021-07-10 17:45:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
zigbee-herdsman:adapter:zStack:znp:log Serialport opened +6ms
zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +457ms
zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
zigbee-herdsman:adapter:zStack:znp:log Port closed +1s
Zigbee2MQTT:error 2021-07-10 17:45:22: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-07-10 17:45:22: Failed to start zigbee
Zigbee2MQTT:error 2021-07-10 17:45:22: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-07-10 17:45:22: Exiting...
Zigbee2MQTT:error 2021-07-10 17:45:22: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)
at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:93:31)
at Generator.throw (<anonymous>)
at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

I tried to disconnect and reconnect the CC2531 and restart Zigbee2MQTT.
This was previously fixing my starting issues, but not this time.

## What did you expect to happen
Zigbee2MQTT was working a few days ago.
I have CC2531 with 40 devices, sometimes some devices get disconnected, but now nothing is working.

## How to reproduce it (minimal and precise)
Zigbee2MQTT run on a Synology NAS DS918+ on DSM 6.2.4-25556

docker run -itd --name="Zigbee2MQTT" --net=host -e TZ=America/Montreal -e DEBUG=zigbee-herdsman* --device=/dev/ttyACM0 --restart=always -v /volume1/docker/zigbee2mqtt/data:/app/data koenkk/zigbee2mqtt:latest

## Debug info
Zigbee2MQTT version: 1.20.0 (commit #41b67fd)
Adapter hardware: CC2531
Adapter firmware version: 1.2 20201127/20201128 (I suppose, I don't know how to check it)

Thanks in advance for any help!

Koenkk, 2021-07-12:
Likely point 2 of https://www.zigbee2mqtt.io/information/FAQ.html#error-srsp---sys---ping-after-6000ms

LucasJanin, 2021-07-19:
Hi @Koenkk ,

Yes, you're right! I reflashed using source_routing firmware and remove 4 unused devices. Now all look working.
But for the long term, I need to switch to a more powerful adapter. Thanks for your help.

closed 2021-07-20 by Koenkk