Koenkk/zigbee2mqtt #8014: Request support Aqara ZNCZ15ML

xfzha0, 2021-07-11:
## Information about the device + link
Aqara smart plug  (ZNCZ15ML; lumi.plug.macn01)

It has 3 parameters for customization:

Maximum Power (100-2500w)
Power-off memory (on off)
Turn off indicator light (on off)

https://www.aqara.com/cn/productDetail/d37
[image]

## data/database.db entry of the device
{"id":4,"type":"Router","ieeeAddr":"0x04cf8cdf3c82f5fe","nwkAddr":54031,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug.macn01","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.plug.macn01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"05-12-2020"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237b9e60","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"05-12-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1625976508209}

<details>
<summary>Zigbee2mqtt pairing log</summary>
<pre><code>
info Device '0x04cf8cdf3c82f5fe' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c82f5fe","ieee_address":"0x04cf8cdf3c82f5fe"},"type":"device_joined"}'
Info Starting interview of '0x04cf8cdf3c82f5fe'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c82f5fe","ieee_address":"0x04cf8cdf3c82f5fe","status":"started"},"type":"device_interview"}'
Debug Device '0x04cf8cdf3c82f5fe' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c82f5fe","ieee_address":"0x04cf8cdf3c82f5fe"},"type":"device_announce"}'
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.plug.macn01"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"appVersion":31}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{"dateCode":"05-12-2020"}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug Skipping message, definition is undefined and still interviewing
Info Successfully interviewed '0x04cf8cdf3c82f5fe', device has successfully been paired
Warning Device '0x04cf8cdf3c82f5fe' with Zigbee model 'lumi.plug.macn01' and manufacturer name 'LUMI' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x04cf8cdf3c82f5fe","ieee_address":"0x04cf8cdf3c82f5fe","status":"successful","supported":false},"type":"device_interview"}'
Debug Received Zigbee message from '0x04cf8cdf3c82f5fe', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":31,"imageType":12,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'lumi.plug.macn01' and manufacturer name 'LUMI'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
</code></pre>
</details>
<details>
<summary>Dump zigbee-packets with ZBOSS Sniffer and Wireshark</summary>
<pre><code>
[ZNCZ15ML.zip](https://github.com/Koenkk/zigbee2mqtt/files/6796582/ZNCZ15ML.zip) - encrypted pcapng-dump (Network Key for decryption entries: 23ec08029d40285b60f0912683cff67b)
Recorded events in the dump:
  * device pairing
  * Report Plug in electrical appliances = true  - Frame: 101
  * Report Power - Frame: 113
  * changing device settings
      * Turn on - Frame: 106
      * Change Maximum Power = 2500W - Frame: 132
      * Enable Power-off memory = true - Frame: 149
      * Turn off indicator light = false - Frame: 163
Environment:
  Plug Firmware Version: 0.0.0_0031
  Aqara Hub G3 (ZNSXJ13LM) Firmware Version: 3.27_0019.0004
  Aqara Home APP Version: 2.3.12

</code></pre>
</details>

xfzha0, 2021-07-17:
I tried the tutorial [How to support new devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html).
But no useful logs are reported in the debug log.

xfzha0, 2021-07-18:
Please help me.. @Koenkk

Koenkk, 2021-07-19:
With the following external converter you should get power reading:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.plug.macn01'],
    model: 'ZNCZ15LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 plug',
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_switch_opple_basic],
    toZigbee: [tz.on_off, tz.xiaomi_power],
    exposes: [
        e.switch(), e.power().withAccess(ea.STATE_GET), e.energy(), e.temperature().withAccess(ea.STATE),
        e.voltage().withAccess(ea.STATE), e.current()],
};

module.exports = definition;
```

For other features like disabling led and power outage memory please provide a sniff when pairing this device with the Xiaomi gateway and enabling these features: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

xfzha0, 2021-07-20:
Thanks for your help.
 on/off is work well.
power can't get 
log:
<code>
Debug Received MQTT message on 'zigbee2mqtt/ZNCZ15LM/get' with data '{"power":""}'&nbsp;
Debug Publishing get 'get' 'power' to 'ZNCZ15LM'&nbsp;
Error Publish 'get' 'power' to 'ZNCZ15LM' failed: 'TypeError: Cannot read property 'read' of undefined'&nbsp;
Debug TypeError: Cannot read property 'read' of undefined at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1727:28) at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:276:37) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44) at process._tickCallback (internal/process/next_tick.js:68:7)&nbsp;
Debug Received Zigbee message from 'ZNCZ15LM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,40]}' from endpoint 1 with groupID 0&nbsp;
Debug No converter available for 'ZNCZ15LM' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,40]}'&nbsp;
Debug Received Zigbee message from 'ZNCZ15LM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":43}' from endpoint 1 with groupID 0&nbsp;
Debug No converter available for 'ZNCZ15LM' with cluster 'haElectricalMeasurement' and type 'attributeReport' and data '{"activePower":43}'
</code>

I used device_tempure to get the temperature normally.

power outage memory used xiaomi_switch_power_outage_memory  is work fine.

disabing led can't work.

The sniffed packet is already provided above.

Thanks again!

xfzha0, 2021-07-20:
```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.plug.macn01'],
    model: 'ZNCZ15LM',
    vendor: 'Xiaomi',
    description: 'Aqara T1 power plug',
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_switch_opple_basic],
    toZigbee: [tz.on_off, tz.xiaomi_power, tz.xiaomi_switch_power_outage_memory, tz.xiaomi_led_disabled_night],
    exposes: [e.switch(), e.power().withAccess(ea.STATE_GET), e.energy(), e.temperature().withAccess(ea.STATE), e.voltage().withAccess(ea.STATE), e.current(), e.consumer_connected().withAccess(ea.STATE), e.power_outage_memory(), e.led_disabled_night().withAccess(ea.STATE_SET),],
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
        },
};

module.exports = definition;
```
The above code, most of the functions are normal, but power can only wait for the report to be updated, and the active acquisition will report an error :
```
Debug Received MQTT message on 'zigbee2mqtt/ZNCZ15LM/get' with data '{"power":""}'
Debug Publishing get 'get' 'power' to 'ZNCZ15LM'
Error Publish 'get' 'power' to 'ZNCZ15LM' failed: 'TypeError: Cannot read property 'read' of undefined'
Debug TypeError: Cannot read property 'read' of undefined at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1727:28) at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:276:37) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:397:44) at process._tickCallback (internal/process/next_tick.js:68:7)
 ```
Can you help me fix this error?

Koenkk, 2021-07-20:
- Does `await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});` really make a difference?
- `tz.xiaomi_switch_power_outage_memory` doesn't support the ZNCZ15LM so this for sure won't work
- for reading power, can you try a read of the genAnalogInput attribute `presentValue` via the frontend, does this return a value?

[image]

xfzha0, 2021-07-21:
* Yes, after adding <code>await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});</code>  <code>xiaomi_switch_opple_basic</code> will take effect. It seems that it will make the device start reporting its attributes and make it readable by <code>xiaomi_switch_opple_basic</code>.
* <code>tz.xiaomi_switch_power_outage_memory</code>because after <code>await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});</code> is added, it also takes effect.
I modified <code>xiaomi_switch_power_outage_memory</code> in toZigbee.js:
```js
xiaomi_switch_power_outage_memory: {
        key: ['power_outage_memory'],
        convertSet: async (entity, key, value, meta) => {
            if (['ZNCZ04LM', 'QBKG25LM', 'SSM-U01', 'QBKG39LM', 'ZNCZ15LM'].includes(meta.mapped.model)) {
                await entity.write('aqaraOpple', {0x0201: {value: value ? 1 : 0, type: 0x10}}, manufacturerOptions.xiaomi);
            } else if (['ZNCZ02LM', 'QBCZ11LM'].includes(meta.mapped.model)) {
                const payload = value ?
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x07, 0x01, 0x10, 0x01], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x08, 0x01]] :
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x09, 0x01, 0x10, 0x00], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x0a, 0x01]];

                await entity.write('genBasic', {0xFFF0: {value: payload[0], type: 0x41}}, manufacturerOptions.xiaomi);
                await entity.write('genBasic', {0xFFF0: {value: payload[1], type: 0x41}}, manufacturerOptions.xiaomi);
            } else {
                throw new Error('Not supported');
            }

            return {state: {power_outage_memory: value}};
        },
        convertGet: async (entity, key, meta) => {
            if (['ZNCZ04LM', 'QBKG25LM', 'SSM-U01', 'QBKG39LM', 'ZNCZ15LM'].includes(meta.mapped.model)) {
                await entity.read('aqaraOpple', [0x0201]);
            } else if (['ZNCZ02LM', 'QBCZ11LM'].includes(meta.mapped.model)) {
                await entity.read('aqaraOpple', [0xFFF0]);
            } else {
                throw new Error('Not supported');
            }
        },
    },
```
* I can't read of the <code>genAnalogInput</code> attribute <code>presentValue</code> via the frontend, this will return an error:
<pre><code>Error Publish 'set' 'read' to 'ZNCZ15LM' failed: 'Error: Read 0x04cf8cdf3c82f5fe/1 genAnalogInput(["presentValue"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46131 - 1 - 10 - 12 - 1 after 10000ms)'
</code></pre>

xfzha0, 2021-07-21:
Wow, when I re-pair the device, if there is no <code>await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f})</code>, then many states cannot be read, and then I re-add <code>await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f})</code> and re-pair the device, many states can be displayed normally, but every time zigbee2mqtt is restarted, an error will be reported 3 times :
<pre>
error 2021-07-22 01:26:41: Failed to configure '0x04cf8cdf3c82f5fe', attempt 1 (Error: Write 0x04cf8cdf3c82f5fe/1 aqaraOpple({"mode":1}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8273 - 1 - 13 - 64704 - 4 after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
error 2021-07-22 01:27:04: Failed to configure '0x04cf8cdf3c82f5fe', attempt 2 (Error: Write 0x04cf8cdf3c82f5fe/1 aqaraOpple({"mode":1}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8273 - 1 - 14 - 64704 - 4 after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
error 2021-07-22 01:27:28: Failed to configure '0x04cf8cdf3c82f5fe', attempt 3 (Error: Write 0x04cf8cdf3c82f5fe/1 aqaraOpple({"mode":1}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8273 - 1 - 20 - 64704 - 4 after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
</pre>

PS: Sorry for my bad English, translated from Google.

Koenkk, 2021-07-21:
Try changing to `await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true})`

closed 2021-07-24 by xfzha0

Chengxuyuan0, 2024-12-11:
> ```js
> ZNCZ15LM
> ```

Hello, can you obtain the voltage values now? Looking forward to your reply!