Koenkk/zigbee2mqtt #8016: Ikea E1525 motion sensor binds to wrong group ID

Eddy805, 2021-07-11:
## What happened

Ikea E1525 motion sensor seems to be binding to the wrong group ID. Logging shows binding to the correct group (ID 2), but when triggered, it is sending messages for a non existing group (ID 48000).

Creating a group with the reported group ID when triggering gets a working result. Looks like either the sensor accepts just the one group ID, or Z2M is sending a wrong ID when binding.

Loggings: https://hastebin.com/sojenifaja.yaml

Reproduced with 2 motion sensors (other sensor uses another group)

## What did you expect to happen

Message sent for correct group ID when triggered.

## How to reproduce it (minimal and precise)

- Create group, add light
- Bind E1525 to group
- Trigger occupancy

## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: CC2531
Adapter firmware version: default 20201127

MattWestb, 2021-07-12:
E1525 is the old motion sensor and its using ZLL and not ZB3 and is making one random ZLL  group its controlling then being retested.

So you shall using that group for adding your devices to and its working.

Changing the bonded group is only possible with touch link in ZLL devices that is not implanted in Z2M.

Eddy805, 2021-07-16:
Thanks for the explanation! Guess it's working as supposed to, then.

closed 2021-07-16 by Eddy805