Koenkk/zigbee2mqtt #8020: Device - model 'TS011F' and manufacturer name '_TZ3000_jvzvulen'

kotyla, 2021-07-12:
Hello,

I've got another TS011F plug not supported. I've tried to use instruction how to support new devices but I failed (seems too complicated for me). I've also tried dev branch.

Is it possible to add it ?
Thank you.

`Device '0x60a423fffeff96db' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_jvzvulen' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

[image]
[image]

DavidFFerreira, 2021-07-12:
you need to folow de labels for request a  new device...and this device i think is only add a name to an existing code device =)

Koenkk, 2021-07-12:
Added it, please let me know if the power monitoring works correctly.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kotyla, 2021-08-03:
It's working in HA after update to 1.21.0-1. Thank you very very much.

closed 2021-08-03 by kotyla