Koenkk/zigbee2mqtt #8022: Problem with binding

umutcelebi, 2021-07-12:
First of all i am a verry new with Home Assistant and Z2M.
I am trying to do easy and simple things like this but getting stuck.
Also the bulbs are verry slow going on and off.
All the LQI is 150+ so i was thinking the binding would help?
But cant do that..
If you need more info please ask me because i am not sure if i do this right.

## What happened
I want to bind my all 4 GU10 IKEA TRADFRI bulbs to my remote from IKEA to work.
## What did you expect to happen
It is not binding and giving a error
## How to reproduce it (minimal and precise)
In the UI just choose group "Keuken" and select OnOff and click Bind

## Debug info
Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: Coordinator revision
20201128

Link log:
https://pastebin.com/ZHVgPsbM

MattWestb, 2021-07-13:
Is you trying binding the remote to one light or to one light group ?

IKEA controlling devices is only supporting binding to light groups and not direct to devices so first making one group for the lights and the binding the remote to the group.

umutcelebi, 2021-07-13:
Yes it is using light group. Otherwise it is probably going more wrong.

MattWestb, 2021-07-13:
Looks like one timeout then sending the bind request.
Did you "waking" the remote up then sending the commands so its not sleeping ?

(The IKEA test firmware is supporting device binding but no grope but its very likely some wrong in the firmware then its not moved to the production feed for around on mouth.)

umutcelebi, 2021-07-13:
I think it is now bound to the group. But when i turn off my z2m then i cant use them. Do i need to do anything special after binding?

MattWestb, 2021-07-13:
Normally not. If light is in the group and the remote is bonded to it then it shall working also if the coordinator is off line. But it can being that the remote is having the coordinator as its parent and need some time for finding one new router for connecting the the network before its working OK

umutcelebi, 2021-07-13:
Hmm okay, it did not work. Do i need to wait like 5 to 10 minutes or so? Also when i check the map i do not see them bounded with each other. And 2 spots are not even connected to the coordinator or as router.[image]

MattWestb, 2021-07-13:
No is only need some seconds for doing  the binding.

But on the map its looks like you remote is having the light as its parent = good but  the parent dont have connection to the coordinator so its perhaps was not getting  the bind command you was sending.

umutcelebi, 2021-07-13:
First of all thanks for your fast responses!
What can i do?

umutcelebi, 2021-08-04:
Issue seems to be solved after moving to celeron minipc from rpi3.

MattWestb, 2021-08-04:
If the PI is running Z2M and the celeron is having USB3 ports with connected devices it can blocking the radio with interference so nothing is working at all (normal with CornBee radio adapters that need one USB extrication cable for working OK) but many other things can doing the same like WiFi, Bluetooth, Ethernet and power supply adapters.

Great that you was getting it working well !!

umutcelebi, 2021-08-04:
I think that i had to many integrations and addons on my rpi3 which slowered up everything.

closed 2021-08-09 by umutcelebi