Koenkk/zigbee2mqtt #8023: Permit join not persist

aureliosaraiva, 2021-07-12:
## What happened

The **permit join** option is not persisting. Even marking the setting as true.

[image]

## What did you expect to happen

I expect that and I set permit join to true, my coordinator will always allow entry from new devices.

Whenever I need to add a new device, I need to enable allow join again, even if it is permanently enabled.

## Debug info
**Zigbee2MQTT version**: 1.20.0 commit: 41b67fd
**Coordinator type**: zStack3x0
**Coordinator revision**: 20201026
**Frontend version**: 0.4.29

MattWestb, 2021-07-13:
For current Zigbee standard (Zigbee 3) its not allowed having Permit join open more then 255 seconds (security risk).

If you is having one HA 1.x coordinator its OK but not recommended but if running on one Zigbee 3 firmware it shall not being possible opening the network for longer time.

So I recommending you doing one firmware downgrade of your coordinator to one HA 1.x firmware but if you is having Zigbee 3 routers they is timing out if you is setting to long time then they must doing that  by Zigbee 3 standard or relearning how one Zigbee system is working in real.

Koenkk, 2021-07-13:
@MattWestb this is not true for z2m since it will send the permit join request again within the 255 seconds.

When joining fails, do you still see disable join in the frontend?

[image]

aureliosaraiva, 2021-07-13:
@Koenkk Yes, whenever it resets, I have to reactivate it again.

@MattWestb Unfortunately it's the way I use today to get xiaomi/aqara devices back up and running after switching repeaters.

I understand the security issue, but today analyzing the use of automations using zigbee in my region and even in my country, the risk is almost zero in leaving it enabled.

At least until I find a cheaper and better solution than xiaomi

MattWestb, 2021-07-13:
I dont have problems with Aqara Magnet and Weather version 1 and 2 but its depends of the parent they is having if they is leaving or staying in the network.

You can doing one automation that is closing the network and then opening it a gen some seconds every 250 seconds after and you can have 24/7 open if you like. 
If not sending closing i think some Zigbee 3 devices is refusing extending the time then its have "running out".

aureliosaraiva, 2021-07-14:
@MattWestb I think I'll do that anyway, at least until I solve this problem.

Koenkk, 2021-07-14:
@aureliosaraiva could you provide the herdsman debug logging from starting z2m till the permit join is automatically disabled?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

aureliosaraiva, 2021-07-16:
@Koenkk Problem found and solved. There was an automation in HomeAssistant that turned it off automatically. Apparently this automation did not work before and with the update of zigbee2mqtt started to work.

Well, anyway, it's working now, sorry for opening an issue with that.

Thanks.

closed 2021-07-16 by aureliosaraiva