Koenkk/zigbee2mqtt #8026: HUSBZB-1 controller possibly hanging application

AutomanRJJ, 2021-07-12:
## What happened

I am trying to set this up on something that is not going to eradicate access to usb devices (Synology DSM 7) by setting up an older workstation as a Fedora Linux with Docker running on it to house my Zigbee and Z-Wave interface.
I am using the HUSBZB-1 controller... somewhere in here I read to use that configure for adapter: ezsp
I think maybe the application is internally hung, maybe its something to do with the ezsp adapter?  
I am wondering if that might have it hung, not even initializing. It doesn't even log into MQTT.
It appears to be up and working but I can't get to the frontend. I have tried every conceivable configuration I can think of to get to the frontend.
I am running it inside of a docker running on Fedora linux on ip 10.0.0.70. I know I can access other docker instances because I have portainer and zwave2mqtt up and running.
I have tried the default bridge network like portainer is on, and forwarding port 8080:8080. I have even tried switching over to use the host network. Firewall is enabled for port 8080 allowed through.
I switched to 8090 on host network, firewall configured to enable that port, no change so I don’t believe is a problem with docker nor firewalls.
No matter what I do, I can not access the frontend from my windows system (currently IP of windows PC 10.0.0.30) by going to http://10.0.0.70:8080 to try and get to the front end. I get ERR_CONNECTION_REFUSED from Edge and Chrome.

## Debug info
Zigbee2MQTT version:
    Docker Image created 7/1/2021, version 1.20.0 (commit #41b67fd)

Adapter hardware: HUSBZB-1

Included the configuration.yaml, the last log and the state.json
[Config_Logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/6803866/Config_Logs.zip)

FM-17, 2021-07-26:
Experiencing the same problem here with a similar setup. Running Zigbee2MQTT and mosquitto via docker on my home-assistant host (Ubuntu 20.04). 

Zigbee2MQTT version: `1.29.2`
Adapter Hardware: HUSBZB-1 
Adapter Firmware: [ncp-uart-sw-6.7.8.ebl](https://github.com/walthowd/husbzb-firmware/blob/master/ncp-uart-sw-6.7.8.ebl)

[Config_Logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/6874884/Configs.zip)

travipross, 2021-07-26:
+1 

Identical setup as @FM-17, and same issue.

AutomanRJJ, 2021-07-29:
I was able to get the HUSBZB-1 working on my docker instance.
I pulled down the dev channel;
  docker pull koenkk/zigbee2mqtt:latest-dev

you MUST configure the HUSBZB-1 as follows (obviously change to the appropriate port);

serial:
  port: /dev/ttyUSB1
  adapter: ezsp

advanced:
  baudrate: 57600

closed 2021-07-29 by AutomanRJJ

FM-17, 2021-07-29:
I'm still having trouble getting that configuration to work unfortunately. Specifically, I'm uncertain about my `server:` block syntax. Would you mind providing your fully zigbee2mqtt configuration.yaml file? 

I've tried setting `server:` to `mqtt://mqtt`, `mqtt://mqtt:1883`, `mqtt://<host IP>:1883`, and so on.

AutomanRJJ, 2021-07-30:
Sure;
[configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/6905509/configuration.yaml.txt)

RyanWor, 2022-01-29:
> Sure; [configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/6905509/configuration.yaml.txt)

Thanks for sharing this. Seems to work well. May I ask how many Zigbee devices you are running on your network with the HUSBZB-1 running under zigbee2mqtt? I am curious if this will scale up in network size as effective as a CC2652P does.

AutomanRJJ, 2022-01-30:
I am at 55 devices right now on my network.
By device type
  End devices: 35
  Router: 20 (Some of these are light bulbs)
By vendor
  LUMI: 13
  Keen Home Inc: 13
  eWeLink: 10
  Samjin: 8
  sengled: 6
  CREE : 5

xplreitr, 2022-03-11:
I can confirm that I was also able to link my HUSBZB-1 after 

1. Using the firmware mentioned by @FM-17 
2. Using the configuration posted by @AutomanRJJ 

Thanks for your help!

Hedda, 2022-03-14:
Since the EZSP support is still experimental recommend trying to keep discussions to that issue-> https://github.com/Koenkk/zigbee-herdsman/issues/319

kalyway101, 2022-05-01:
This seems like the best issue to comment on...

I have tried adding the configurations from this post but I'm still getting issues. Here is my Supervisor log:

`22-04-30 19:10:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
22-04-30 19:10:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/Poeschl/Hassio-Addons repository
22-04-30 19:10:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/tsightler/ring-mqtt-ha-addon repository
22-04-30 19:10:41 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-04-30 19:10:41 INFO (MainThread) [supervisor.store] Loading add-ons from store: 49 all - 0 new - 0 remove
22-04-30 19:10:41 INFO (MainThread) [supervisor.store] Loading add-ons from store: 49 all - 0 new - 0 remove
22-04-30 19:10:41 ERROR (SyncWorker_5) [supervisor.docker.interface] Container addon_a0d7b954_portainer is not running
22-04-30 19:10:43 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-04-30 19:10:48 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_core_zwave_js application
22-04-30 19:10:48 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-zwave_js with version 0.1.57
22-04-30 19:10:51 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_zwave_js
22-04-30 19:13:27 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
22-04-30 19:13:27 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
22-04-30 19:13:27 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
22-04-30 19:13:27 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)
22-04-30 19:13:27 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt Edge (45df7312_zigbee2mqtt_edge)`

I have updated to the latest firmware and this is my zigbee2MQTT-dev hassio config:

`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: >-
    https://github.com/Koenkk/zigbee-herdsman/issues/319tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: [OMITTED]
  password: [OMITTED]
serial:
  port: /dev/ttyUSB1
homeassistant: true
permit_join: false
frontend:
  port: 8099
advanced:
  baudrate: 115200
  channel: 11
  log_level: info
  log_output:
    - console
    - file`

AutomanRJJ, 2022-05-03:
I think you have the baudrate too high (115200).  I could not get it to work with zigbee2mqtt until I dropped its baudrate to 57600

kalyway101, 2022-05-03:
> I think you have the baudrate too high (115200). I could not get it to work with zigbee2mqtt until I dropped its baudrate to 57600

I actually think this is the real fix. Thank you.

fisher60, 2023-05-09:
> I was able to get the HUSBZB-1 working on my docker instance. I pulled down the dev channel; docker pull koenkk/zigbee2mqtt:latest-dev
> 
> you MUST configure the HUSBZB-1 as follows (obviously change to the appropriate port);
> 
> serial: port: /dev/ttyUSB1 adapter: ezsp
> 
> advanced: baudrate: 57600

Hello, here from modern times. I have the nortek gocontrol flashed with the [new firmware](https://github.com/walthowd/husbzb-firmware). This config is what worked for me (after hours of troubleshooting). I was missing the baudrate.

PeteAndersen, 2023-08-07:
Another modern times update: [Baud rate item moved under `adapter` key](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter). Example:

```yml
serial:
  port: /dev/ttyUSB1
  adapter: ezsp
  baudrate: 57600
```