Koenkk/zigbee2mqtt #8027: Cannot add some device to group - cannot read property 0

sybux, 2021-07-12:
## What happened
After creating a group with the web interface, I cannot add some devices. I can for exemple add a Hue device but not a Trust one. I'm getting the "cannot read property '0' to undefined" error

When adding a Hue device, the endpoint value is 11 but on this trust device it's 1 or 2. Don't think if it can explain the error. This is actually the only difference I've seen.

## What did you expect to happen

## How to reproduce it (minimal and precise)
add a device to the group

## Debug info
Zigbee2MQTT version: 1.20
Adapter hardware: CC2531
Adapter firmware version: can't remember where to check

Log extract :
info  2021-07-12 22:57:19: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x00432432432432/1","group":"gnd_corridor"},"error":"Failed to add from group (Cannot read property '0' of undefined)","status":"error","transaction":"d49t6-8"}'
error 2021-07-12 22:57:19: Failed to add from group (Cannot read property '0' of undefined)

Koenkk, 2021-07-13:
Please provide the debug log of this.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

sybux, 2021-07-13:
Not too much more infos :( 

`Info MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"d49t6-9"}'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x054543543534537/1","group":"gnd_corridor","transaction":"d49t6-10"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x054543543534537/1","group":"gnd_corridor"},"error":"Failed to add from group (Cannot read property '0' of undefined)","status":"error","transaction":"d49t6-10"}'
Error Failed to add from group (Cannot read property '0' of undefined)`

Koenkk, 2021-07-14:
I've added some more logging in the latest dev branch, can you provide the logging again with this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

sybux, 2021-07-14:
I'm sorry but no more output. one other strange thing i've just noticed in groups.js on line 48, I should have an info message about the requested action "adding to group" but I can't see it the logs.
I'm not enough good in JS to try debuging without your help.

sybux, 2021-07-14:
Here is a most reading logs about what I've just said before :

Working log of a hue light added to groups
```
debug 2021-07-15 00:00:56: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x0054353454354353/11","group":"gnd_corridor","transaction":"s5vz1-4"}'
info  2021-07-15 00:00:56: Adding 'gnd_office_light_cell' to 'gnd_corridor'
info  2021-07-15 00:00:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"gnd_office_light_cell","group":"gnd_corridor"},"type":"device_group_add"}'
info  2021-07-15 00:00:56: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x0054353454354353/11","group":"gnd_corridor"},"status":"ok","transaction":"s5vz1-4"}'
```

none working log of a trust light. You can notice that the 2nd line is not present here. We got directly the error
```
debug 2021-07-15 00:01:01: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x032132132132113/1","group":"gnd_corridor","transaction":"s5vz1-5"}'
info  2021-07-15 00:01:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x032132132132113/1","group":"gnd_corridor"},"error":"Failed to add from group (Cannot read property '0' of undefined)","status":"error","transaction":"s5vz1-5"}'
error 2021-07-15 00:01:01: Failed to add from group (Cannot read property '0' of undefined)
```

Koenkk, 2021-07-15:
Are you using the dev branch (or edge for the HA addon?)  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sybux, 2021-07-15:
yes,i'm running 1.20.0-dev commit: f8742344

Koenkk, 2021-07-15:
Turns out i added the logging to the wrong location, fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sybux, 2021-07-15:
This time we got a trace :) 

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x032132132132113/1","group":"gnd_corridor","transaction":"0krzs-1"}'
Debug TypeError: Cannot read property '0' of undefined at Groups.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/groups.js:298:101) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:408:81) at processTicksAndRejections (internal/process/task_queues.js:97:5)
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x032132132132113/1","group":"gnd_corridor"},"error":"Failed to add from group (Cannot read property '0' of undefined)","status":"error","transaction":"0krzs-1"}'
Error Failed to add from group (Cannot read property '0' of undefined)
```

Koenkk, 2021-07-16:
Can you share the data/database.db entry of `0x032132132132113`?

sybux, 2021-07-16:
here it is (of course this not the real name - security design :) )

`{"id":23,"type":"Router","ieeeAddr":"0x032132132132113","nwkAddr":53128,"manufId":4151,"manufName":"Trust International B.V.\u0000","powerSource":"Mains (single phase)","modelId":"ZLL-ExtendedColo","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":528,"inClusterList":[0,4,3,6,8,5,768,65535,25],"outClusterList":[0,4,3,6,8,5,768,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZLL-ExtendedColo","manufacturerName":"Trust International B.V.\u0000","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003"}},"genLevelCtrl":{"attributes":{"currentLevel":221}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1626470399471}`

Koenkk, 2021-07-17:
Should be fixed now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

However for this device you should add endpoint 2, not 1.

sybux, 2021-07-17:
I've just run an update to get the latest dev version. 1.20.0-dev commit: c6fdb253
But now, all my devices had left.
The devices.yaml and database.db contains all informations but the list on the frontend is empty.

sybux, 2021-07-17:
forget previous message, finally got it running....
But I still cannot add device to group

When adding endpoint 2
```
Info Adding 'gnd_corridor_wall' to 'gnd_corridor'
Error Failed to add from group (Command 0x032132132132113/1 genGroups.add({"groupid":3,"groupname":""}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53128 - 1 - 7 - 4 - 0 after 10000ms))
Error Device 'undefined/2' does not exist
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"undefined/2","group":"gnd_corridor"},"error":"Device 'undefined/2' does not exist","status":"error","transaction":"t9q1s-3"}'
Error Device 'undefined/2' does not exist
```

When adding endpoint 1
```
Info Adding 'gnd_corridor_wall' to 'gnd_corridor'
Debug Error: Command 0x032132132132113/1 genGroups.add({"groupid":3,"groupname":""}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53128 - 1 - 8 - 4 - 0 after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7)
Error Failed to add from group (Command 0x032132132132113/1 genGroups.add({"groupid":3,"groupname":""}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53128 - 1 - 8 - 4 - 0 after 10000ms))
```

Koenkk, 2021-07-19:
This timeout indicates that the device does not respond to the request, are you able to turn it on/off? If yes, that would mean that the device is able to receive commands, it could be that this device does not support groups in that case.

sybux, 2021-07-19:
Yes I can act on it with the Expose TAB.
One thing you glitch to my hear is the support of groups. The strange thing is that the light is Trust ZLED-TUNE9, but in ZB, it's reported as : 
Manufacturer           AduroSmart
Model                      81809/81813

and all my Trust light are the same. Which was not the case in previous release. But I can't remember when it changed 1.17, 1.18 ???

Koenkk, 2021-07-19:
Should be fixed now. (note that this will only be a cosmetical change)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sybux, 2021-07-19:
Ok so one step a head and one step back :(
Devices are now Trust signed, but I cannot control them anymore. I can control my philips, my power command Ikea but not trust device.

groenmarsmannetje, 2021-08-06:
I am also not able anymore to control the Trust bulbs.

sybux, 2021-08-06:
Back from holdiays. I've decided to switch back to stable version 1.20. I can switch back to dev if needed for more test.

wixoff, 2021-08-10:
I commented on the commit (https://github.com/Koenkk/zigbee-herdsman-converters/commit/12381ff4e1bfa4eb7f71234d2860e1879563f2a3) but my previously-working AduroSmart 81813 bulbs are now misrecognized as Trust ZLED-TUNE9, and they can't be controlled.  Rolling back to 1.20 solves it.

Koenkk, 2021-08-10:
@sybux @groenmarsmannetje should be fixed now

@wixoff should also be fixed now

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Koenkk, 2021-08-10:
Due to an outage in github actions i cannot update currently, will let you know once it is in the dev branch.

Koenkk, 2021-08-11:
Fixes are in the latest dev branch in 1 hour from now.

wixoff, 2021-08-11:
Thanks, the latest-dev package is working great for me now!

AndrewLinden, 2021-08-24:
I am using current dev branch and at the moment some things are mixed up.
I own Trust bulbs with model names: ZLED-TUNE9 and ZLED-RGB9. TUNE9 is white bulb with color temperature support and RGB9 is a RGB bulb.
My bulbs report the following modelID:
ZLED-TUNE9 -> ZLL-ColorTempera
ZLED-RGB9 -> ZLL-ExtendedColo

At the moment ZLED-RGB9 is detected as ZLED-TUNE9. I can't change colors, only change white color temperature.
I think maybe the error came from the database.db entry posted by @sybux in https://github.com/Koenkk/zigbee2mqtt/issues/8027#issuecomment-881727513 . Maybe this entry was for ZLED-RGB9 not ZLED-TUNE9

If I comment out the line added in https://github.com/Koenkk/zigbee-herdsman-converters/commit/12381ff4e1bfa4eb7f71234d2860e1879563f2a3 my ZLED-RGB9 is recognized as "ERIA colors and white shades smart light bulb A19/BR30 (81809/81813) by AduroSmart" and works correctly as RGB bulb.

Here are my database.db entries:

ZLED-RGB9 (detected as AduroSmart, working correctly):
`{"id":17,"type":"Router","ieeeAddr":"0x00158d0000aff39a","nwkAddr":25219,"manufId":4151,"manufName":"Trust International B.V.\u0000","powerSource":"Mains (single phase)","modelId":"ZLL-ExtendedColo","epList":[1,
2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":528,"in
ClusterList":[0,4,3,6,8,5,768,65535,25],"outClusterList":[0,4,3,6,8,5,768,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZLL-ExtendedColo","manufacturerName":"Trust International B.V.\u0000","powerSource":
1,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1629817969404}
`

ZLED-TUNE9:
`{"id":28,"type":"Router","ieeeAddr":"0x00158d0001cca6f3","nwkAddr":60107,"manufId":4151,"manufName":"Trust International B.V.\u0000","powerSource":"Mains (single phase)","modelId":"ZLL-ColorTempera","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,4,3,6,8,5,768,65535,65535,25],"outClusterList":[0,4,3,6,8,5,768,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":3},"attributes":{}},"genScenes":{"dir":{"value":3},"attributes":{}},"genOnOff":{"dir":{"value":3},"attributes":{}},"genLevelCtrl":{"dir":{"value":3},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"lightingColorCtrl":{"dir":{"value":3},"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1629819160859}`

Koenkk, 2021-08-24:
@sybux can you confirm this?

@AndrewLinden does it work with the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2949/files ? Off topic, good to see your hear, the real z2m OG :smile:

AndrewLinden, 2021-08-24:
> @AndrewLinden does it work with the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2949/files ? 

ZLED-RGB9 also needs `meta: {applyRedFix: true},`
Apart from that, all seems fine. I deleted one bulb of both types and joined again. Both detected correctly and worked fine in HA

> Off topic, good to see your here, the real z2m OG smile

I have been using z2m all the time. Many thanks for all the work you have put into this project!

sybux, 2021-08-29:
Sorry for no feedback before, it was Holiday time :)
I've just upgraded to version 1.21, Trust bulb are seen as trust but with a wrong identification. There are seen as ZLED-TUNE9 in stead of ZLED-RGB9, and I can't command them such as a simple power on/off

I'm getting the error :

Error Publish 'set' 'state' to 'gnd_corridor_wall' failed: 'Error: Command 0x00158d00011d64c7/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53128 - 1 - 166 - 6 - 11 after 10000ms)'

Do you think I need to switch to dev branch to test it again ?

sybux, 2021-08-29:
Finally updated to dev release, but still getting error on a simple power on/off :

```
debug 2021-08-29 23:09:09: Error: Command 0x00158d00011d64c7/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53128 - 1 - 8 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
info  2021-08-29 23:09:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'gnd_corridor_wall' failed: 'Error: Command 0x00158d00011d64c7/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 53128 - 1 - 8 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"gnd_corridor_wall"},"type":"zigbee_publish_error"}'
```

Koenkk, 2021-08-30:
@sybux can you first confirm https://github.com/Koenkk/zigbee2mqtt/issues/8027#issuecomment-881727513 is a ZLED-TUNE9 or ZLED-RGB9

sybux, 2021-08-30:
yes, my ZLED-RGB9 are shown as ZLED-ZLED-TUNE9. With that I can't controll this lights.

database export :
`{"id":23,"type":"Router","ieeeAddr":"0x00158d00011d64c7","nwkAddr":53128,"manufId":4151,"manufName":"Trust International B.V.\u0000","powerSource":"Mains (single phase)","modelId":"ZLL-ExtendedColo","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":528,"inClusterList":[0,4,3,6,8,5,768,65535,25],"outClusterList":[0,4,3,6,8,5,768,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZLL-ExtendedColo","manufacturerName":"Trust International B.V.\u0000","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003"}},"genLevelCtrl":{"attributes":{"currentLevel":221}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160916","swBuildId":"1000-0003","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1630268625551}`

Koenkk, 2021-08-31:
@sybux thanks, then the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2949/files should fix your problem

@sybux  or @AndrewLinden could you provide me the color temp range? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

sybux, 2021-08-31:
Is the changed already in the dev branch ?

Koenkk, 2021-08-31:
No, I'm waiting for the color temp range (https://github.com/Koenkk/zigbee2mqtt/issues/8027#issuecomment-909112342)

sybux, 2021-08-31:
oops sorry.... so next question, how can I run this command ?
zigbee2mqtt/DEVICE_FRIENDLY_NAME/set with payload {"read": {"cluster": "lightingColorCtrl", "attributes": ["colorTempPhysicalMin", "colorTempPhysicalMax"]}}

is there a cmdline way to send this ?

(sorry this is the part I need to develop in my skill)

Koenkk, 2021-08-31:
Might be easier to do via the frontend:

[image]

sybux, 2021-08-31:
here it is :)

`Info Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}`

Koenkk, 2021-08-31:
Thanks, merged it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sybux, 2021-08-31:
Ok @Koenkk, it's working fine for me. Bulb are identified fine, I can control on/off and color changing. I let @AndrewLinden confirmed if it also ok from his part.

closed 2021-10-08 by github-actions[bot]

martdj, 2022-10-08:
I actually have an issue with the ZLED-RGB9. It becomes unavailable after some time. When it is, I can still switch it on and off from the Zigbee2MQTT dashboard, but I can no longer control it from HA. Is it possible to fix this?

JackBaltzer, 2023-03-17:
> I actually have an issue with the ZLED-RGB9. It becomes unavailable after some time. When it is, I can still switch it on and off from the Zigbee2MQTT dashboard, but I can no longer control it from HA. Is it possible to fix this?

I am experiencing the same issue with my 2 ZLED-RGB9 bulbs. I just switched from ZHA to Zigbee2mqtt, and the bulbs turn offline within 30 minutes.
I can press reconfigure from zigbee2mqtt dashboard, and then the bulb will be available for a short while again, but will turn offline automatically.

I get a warning message: 

> Failed to ping 'Pære Trust Stue Lænestol Lampe' (attempt 1/1, Read 0x00158d00011d8a9d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43842 - 1 - 143 - 0 - 1 after 10000ms))

Any ideas or tips on how to get around this?

YannickvanVugt, 2023-08-02:
> I actually have an issue with the ZLED-RGB9. It becomes unavailable after some time. When it is, I can still switch it on and off from the Zigbee2MQTT dashboard, but I can no longer control it from HA. Is it possible to fix this?

> I am experiencing the same issue with my 2 ZLED-RGB9 bulbs. I just switched from ZHA to Zigbee2mqtt, and the bulbs turn offline within 30 minutes. I can press reconfigure from zigbee2mqtt dashboard, [...]

Hey guys,

Look above this comment, the issue with these lights is fixed now. :)

If you still have some and disabled availability check for these, you can enable it again.

Best regards,
Yannick