Koenkk/zigbee2mqtt #8032: max_mireds from bulbs in a group is not exposed to HA during auto discovery.

LucidityCrash, 2021-07-13:
## What happened
Light group auto discovered in HA from z2m doesn't have max_mireds property

## What did you expect to happen
Both the bulbs in the group (Zigbee 3.0 LED-bulb, RGBW LED) have a max_mireds value (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2473/commits/ecabe5b65288d662d0bce6e780851c698ba1c97e) I expected this to be exposed in the group discovery config and it is not so the discovered light entity in HA has a temperature slider that can go up to 500.  When I was adding this manually to HA the max temperature on the slicer is limited to max_mireds.

## How to reproduce it (minimal and precise)
Add lights that have a non default max_mireds (ie not 500) to a group.  The discovered light entity in HA has a temperature slider that can go up to 500

## Debug info
Zigbee2MQTT version:  1.20.0 commit: 41b67fd

jwoodard80, 2021-07-14:
I actually just noticed this as well. I have 4 identical bulbs that run 153-370. I was able to edit the definition of the bulb and a PR was submitted by someone but the group does not have a place for that.

Might I suggest an option in the configuration.yaml to define a max/min if we are not pulling a value?

Koenkk, 2021-07-14:
Added, please let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jwoodard80, 2021-07-14:
Thank you very much for this! Appreciate the quick addition!

Koenkk, 2021-07-15:
Need some input on this discussion I'm having with @sjorge  here: https://github.com/Koenkk/zigbee2mqtt/commit/515599ad139c27ddf9a776a5023f9916f1273456#r53518817 

Currently if a group has 2 bulbs with different color temperatures ranges:
- bulb A: range 100 - 350
- bulb B: range 150 - 400

In the current implementation (solution A), the range is set to the maximum one (in this case 100 - 400). However if you would now set the group color temperature to lets say 370 it would result in:
- bulb A: 350 (max since it cannot handle 370) <-- inconsistent state
- bulb B: 370 (as expected)

An alternative implementation (solution B) could be that the range is set such that it fits all bulbs (in this case 150 - 350). The downside is that you cannot set bulb B to the max range anymore via the group.

What would be the preferred solution (a or b)?

Koenkk, 2021-07-16:
Implemented solution B now thanks for your input in https://github.com/Koenkk/zigbee2mqtt/commit/515599ad139c27ddf9a776a5023f9916f1273456#r53548408

closed 2021-07-16 by Koenkk