Koenkk/zigbee2mqtt #8033: 'value_template' option is deprecated warning for switches exposed as lights

LucidityCrash, 2021-07-13:
## What happened
Warnings in HA logs
`2021-07-13 16:47:57 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_basic] The 'value_template' option is deprecated, please replace it with 'state_value_template'`
these are generated by 2 switches I have configured to be lights using https://www.zigbee2mqtt.io/integration/home_assistant.html#exposing-switch-as-a-light

looking at the config topic for the device in question the json does indeed include 
`"value_template": "{{ value_json.linkquality }}"`

## What did you expect to happen
These warning not to be present.

## How to reproduce it (minimal and precise)
Add a switch as a light using https://www.zigbee2mqtt.io/integration/home_assistant.html#exposing-switch-as-a-light after discovery in HA if the warning is not present restart HA and they should be logged during startup.

## Debug info
Zigbee2MQTT version: 1.20

chumbazoid, 2021-07-19:
Also experiencing this issue and unclear how to resolve. Originates w/ Home Assistant 2021.7 [Deprecate support for undocumented value_template in MQTT light #51589](https://github.com/home-assistant/core/pull/51589).

Koenkk, 2021-07-19:
Can you try with:

```yaml
devices:
  "0x12345678":
    friendly_name: my_switch
    homeassistant:
      switch:
        type: light
        object_id: light
        discovery_payload: {value_template: null, state_value_template: '{{ value_json.state }}'}
```

brokenmass, 2021-07-23:
@Koenkk could it be related to https://github.com/Koenkk/zigbee2mqtt/issues/6503 ?

Koenkk, 2021-07-24:
@brokenmass can you try the config from https://github.com/Koenkk/zigbee2mqtt/issues/8033#issuecomment-882755004 and post the debug log when starting z2m in case it does not work.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

LucidityCrash, 2021-07-24:
Sorry for the delay ... I tried adding that and it doesn't appear to have had any effect.

```
'0x5c0272fffece4eb8':
  friendly_name: Back Door
  homeassistant:
    switch:
      type: light
      object_id: light
      name: back_door
      discovery_payload:
        value_template: null
        state_value_template: '{{ value_json.state }}'
  optimistic: true
'0x5c0272fffece6d93':
  friendly_name: Patio Door1
  homeassistant:
    switch:
      type: light
      object_id: light
      name: patio_door
      discovery_payload:
        value_template: null
        state_value_template: '{{ value_json.state }}'
  optimistic: true
```

Log file attached 

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6872273/log.txt)

Koenkk, 2021-07-24:
The following config works for me:

```yaml
devices:
  '0x00158d00018255df':
    friendly_name: '0x00158d00018255df'
    homeassistant:
      switch:
        type: light
        object_id: light
      light:
        value_template: null
        state_value_template: '{{ value_json.state }}'
```

Also updated the docs with this.

Mr-Groch, 2021-08-06:
> Also updated the docs with this.

Unfortunately, there is no value visible in docs for `state_value_template`:

[image]

Koenkk, 2021-08-06:
@Mr-Groch fixed.

closed 2021-09-13 by github-actions[bot]

adizag, 2022-02-16:
@Koenkk 
Updated to HA latest, updated Z2M conf file to latest format, but still getting no entities and HA log error.
What could get wrong?

HA screenshot:
[image]

Z2M conf:
```
  "0xec1bbdfffe2c55be":
    friendly_name: Mamad 2nd
    homeassistant:
      switch_l1:
        type: light
        object_id: mamad_2nd_light_l1
      light_l1:
        name: mamad_2nd_l1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      switch_l2:
        type: light
        object_id: mamad_2nd_light_l2
      light_l2:
        name: mamad_2nd_l2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
```

HA log:
```
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Mamad 2nd/l2/set', 'device': {'identifiers': ['zigbee2mqtt_0xec1bbdfffe2c55be'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Mamad 2nd'}, 'json_attributes_topic': 'zigbee2mqtt/Mamad 2nd', 'name': 'Mamad 2nd mamad 2nd light l2', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Mamad 2nd', 'unique_id': '0xec1bbdfffe2c55be_mamad_2nd_light_l2_zigbee2mqtt', 'value_template': '{{ value_json.state_l2 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Mamad/l1/set', 'device': {'identifiers': ['zigbee2mqtt_0x5c0272fffe8ef714'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Mamad'}, 'json_attributes_topic': 'zigbee2mqtt/Mamad', 'name': 'Mamad mamad light 1', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Mamad', 'unique_id': '0x5c0272fffe8ef714_mamad_light_1_zigbee2mqtt', 'value_template': '{{ value_json.state_l1 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return s [... cut ...]
Zigbee2MQTT:debug 2022-02-16 10:13:03: Received Zigbee message from 'Mamad 2nd', type 'attributeReport', cluster 'genBasic', data '{"65506":19,"appVersion":128}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-16 10:14:06: Received Zigbee message from 'Stairs Down', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-16 10:15:27: Received Zigbee message from 'Mamad', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
```