Koenkk/zigbee2mqtt #8035: Lots of timeout recenlty

arturdobo, 2021-07-13:
## What happened

I've started experiencing a lot of timeouts in my network recently. Typically the only thing I need to do is send a couple of actions sequentially to achieve significant lag like turning a light on after a couple of seconds or timeouts. 

I thought maybe there is some performance issue because of lunching zigbee2mqtt + mosquito on raspberry pi zero. After migration to raspberry pi 3b + the issue still exists. 

## What did you expect to happen

Not so many timeouts. It worked previously almost flawlessly. 

## How to reproduce it (minimal and precise)

See attached logs. I set ON on `livingroom_moss_and_plants_light_switch` and next
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6811140/log.txt)
 it failed on setting OFF after couple of seconds

Zigbee2MQTT version: 1.20.0 commit: 41b67fdd
Adapter hardware: CC2652R (zzh)
Adapter firmware version: 20210120

bartekluk, 2021-07-14:
Are you using an SSD or other usb drive with your Pi? I had an issue when noise from USB drive would mess with my zigbee dongle when I moved it away by using an extension cable all went away.

arturdobo, 2021-07-14:
Yes, an SSD drive is connected to rpi next to Zigbee dongle. Let me test your suggestion

arturdobo, 2021-07-15:
Unfortunately, it didn't help

arturdobo, 2021-07-16:
I flashed `CC2531` to compare and check whether the adapter is not an issue. In the beginning, it worked quite nicely, but after a couple of hours (I didn't measure it precisely), it stops work almost completely, and replugging was needed. For sure I didn't expect too much from this device. @Koenkk do you think this issue can be firmware-related?

Koenkk, 2021-07-16:
Timeouts are caused by interference (e.g. from SSD close to the adapter antenna)
- Make sure to connect your adapter through a USB extension cable and position it +30 cm from your computer
- Check wifi: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html

arturdobo, 2021-07-17:
The adapter is now connected using an extension cable 1.5 m long and I tested a lot of positions. Behaviour is more or less the same. When it comes to wifi interference my network at home works on channel 1 while the Zigbee adapter is set to work on channel 25. They shouldn't interfere each other

davidnite, 2021-07-18:
I'm having the same problem with a zzh, and it is definitely not related to interference.   I am not running a Pi, but an ubuntu server running docker with a 1.5 meter cable to separate the zzh adapter from the server.  Adapter has been in place since January, with no moves or firmware changes.  Something seems to have happened with a recent update.

I am seeing less lag when issuing commands from home assistant, but when I attempt to use to use my Tuya TS0044 switches I generally have to "wake them up" by pressing the button and then pressing it again to get the desired result.  This specific issue has only been happening since the latest update.  It's very reminiscent of when I had a CC2531 with 20+ devices on a single network.

arturdobo, 2021-07-19:
I ordered slaeh's  CC2652RB adapter to compare whether it behaves similarly or not. Once it arrives I'll share with you my result

nashant, 2021-07-24:
I'm getting *loads* of timeouts. It's particularly bad when trying to pair new devices and send commands to existing devices at the same time. Will try downgrading to 1.19 from 1.2 0 to see if that helps. Hopefully no database migrations 1.19->1.20!

arturdobo, 2021-07-24:
I got the mentioned above adapter and I'm going to perform some tests during the weekend and compare it with zzh

arturdobo, 2021-07-24:
@Koenkk how to figure out the adapter with router firmware is connected to the network? I flashed both `CC2531` and zzh using a router firmware, but I didn't find any info on how to check they're connected.

Koenkk, 2021-07-25:
@arturdobo you can check the networkmap (can be generated in the frontend), if it has outgoing lines (children) it's connected.

closed 2021-09-01 by github-actions[bot]